raspiblitz/raspiblitz #3308: WebUI Balance is too low

Yonben, 2022-08-31:
Hi,

Just wanted to inform, I've been getting validation errors on "sending" fromthe WebUI telling me my balance is too low to send funds when it's not (and I can still submit and it works).

Willing to work on fixing it, just wanted to at least document the bug first :)

Yonben, 2022-08-31:
Similarly:
[image]
It seems I don't use 97% CPU. Is it something I'm not reading properly in `top` or a display error in the webui?

@fusion44 I'd love some direction on how to work on the webui to try and tackling these myself :)

cstenglein, 2022-09-03:
Hi @Yonben 

The first bug is related to https://github.com/cstenglein/raspiblitz-web/issues/401 - sorry for not documenting much, I'm mostly working alone and time is limited. If you want I can write more information in the issues if you want to work on them.

Regarding the CPU: It's calculated from the load. See the formula the WebUI uses here: https://github.com/cstenglein/raspiblitz-web/blob/master/src/pages/Home/HardwareCard.tsx#L29

Here is an article which explains the load averages: https://www.tecmint.com/understand-linux-load-averages-and-monitor-performance/

So that means you look at the load average, you are using **3.78** Cores out of 4 in the last minute => 94,5%, so seems right but I understand if it's confusing. Maybe we should display that differently?

Yonben, 2022-09-05:
@cstenglein thanks for details!
I'll take a look at the repo tonight and check what i can find. I'll comment on the issue there :)

cstenglein, 2022-09-11:
FYI @Yonben fixed that issue in https://github.com/cstenglein/raspiblitz-web/pull/449

Can be closed IMO

closed 2022-09-13 by rootzoll