raspiblitz/raspiblitz #3315: Blockchain progress stuck

merce2, 2022-09-10:
Hello! I have tried to synch the blockchain and for the first 78 % everything went fine. But now there is no progress anymore and it seems like the raspiblitz is trying to restart all the time.

[image]

Thanks for your help!

rootzoll, 2022-09-13:
The restarts dont look good. Exit with CTRL+c to the terminal and the execute command `debug` - please post the sections under `*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***` to check for errors.

cstenglein, 2023-06-07:
@merce2  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.