raspiblitz/raspiblitz #3316: Password WebUI don't work

Tinus016, 2022-09-11:
I cant login on the WebUI whit my SSH password, (password A) on the web page. I have already changed the password once but without success, logging in via SSH is possible. I've tried to patch the WebUI it worked but stil not able to log in.

cstenglein, 2022-09-11:
Do you have special characters except `.` `_` `-` in your password A?

Tinus016, 2022-09-11:
Nope only <code>-</code>

rootzoll, 2022-09-13:
@fusion44 we need better debug data on this - what would be the best way to get any possible exceptions happening here? Is there anything for v1.8.1 we can add to capture this better?

rootzoll, 2022-09-16:
TODO: make ssh script that test password give more debug details when password is wrong and loop this thru API. Also make the rest of the API code on checking password giving back more debug info in general when password is wrong.

hashes4merkle, 2022-11-26:
Also seeing this bug on my node. @cstenglein -  Any notable areas I should review to debug this issue?

cstenglein, 2022-12-17:
@hashes4merkle check if the API is running

`sudo service blitzapi status`

rootzoll, 2022-12-23:
Had a problem with a password containing a `-` on SSH & WebUI on testing RC1 - so todo for RC2:
- [ ] check setup & recover passwords with special chars for RC2

closed 2023-05-10 by rootzoll