raspiblitz/raspiblitz #3324: Make sure Proxmos guide is linked in documentation

rootzoll, 2022-09-21:
https://cercatrova.blog/en/install-raspiblitz-on-proxmox/

cercatrova21, 2022-09-22:
I can copy the content to a md file here on github. I'll create under alternative.platforms a new folder called "Proxmox" and save the guide there. Ok?

rootzoll, 2022-09-22:
Sounds good with me, thx.

cercatrova21, 2022-09-24:
Yes this would be my update strategy. Will test with the 1.8.1 release and add that to the guide

Surenic, 2022-10-04:
Not sure if anyone else has tested this yet, but I did and ran into the problem, that RTL is not working in my VM, anything else is.

At first I migrated the node through the ssd, RTL was installed. HTTP not working at all, HTTPS calling a bad gateway.

A fresh install did not let me install RTL by ssh menu. It failed, asking me to install it manually, which I did. This was leading to the errors above. Guess this is not priority at all, but maybe someone has a fast idea.

cercatrova21, 2022-10-04:
Yes looks like c-lightning is the problem here. I tested my setup with LND only and got no problems with REST connection

Surenic, 2022-10-04:
Found a solution, seems that Raspiblitz by itself did not install some dependencies. Actually I was surprised of how fast RTL was installed.

´´´
cd /home/bitcoin/c-lightning-REST
sudo -u bitcoin npm install
sudo -u bitcoin npm update
´´´

CLREST was starting then, RTL is running

nazgul6, 2022-10-04:
Great guide. I would add that a VM disk can be used as a storage for blockchain data as well. No hiccups here after few months.

SomeBTChomer, 2022-10-05:
> Not sure if anyone else has tested this yet, but I did and ran into the problem, that RTL is not working in my VM, anything else is.
> 
> At first I migrated the node through the ssd, RTL was installed. HTTP not working at all, HTTPS calling a bad gateway.
> 
> A fresh install did not let me install RTL by ssh menu. It failed, asking me to install it manually, which I did. This was leading to the errors above. Guess this is not priority at all, but maybe someone has a fast idea.
> 
> EDIT: Maybe this causes the trouble, c-lightning REST is not able to start
> 
> ```
> Okt 04 10:38:39 RaspiBlitz systemd[1]: clrest.service: Scheduled restart job, restart counter is at 85.
> Okt 04 10:38:39 RaspiBlitz systemd[1]: Stopped c-lightning-REST daemon for mainnet.
> Okt 04 10:38:39 RaspiBlitz systemd[1]: Started c-lightning-REST daemon for mainnet.
> Okt 04 10:38:39 RaspiBlitz node[1616508]: node:internal/modules/cjs/loader:936
> Okt 04 10:38:39 RaspiBlitz node[1616508]:   throw err;
> Okt 04 10:38:39 RaspiBlitz node[1616508]:   ^
> Okt 04 10:38:39 RaspiBlitz node[1616508]: Error: Cannot find module 'express'
> Okt 04 10:38:39 RaspiBlitz node[1616508]: Require stack:
> Okt 04 10:38:39 RaspiBlitz node[1616508]: - /home/bitcoin/c-lightning-REST/bitcoin/app.js
> Okt 04 10:38:39 RaspiBlitz node[1616508]: - /home/bitcoin/c-lightning-REST/bitcoin/cl-rest.js
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Function.Module._load (node:internal/modules/cjs/loader:778:27)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Module.require (node:internal/modules/cjs/loader:1005:19)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at require (node:internal/modules/cjs/helpers:102:18)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Object.<anonymous> (/home/bitcoin/c-lightning-REST/bitcoin/app.js:1:13)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Module._compile (node:internal/modules/cjs/loader:1103:14)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1157:10)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Module.load (node:internal/modules/cjs/loader:981:32)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Function.Module._load (node:internal/modules/cjs/loader:822:12)
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     at Module.require (node:internal/modules/cjs/loader:1005:19) {
> Okt 04 10:38:39 RaspiBlitz node[1616508]:   code: 'MODULE_NOT_FOUND',
> Okt 04 10:38:39 RaspiBlitz node[1616508]:   requireStack: [
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     '/home/bitcoin/c-lightning-REST/bitcoin/app.js',
> Okt 04 10:38:39 RaspiBlitz node[1616508]:     '/home/bitcoin/c-lightning-REST/bitcoin/cl-rest.js'
> Okt 04 10:38:39 RaspiBlitz node[1616508]:   ]
> Okt 04 10:38:39 RaspiBlitz node[1616508]: }
> Okt 04 10:38:39 RaspiBlitz systemd[1]: clrest.service: Main process exited, code=exited, status=1/FAILURE
> Okt 04 10:38:39 RaspiBlitz systemd[1]: clrest.service: Failed with result 'exit-code'.
> ```

I think the solution is here: https://github.com/rootzoll/raspiblitz/issues/3297

rootzoll, 2024-04-30:
For future update of the documentation also the information here is to be considered to include:
https://github.com/raspiblitz/raspiblitz/discussions/4282#discussioncomment-9172010

closed 2025-11-11 by rootzoll