raspiblitz/raspiblitz #3326: Logs filled whole SD Card and (maybe some autoclearing process needed?)

donovan-ringley, 2022-09-22:
I have a Raspi with 32GB SD Card. At some point, the node simply stopped working because "No space left on device". 

The full logs was probably not root cause of my issues. I think that for some other reason bitcoind kept restarting filling the logs.

However I would like to ask to add some mechanism that wont allow logs from filling up whole SD card. (Probably some autodelete.)

This was quite bad situation for me. I am not home much and my node died when I was abroad so I lost access to it. I was quite unhappy that this happed for such a reason.

rootzoll, 2022-09-23:
Logs are already on auto-rotate-delete  ... could you spot which log made the problem? Which bonus apps do you have installed?

donovan-ringley, 2022-09-23:
Sadly I have already deleted them. But I believe it was `/var/log/syslog.1` and maybe one more that I am not able to recall.

donovan-ringley, 2022-09-23:
My enabled services:
[image]

rootzoll, 2022-09-23:
TODO: Setup test device with this services, keep running for some days and check for growing logs.

rootzoll, 2022-12-13:
OK logs are reduced. Open for final testing.

closed 2022-12-15 by rootzoll