raspiblitz/raspiblitz #3329: [Discussion] BItcoin Core Initial Block Download over clearnet, I2P and Tor by default

openoms, 2022-09-24:
Aware of this is a sensitive topic, but discussing is necessary. 

## The problem

The environment keeps changing with Tor currently practically broken: https://status.torproject.org/

Having report of IBD on RPi4+SSD taking 2 weeks and misidentifying the bottleneck being the RP, when it is actually Tor.

See for comparison the 3 years old tweet about the full sync over Tor-only in 80 hours: https://twitter.com/openoms/status/1147393450171674624

Possible improvements:

## Hybrid mode for Bitcoin Core

 Simply need to delete the line:
```
onlynet=onion
```
from your bitcoin.conf (menu - SYSTEM - bitcoinCONF) to sync connect to peers over clearnet also.
Just be aware that your ISP will see that you are using bitcoin - it is obvious anyway even over from the traffic and download spikes on the new blocks and is ok in most countries.

## I2P as an alternative network
can achieve significant speed improvements, but more peers are needed.
See how to set up easily in the command line:
https://github.com/rootzoll/raspiblitz/issues/2413#issuecomment-1013855483

## what can we do?
- start nodes in hybrid mode (for bitcoin core, not lightning) by default with a warning
- implement I2P to be used by default as well as Tor
- wonder if adding a warning and an option to the web-gui to avoid exposing the clearnet traffic is reasonably easy @cstenglein?

Comments and suggestions are most welcome.

1440000bytes, 2022-09-24:
Instead of removing `onlynet=onion`, we could add `onlynet=i2p` so that Tor and i2p are used for outbound connections.

In case this doesn't help, IPv4 and IPv6 peers are required, a free VPN like [riseup](https://riseup.net/en/vpn/linux) could be used.

Emzy, 2022-09-24:
As you said, the only downside is that your ISP would see more clearly you run a Bitcoin Core node.
It is very important for your LN node to get all new blocks and don't get stuck.
I think running more than one network (Clearnet + Tor) is the safer option anyway. 
Clearly this should be an option, or even the default option.

openoms, 2022-09-24:
Adding I2P by default seems to be a no-brainer and probably the advanced option to activate clearnet for Bitcoin Core is the best choice for a start.

kristapsk, 2022-09-24:
I would suggest you to start with adding I2P by default for Bitcoin Core. As I already mentioned in #2413, I2P is now optional for RaspiBolt, you can likely just copy necessary steps from there. https://raspibolt.org/guide/bonus/bitcoin/i2p.html

Also, CJDNS is another option for Bitcoin Core before clearnet, but haven't yet tried that one myself.

ajtowns, 2022-09-24:
Could publish `*.blk` files over https somewhere, download them, and import them via `-loadblock=blkN.dat` -- that still does the IBD work, just avoids downloading over P2P. Can probably cat them together before publishing if files larger than 100MB are okay; maybe cat every 40 of them together for ~80 files at ~4GB each?

rootzoll, 2022-09-24:
@ajtowns in the past we had torrent download of blockchain data - it was quite messy & I am glad that we go rid of it .. http would introduce a central service I would like to avoid as much as possible

rootzoll, 2022-09-24:
> Adding I2P by default seems to be a no-brainer and probably the advanced option to activate clearnet for Bitcoin Core is the best choice for a start.

Adding I2P would be a quick fix for v1.8.1 and we could then see if the the practical results are already satisfying. With the WebUI we have now the possibility to present options to the user to speed up syncing ... adding/activating clearnet could be one of them that we could start offering with a v1.8.3

benjamin-wilson, 2022-10-13:
It seems removing `onlynet=onion` alone is insufficient, blocks were not syncing at full speed. After commenting out the whole Tor section, things started syncing at full speed.

Darth-Coin, 2022-11-23:
Is it possible to have an option in the first step when user open the webUI and choose the data disk etc, also to set through which network want to have the IBD?

rootzoll, 2022-11-27:
> Is it possible to have an option in the first step when user open the webUI and choose the data disk etc, also to set through which network want to have the IBD?

To add more options to the WebUI for setups is the plan - but will take some time to add step by step.

> Also to set a local IP node for example from where could have a direct connection to sync
related to this: ´bitcoin-cli addnode "192.168.x.x:8333" "onetry"´ where 192.168.x.x is another bitcoin node in the same LAN.

OK that is an interesting idea I never tried. Do you have some experience on how fast the transfere/validation of blocks with another node as peer in the LAN? I thinkp the limiting factor here will be the "block validation" and not the transfere - so might not be a big speed up against clearnet peers the internet in general - but happy for some experience measurements.

rootzoll, 2022-11-27:
TODO:

- [ ] test & merge I2PD PR  #3330
- [ ] make new issue for adding option for clearnet mining to webui and SSH blockchain sync options (with 1.10.0)

Darth-Coin, 2022-11-27:
> I thinkp the limiting factor here will be the "block validation" and not the transfere - so might not be a big speed up against clearnet peers the internet in general - but happy for some experience measurements.

Yes, I did that process 3 times in the past. I always have a bitcoin node in LAN that only is doing this - keep staying syncronized. So in case I have a failure for others (shit can happen), I just take the blocks from it.

Yes, this procedure is not super fast as we want to think, it still have to validate the blocks, but at least:
- you don't have to stop the continuous sync on the "mother node"
- you don't have to take the drives and put them into another PC to copy the data manually
- you do not make too much internet traffic, because all the "new sync" is done between LAN
- once is finished, you can just remove the local sync and leave the new node to sync normally, without even restart or shut down, all very smooth.

Yes, not every node runner will have another node in the house syncing... but is a nice option.

openoms, 2022-11-27:
@Darth-Coin addnode on LAN is a good option, but should remain a manual option in CLI due to implementation complexity.
The good news is my I2P nodes remain connected on the LAN even with the internet down.
I think I2P adds the required speed-up for the random IBD as well.
Great summary here: https://jonatack.github.io/articles/using-alternative-p2p-networks-with-bitcoin-core

Darth-Coin, 2022-11-28:
Agreed. Not all users would have another node in LAN. And those who have it, already know what to do to connect with it.

rootzoll, 2022-12-07:
OK ip2 got merged #3330 for v1.9.0 release and created #3500 to consider clearnet option in v1.10.0 - closing this issue
if you wanna help testing wait for v1.9.0rc1 and report on issue #2413

closed 2022-12-07 by rootzoll