raspiblitz/raspiblitz #3331: WebUI: after reinstalling the app page is not openning

1thales, 2022-09-25:
So i deleted webUI then decided to reenable it again. Homepage is opening but when i go to the apps page i am met with a forever rotating wheel :(

cstenglein, 2022-11-10:
If you update the api & webUI, it should be fixed now:

```
sudo ./config.scripts/blitz.web.api.sh update-code
sudo ./config.scripts/blitz.web.ui.sh update
```

Otherwise fixed with the next version.

Problem was the branches were broken, we will change the branching strategy with the next version, so it doesn't break midway.

1thales, 2022-11-15:
this is what i get when i run the first command

admin@192.168.0.93:~ ₿ sudo ./config.scripts/blitz.web.api.sh update-code
ls: cannot access '/etc/systemd/system/blitzapi.service': No such file or directory

cstenglein, 2022-11-15:
> this is what i get when i run the first command

looks like you uninstalled the API, as seen by the comment:

>  #blitzapi not active

Then you can use:

```sudo ./config.scripts/blitz.web.api.sh on```

1thales, 2022-11-15:
`admin@192.168.0.93:~ ₿ sudo ./config.scripts/blitz.web.api.sh on
# INSTALL Web API ...
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
adduser: The user `blitzapi' already exists.
# allowing user as part of the bitcoin group to RW RPC hook
chmod: cannot access '/home/bitcoin/.lightning/bitcoin': No such file or directory
chmod: cannot access '/home/bitcoin/.lightning/bitcoin/lightning-rpc': No such file or directory
cat: /home/bitcoin/.lightning/config: No such file or directory
tee: /home/bitcoin/.lightning/config: No such file or directory
rpc-file-mode=0660
Cloning into '/home/blitzapi/blitz_api'...
remote: Enumerating objects: 2199, done.
remote: Counting objects: 100% (794/794), done.
remote: Compressing objects: 100% (307/307), done.
remote: Total 2199 (delta 602), reused 617 (delta 485), pack-reused 1405
Receiving objects: 100% (2199/2199), 1.23 MiB | 3.99 MiB/s, done.
Resolving deltas: 100% (1487/1487), done.
Already on 'main'
Your branch is up to date with 'origin/main'.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring cffi: markers 'python_version >= "3.9" and python_version < "4.0" and implementation_name == "pypy"' don't match your environment
Ignoring colorama: markers 'python_version >= "3.9" and python_version < "4.0" and platform_system == "Windows"' don't match your environment
Ignoring py: markers 'python_version >= "3.9" and python_version < "4.0" and implementation_name == "pypy"' don't match your environment
Ignoring pycparser: markers 'python_version >= "3.9" and python_version < "4.0" and implementation_name == "pypy"' don't match your environment
Requirement already satisfied: aiohttp==3.8.1 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 1)) (3.8.1)
Requirement already satisfied: aiojobs==1.0.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 74)) (1.0.0)
Collecting aioredis==2.0.1
  Downloading https://www.piwheels.org/simple/aioredis/aioredis-2.0.1-py3-none-any.whl (71 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 71.2/71.2 kB 776.9 kB/s eta 0:00:00
Requirement already satisfied: aiosignal==1.2.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 85)) (1.2.0)
Requirement already satisfied: anyio==3.6.1 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 89)) (3.6.1)
Requirement already satisfied: async-timeout==4.0.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 93)) (4.0.2)
Collecting attrs==22.1.0
  Downloading https://www.piwheels.org/simple/attrs/attrs-22.1.0-py2.py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.8/58.8 kB 1.1 MB/s eta 0:00:00
Requirement already satisfied: cchardet==2.1.7 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 101)) (2.1.7)
Collecting certifi==2022.9.24
  Downloading https://www.piwheels.org/simple/certifi/certifi-2022.9.24-py3-none-any.whl (161 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 161.1/161.1 kB 3.6 MB/s eta 0:00:00
Collecting charset-normalizer==2.1.1
  Downloading https://www.piwheels.org/simple/charset-normalizer/charset_normalizer-2.1.1-py3-none-any.whl (39 kB)
Requirement already satisfied: click==8.1.3 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 204)) (8.1.3)
Requirement already satisfied: deepdiff==5.8.1 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 211)) (5.8.1)
Collecting deprecated==1.2.13
  Downloading https://www.piwheels.org/simple/deprecated/Deprecated-1.2.13-py2.py3-none-any.whl (9.5 kB)
Collecting fastapi-plugins==0.10.0
  Downloading https://www.piwheels.org/simple/fastapi-plugins/fastapi_plugins-0.10.0-py3-none-any.whl (23 kB)
Collecting fastapi==0.82.0
  Using cached https://www.piwheels.org/simple/fastapi/fastapi-0.82.0-py3-none-any.whl (55 kB)
Collecting frozenlist==1.3.1
  Using cached frozenlist-1.3.1-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (159 kB)
Collecting googleapis-common-protos==1.56.4
  Downloading https://www.piwheels.org/simple/googleapis-common-protos/googleapis_common_protos-1.56.4-py2.py3-none-any.whl (211 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 211.7/211.7 kB 1.9 MB/s eta 0:00:00
Collecting grpcio-tools==1.48.1
  Using cached grpcio_tools-1.48.1-cp39-cp39-manylinux_2_17_aarch64.whl (31.5 MB)
Collecting grpcio==1.48.1
  Using cached grpcio-1.48.1-cp39-cp39-manylinux_2_17_aarch64.whl (63.9 MB)
Collecting h11==0.14.0
  Downloading https://www.piwheels.org/simple/h11/h11-0.14.0-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.3/58.3 kB 732.2 kB/s eta 0:00:00
Requirement already satisfied: hiredis==2.0.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 388)) (2.0.0)
Collecting idna==3.4
[... 78 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 70, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 101, in main
    return self._main(args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
    self.handle_pip_version_check(options)
[... cut ...]
  File "/usr/lib/python3.9/logging/__init__.py", line 1595, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1657, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 948, in handle
    self.emit(record)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 179, in emit
    self.handleError(record)
Message: '[present-rich] %s'
Arguments: (UpgradePrompt(old='22.1.2', new='22.3.1'),)
error='pip install failed'`

1thales, 2022-11-15:
didn't work :(

ChuckNorrison, 2022-11-15:
faced the same errors here
https://github.com/rootzoll/raspiblitz/issues/3262#issuecomment-1272602987

Its still unknown how this happened. Fix manually as described or reflash sd card works for me.

cstenglein, 2022-11-15:
@rootzoll @fusion44 can we fix the pip issue in the scripts / the API?

Also occurred on my test node.

rootzoll, 2022-12-11:
@cstenglein does this problem still exists when testing with the new dev image?

cstenglein, 2022-12-12:
No, issue was fixed some time ago in the blitzAPI.

Main issue for using that fix was https://github.com/rootzoll/raspiblitz/pull/3466 .

Can be closed now or with 1.9.0 release