raspiblitz/raspiblitz #3340: No update option after emergency mode

klnit, 2022-10-02:
Hi all,

I have my blitz running for years and did several upgrades. So I know how it works but now I am stuck. I had 1.7.2 running and accidently pulled the power plug. After that it only started into emergency mode and I couldn't login. As I wanted to upgrade anyway I burned the SD with version 1.8.0. But the welcome screen only states the options "FreshSetup", "FromBackup" and "Shutdown". No option to "Update". Then I tried a fresh 1.7.2 again but it is the same.
I couldn't find something similar on the internet. Can someone point to what I can do now? (I have a thumbdrive always plugged in where the channel states should be backed up. But I have no system backup I could load.)

Thanks

rootzoll, 2022-10-06:
> (I have a thumbdrive always plugged in where the channel states should be backed up. But I have no system backup I could load.)

OK remove and keep that thumbdrive safe.

> But the welcome screen only states the options "FreshSetup", "FromBackup" and "Shutdown". No option to "Update".

This should only happen if the new sd card image cannot find the "raspiblitz.conf" file under "/mnt/hdd". Most prob thru the power outage the HDD/SSD does not mount correctly and then it cannot find that file and treats the HDD/SSD as a fresh drive.

If you have a Linix laptop you can try to mount the HDD/SSD there for a test. Otherwise this will become a tricky recovery journey. Can you check on your laptop what is on the Thumbdrive you have?

klnit, 2022-10-07:
Klappt!
It is working now. I just plugged it into an Ubuntu, it was mounting and I could see the files. Then unmounted. Now it is in the updating process with the raspi. Thumbdrive looks good, too. A lot of channel states.
Du bist der Beste! Danke

klnit, 2022-10-08:
Had to compactdb but now it is running

closed 2022-10-08 by klnit