raspiblitz/raspiblitz #3342: Public IP detection failed

ChuckNorrison, 2022-10-04:
On my Setup with OpenVPN on top of my raspiblitz installation, the public IP detection recognized 127.0.0.1 after some days. This IP is also written to my lnd.conf. 

Iam using a static public IP via VPS iam connecting to over OpenVPN. 

This was solved here:
https://github.com/rootzoll/raspiblitz/issues/1718

rootzoll, 2022-10-06:
Thanks for the report .. such sporadic errors are hard to get by. But will mark this issue for v.1.8.2 where I hope to give the whole IP config situation a fresh look.

ChuckNorrison, 2022-10-09:
After disconnecting the Router and reconnect, the detected IP is 127.0.0.1. 

`internet.sh status global` command result with the expected "globalip". "publicip" and "cleanip" was set to 127.0.0.1.

Output form internet.sh (replaced my correct public IP with 1.2.3.4):

> \# getting public IP from third party service
> \##  curl returned:  1.2.3.4
> \##  curl exit code: 0
> \# sanity check of IP data:
> \# OK IPv4 for 1.2.3.4
> \### LOCAL INTERNET ###
> localip=192.168.8.160
> localiprange=192.168.8.0/24
> dhcp=1
> configWifiExists=0
> network_device=eth0
> network_rx='2.9GiB'
> network_tx='18.3GiB'
> \### GLOBAL INTERNET ###
> ipv6=off
> \# globalip --> ip detected from the outside
> globalip=1.2.3.4
> \# publicip --> may consider the static IP overide by raspiblitz config
> publicip=127.0.0.1
> \# cleanip --> the publicip with no brackets like used on IPv6
> cleanip=127.0.0.1

Raspiblitz.conf publicIP was set to 127.0.0.1.

ChuckNorrison, 2022-10-18:
00infoBlitz.sh calls _cache.sh which will try to `redis-cli get ln_lnd_mainnet_address`. The wrong IP 127.0.0.1 is returned here. 

_background.scan.sh calls `lnd.monitor.sh mainnet info` and retrieves the "uris" via lncli, which is still wrong and sets it to redis db with help of _cache.sh

So the issue is produced by lncli in some way. I observed this behaviour on restarting lnd.service and was able to fix it with temporary with `redis-cli set ln_lnd_mainnet_address xyz@1.2.3.4`

ChuckNorrison, 2022-10-24:
i did a pull request
https://github.com/rootzoll/raspiblitz/pull/3382