raspiblitz/raspiblitz #3343: Electrum Rust Server running but wallet is "not connected"

lupn3, 2022-10-05:
Raspiblitz 1.8.0 running, full synced

Electrs

    is activated
    has opened port 50002
    is connected to localhost

but suddenly out of the nowhere the connection to the wallet isn't possible anymore. It seems that Electrs is receiving the request from the wallet (not: electrum wallet), but not responding to it, even after reinstall an re-indexing Electrs.

rootzoll, 2022-10-06:
I can just mark this for a re-test before v1.8.1 release right now, because this seems hard to reproduce. Uf anybody can also report a similar problem - please report with as much details as possible.

rootzoll, 2022-10-06:
@lupn3 what you can try is to deinstall and then reinstall the service.

lupn3, 2022-10-09:
Thank you for your quick feedback.
Unfortunately, the problem persists, even after deinstall and then reinstall the service. It did not exist before the update to 1.8.0. Also, since the update, neither the ip nor the .onion address is displayed (although the command line sudo cat /mnt/hdd/tor/electrs/hostname helped here). But the problem, that a connection via elelctrs is no longer possible, still exists.

lupn3, 2022-10-14:
Addendum: 
Have completely uninstalled and reinstalled electrs again. The problem still exists. Than I have uninstalled electrs again (by keeping the index) and installed it again. It said "Downloading and building electrs v0.9.7.". 

However, the latest version is v0.9.9. 
Is an update to v0.9.9. recommended? And if so, what is the best way to install it on the raspiblitz?

Nerfzooka, 2022-10-21:
More info from someone with a similar problem:

- Raspiblitz 1.8.0. It was working normally with 1.7.2.
-  My IP and onion address **do** show on my info screen.
- The command` sudo journalctl -n100 -u electrs` shows no errors.
- Electrum wallet 4.3.2 latest release running on Windows 10.
- Electrum wallet will connect to other servers, but not to the Raspiblitz.
- Electrum wallet network screen **is** looking at the Raspiblitz's IP on port 50002.

After uninstalling and reinstalling the electrs service and clearing out the electrum wallet's cert folder It works again!

Nerfzooka, 2022-11-01:
More info from someone with a similar problem:

- Electrum Wallet not synced.
- Raspiblitz stuck on this screen.
[image]

- Reboot doesn't change anything.
- Emptying the wallets cert folder doesn't change anything.

Nerfzooka, 2022-11-04:
Guess it wasn't the problem in question, just the LND crash.  All is working for me after the 1.8.0c update.

poweruser0815, 2022-11-13:
Hi , I do have simelar problem. 

Electrum: Not running - check: sudo journalctl -u electrs
Before I had problems with LND wallet - fixed with LND 0.15.4 - beta

Output of journalctl -u electrs. 

ov 13 01:19:29 raspberrypi electrs[63522]:     0: bitcoind RPC polling failed
Nov 13 01:19:29 raspberrypi electrs[63522]:     1: daemon not available
Nov 13 01:19:29 raspberrypi electrs[63522]:     2: JSON-RPC error: transport error: Didn't receive response data in time, timed out.
Nov 13 01:20:29 raspberrypi electrs[77470]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daem>
Nov 13 01:20:29 raspberrypi electrs[77470]: [2022-11-13T01:20:29.479Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 13 01:20:29 raspberrypi electrs[77470]: [2022-11-13T01:20:29.479Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 13 01:20:29 raspberrypi electrs[77470]: [2022-11-13T01:20:29.526Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 157 SST files, 37.57655091 GB, 4.696187481 Grows
Nov 13 01:20:33 raspberrypi electrs[77470]: [2022-11-13T01:20:33.639Z INFO  electrs::chain] loading 761248 headers, tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f
Nov 13 01:20:37 raspberrypi electrs[77470]: [2022-11-13T01:20:37.528Z INFO  electrs::chain] chain updated: tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f, height=761248
Nov 13 01:21:07 raspberrypi electrs[77470]: [2022-11-13T01:21:07.824Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 13 01:21:07 raspberrypi electrs[77470]: Error: electrs failed
Nov 13 01:21:07 raspberrypi electrs[77470]: Caused by:
Nov 13 01:21:07 raspberrypi electrs[77470]:     0: bitcoind RPC polling failed
Nov 13 01:21:07 raspberrypi electrs[77470]:     1: daemon not available
Nov 13 01:21:07 raspberrypi electrs[77470]:     2: JSON-RPC error: transport error: Didn't receive response data in time, timed out.
Nov 13 01:22:08 raspberrypi electrs[90751]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daem>
Nov 13 01:22:08 raspberrypi electrs[90751]: [2022-11-13T01:22:08.233Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 13 01:22:08 raspberrypi electrs[90751]: [2022-11-13T01:22:08.233Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 13 01:22:08 raspberrypi electrs[90751]: [2022-11-13T01:22:08.281Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 158 SST files, 37.576551714 GB, 4.696187482 Grows
Nov 13 01:22:13 raspberrypi electrs[90751]: [2022-11-13T01:22:13.049Z INFO  electrs::chain] loading 761248 headers, tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f
Nov 13 01:22:16 raspberrypi electrs[90751]: [2022-11-13T01:22:16.807Z INFO  electrs::chain] chain updated: tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f, height=761248
Nov 13 01:22:47 raspberrypi electrs[90751]: [2022-11-13T01:22:47.089Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin

lupn3, 2022-11-14:
Unfortunately, the problem still persists.

The 1.8.0c update has not helped. Also not the manual electrs 0.9.10 update.

Output of sudo journalctl -u electrs:

Nov 14 07:54:36 raspberrypi electrs[12301]: Error: electrs failed
Nov 14 07:54:36 raspberrypi electrs[12301]: Caused by:
Nov 14 07:54:36 raspberrypi electrs[12301]:     0: bitcoind RPC polling failed
Nov 14 07:54:36 raspberrypi electrs[12301]:     1: daemon not available
Nov 14 07:54:36 raspberrypi electrs[12301]:     2: JSON-RPC error: transport error: Didn't receive response data in tim>
Nov 14 07:54:36 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 14 07:54:36 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 14 07:54:36 raspberrypi systemd[1]: electrs.service: Consumed 9.569s CPU time.
Nov 14 07:55:37 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 1.
Nov 14 07:55:37 raspberrypi systemd[1]: Stopped Electrs.
Nov 14 07:55:37 raspberrypi systemd[1]: electrs.service: Consumed 9.569s CPU time.

rootzoll, 2022-11-14:
@lupn3 and @poweruser0815 so it seems the lines
´´´
Nov 13 01:21:07 raspberrypi electrs[77470]: Error: electrs failed
Nov 13 01:21:07 raspberrypi electrs[77470]: Caused by:
Nov 13 01:21:07 raspberrypi electrs[77470]: 0: bitcoind RPC polling failed
Nov 13 01:21:07 raspberrypi electrs[77470]: 1: daemon not available
Nov 13 01:21:07 raspberrypi electrs[77470]: 2: JSON-RPC error: transport error: Didn't receive response data in time, timed out.
´´´

Can you confirm that bitcoind is running .. thats the debug systemd service status on bitcoind?

lupn3, 2022-11-15:
@rootzoll Here is a current excerpt from the debug system status:

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-11-14 07:48:30 GMT; 24h ago
    Process: 7409 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 7430 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 7432 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 7515 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 6h 8min 8.288s
     CGroup: /system.slice/bitcoind.service
             └─7515 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Warning: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Mon 2022-11-14 21:29:44 GMT, ends at Tue 2022-11-15 08:41:59 GMT. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-11-15T07:30:27Z UpdateTip: new best=000000000000000000030bb4dff9c42ec8634d22d859412865214c862d490029 height=763246 version=0x20c00000 log2_work=93.844383 tx=780455902 date='2022-11-15T07:30:15Z' progress=1.000000 cache=130.3MiB(690894txo)
2022-11-15T07:30:27Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2022-11-15T07:40:13Z UpdateTip: new best=00000000000000000005762a48ea7c88e01115a6e7fdf5a501c6f3ee21fee1b4 height=763247 version=0x20600000 log2_work=93.844396 tx=780458572 date='2022-11-15T07:39:55Z' progress=1.000000 cache=131.1MiB(697229txo)
2022-11-15T08:13:12Z UpdateTip: new best=00000000000000000003c0682a573ac2c8659f2950d021ae6310ce192972c84c height=763248 version=0x20200000 log2_work=93.844409 tx=780460120 date='2022-11-15T08:12:51Z' progress=1.000000 cache=131.0MiB(696546txo)
2022-11-15T08:13:26Z UpdateTip: new best=0000000000000000000572c7fc8a83a0815ca756b0b9b359c33505e9d6f90a34 height=763249 version=0x2e470000 log2_work=93.844422 tx=780462193 date='2022-11-15T08:13:00Z' progress=1.000000 cache=131.1MiB(697674txo)
2022-11-15T08:13:26Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2022-11-15T08:13:26Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2022-11-15T08:13:26Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2022-11-15T08:13:26Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2022-11-15T08:14:33Z UpdateTip: new best=000000000000000000004e2e7cf55248b7f8c2a991f86f9f1dd96aa6dd43c753 height=763250 version=0x20a00000 log2_work=93.844435 tx=780464164 date='2022-11-15T08:13:38Z' progress=1.000000 cache=131.5MiB(700785txo)
2022-11-15T08:21:25Z New outbound peer connected: version: 70016, blocks=763250, peer=999 (outbound-full-relay)
2022-11-15T08:24:21Z UpdateTip: new best=000000000000000000046e5d26626a8078c4c18250b7d7a0198b27f3208ed0c7 height=763251 version=0x20200004 log2_work=93.844447 tx=780466129 date='2022-11-15T08:23:21Z' progress=1.000000 cache=132.2MiB(706584txo)
2022-11-15T08:24:59Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-11-15T08:25:58Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-11-15T08:26:19Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-11-15T08:26:41Z UpdateTip: new best=000000000000000000044b601b340b4a30c832a894b9440c50c763fad0077444 height=763252 version=0x21bd4000 log2_work=93.844460 tx=780466566 date='2022-11-15T08:25:28Z' progress=1.000000 cache=132.9MiB(712139txo)
2022-11-15T08:26:50Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-11-15T08:28:23Z UpdateTip: new best=000000000000000000065334b8d84554dafc67641f4bce27699af06baf4796da height=763253 version=0x21574000 log2_work=93.844473 tx=780466954 date='2022-11-15T08:27:13Z' progress=1.000000 cache=133.6MiB(718050txo)
2022-11-15T08:29:34Z New outbound peer connected: version: 70016, blocks=763253, peer=1006 (block-relay-only)
2022-11-15T08:34:37Z UpdateTip: new best=0000000000000000000689680b2577b9dfd8f68d892fcb434e5ebbe13f738ed9 height=763254 version=0x299aa000 log2_work=93.844486 tx=780468069 date='2022-11-15T08:33:31Z' progress=1.000000 cache=134.4MiB(724446txo)

rootzoll, 2022-11-15:
OK bitcoind seems running OK. Can you you check the electrs config with `sudo cat /home/electrs/.electrs/config.toml` - dont post here (because of password) but can you see the line  `auth = "raspibolt:[YOURPASSWORDB]"`?

lupn3, 2022-11-15:
Yes, i can see the line.

rootzoll, 2022-11-15:
Please try to test call the bitcoind RPC from the command line with`curl --user raspibolt --data-binary '{"jsonrpc": "1.0", "id":"curltest", "method": "getblockchaininfo", "params": [] }' -H 'content-type: text/plain;' http://127.0.0.1:8332/`

Use your passwordB when asked for `host password`. You should get something back like this when RPC is OK:
```
{"result":{"chain":"main","blocks":763280,"headers":763280,"bestblockhash":"000000000000000000053610c7f187e1ccf2c700e2246636bf4d98e4cb20d293","difficulty":36762198818467.21,"time":1668517470,"mediantime":1668513729,"verificationprogress":0.9999996407126942,"initialblockdownload":false,"chainwork":"00000000000000000000000000000000000000003979275cb6232949708878e4","size_on_disk":497647684564,"pruned":false,"warnings":""},"error":null,"id":"curltest"}
```

lupn3, 2022-11-15:
According to this output RPC seems OK.

rootzoll, 2022-11-15:
So to sum up so far: Bitcoin RPC is alive and answering, ElectRS has the correct username/password in config BUT fails because RPC times out. @openoms have you an idea what else to check?

poweruser0815, 2022-11-18:
@rootzoll sorry for my delayed reply. Same here..

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-11-13 19:50:23 GMT; 5 days ago
    Process: 7805 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 7821 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 7826 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 7911 (bitcoind)
      Tasks: 18 (limit: 9353)
        CPU: 11h 32min 55.638s
     CGroup: /system.slice/bitcoind.service
             └─7911 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -…

poweruser0815, 2022-11-18:
> Yes, i can see the line.
Same here

poweruser0815, 2022-11-18:
> According to this output RPC seems OK.

Here also okay

lupn3, 2022-11-22:
@rootzoll Maybe there is something wrong with my Bitcoin Core, or the connection Electrs has to it? (https://github.com/sparrowwallet/sparrow/issues/731)

If so, how could this be verified?

rootzoll, 2022-11-22:
There is a new electrs version out 0.9.10 - some people report it fixes their problems. It will be part of the next RaspiBlitz release v1.9.0 .. so maybe wait some weeks and give it a try then.

lupn3, 2022-11-24:
@rootzoll Yes, then I will wait and see. Low time preference here anyway ;) Regardless of this, wouldn't Fulcrum actually be a better alternative for performance reasons alone?

Btw, thanks a lot for your time and effort on this issue here and the raspiblitz in general!