raspiblitz/raspiblitz #3344: v1.8 Installing Lightning Terminal results in 502 Bad Gateway for RTL

skifreak27, 2022-10-05:
Seen this issue posted elsewhere with different revs but have not discovered a resolution.  Get bad gateway 502 error when trying to log into web GUI for RTL on raspiblitz.  LIT GUI works fine. 

Have tried removing and re-installing RTL multiple times.  Have tried rebooting multiple times. 

When entering 
`sudo service RTL status`

I receive the following error/failure; 

[image]

Seems that RTL is not installing or running correctly.

Have also tried service start and stop to no avail.  Any help would be super appreciated.

skifreak27, 2022-10-30:
Is this just a dead project or what?

jimmymsaliba, 2022-11-04:
I'm having a similar issue.  Can access LIT Gui but not the Raspiblitz UI and RTL.

jimmymsaliba, 2022-11-04:
I ended up fixing mine and it looks like it was related to installing JoinMarket instead of LIT.  After checking the /var/log/nginx/error.log I was getting errors related to JM.  Uninstalled JM and removed the related .conf files in /etc/nginx/sites-enabled/ directory.  Nginx started just fine after that.  Hope this helps!

rootzoll, 2022-11-05:
Will recheck for v1.8.1 release

rootzoll, 2022-12-11:
OK I was nozt able to reproduce this. I installed RTL (for lnd) and then installed LIT and JoinMarket (also JAM) and RTL still works. Maybe this got fixed already. Please this on your side as soon a Release Candidate of v1.9 is avalilable and report back.