raspiblitz/raspiblitz #3345: Problem install bonus.mempool.sh

BlockHackTk, 2022-10-05:
Hi,

I tried to install _mempool_ via the menu but got an error advising me to restart Rraspiblitz and do a manual installation with `./bonus.mempool.sh.`

Raspiblitz V 1.8.0

Here is the result:

`admin@192.168.1.103:~/config.scripts ₿ ./bonus.mempool.sh on
# making sure services are not running
# *** INSTALL MEMPOOL ***
nodeJS v16.14.2 is already installed
# txindex is already active
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
mariadb-client is already the newest version (1:10.5.15-0+deb11u1).
mariadb-server is already the newest version (1:10.5.15-0+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
adduser: The user `mempool' already exists.
fatal: destination path 'mempool' already exists and is not an empty directory.
HEAD is now at f901f069 Release v2.4.0
--2022-10-05 17:42:37--  https://github.com/wiz.gpg
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4608 (4.5K) [text/plain]
Saving to: ‘/var/cache/raspiblitz/pgp_keys_wiz.asc’

/var/cache/raspiblitz/pgp_keys_wiz.asc       100%[==============================================================================================>]   4.50K  --.-KB/s    in 0s      

2022-10-05 17:42:38 (19.4 MB/s) - ‘/var/cache/raspiblitz/pgp_keys_wiz.asc’ saved [4608/4608]

pub   rsa4096 2019-01-14 [SC]
      913C5FF1F579B66CA10378DBA394E332255A6173
uid                      J. Maurice (wiz) <j@wiz.biz>
sub   rsa4096 2019-01-14 [A]
sub   rsa4096 2019-01-14 [E]

gpg: key A394E332255A6173: "J. Maurice (wiz) <j@wiz.biz>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
# running: git verify-commit f901f069

gpg: Signature made Thu 09 Jun 2022 15:10:18 BST
gpg:                using RSA key 913C5FF1F579B66CA10378DBA394E332255A6173
gpg: Good signature from "J. Maurice (wiz) <j@wiz.biz>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 913C 5FF1 F579 B66C A103  78DB A394 E332 255A 6173
# goodSignature(1)
# correctKey(1)

##########################################################################
# OK --> the PGP signature of the checked out f901f069 commit is correct
##########################################################################

# npm install for mempool explorer (frontend)

changed 1 package, and audited 1369 packages in 21s

68 packages are looking for funding
  run `npm fund` for details

6 vulnerabilities (2 moderate, 3 high, 1 critical)

To address issues that do not require attention, run:
  npm audit fix

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.

> mempool-frontend@2.4.0 build
> npm run generate-config && ng build --configuration production --localize && npm run sync-assets && npm run build-mempool.js

> mempool-frontend@2.4.0 generate-config
> node generate-config.js

mempool-frontend-config.json file not found, using default config
Copied src/index.mempool.html to src/index.html
mempool version 2.4.0
mempool revision f901f069
No configuration updates, skipping generated-config.js file update
Unknown error: Error: Cannot find module '/home/mempool/mempool/frontend/node_modules/iconv-lite/lib/index.js'. Please verify that the package.json has a valid "main" entry
FAIL - npm run build did not run correctly, aborting (1)`

ChuckNorrison, 2022-10-06:
Can confirm this install process fails

> \# npm install for mempool explorer (frontend)
> npm ERR! code ERR_SOCKET_TIMEOUT
> npm ERR! network Socket timeout
> npm ERR! network This is a problem related to network connectivity.
> npm ERR! network In most cases you are behind a proxy or have bad network settings.
> npm ERR! network 
> npm ERR! network If you are behind a proxy, please make sure that the
> npm ERR! network 'proxy' config is set properly.  See: 'npm help config'
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/mempool/.npm/_logs/2022-10-06T08_46_09_230Z-debug-0.log
> FAIL - npm install did not run correctly, aborting

rootzoll, 2022-10-06:
I marked this for re-test for v1.8.1 release ... but so far I am not able to reproduce this on dev branch. Would need to test with a fresh v1.8.0 sd card.

ChuckNorrison, 2022-10-10:
reflashed sd card with v1.8.0 and install of mempool works

BlockHackTk, 2022-10-18:
All is good for me with **Version 1.8.0b** with lnd 0.15.2 & Core Lightning 0.11.2 and bitcoin 23.0.0

BlockHackTk, 2022-10-18:
if the installation of bonus.mempool.sh does not work with **Version 1.8.0**, install **Version 1.8.0b** and it should be ok.

closed 2022-10-18 by BlockHackTk

7o7yee, 2023-02-12:
hi, i am using raspiblitz 1.8.0c with bluemoon, bitcoin 23.0.0 and CLN_GRPC v0.11.2 on a rpi4b and i just finished downloading rtl but i cant download mempool it just says "installing" all day and idk what to think do i just leave it on until it finishes? it could be days? how long should it take? rtl was quick...