raspiblitz/raspiblitz #3350: LND suddenly inactive

bigsome, 2022-10-08:
Hi there,

I don't know what happens with my node... It was running without issues for months, in august I had a power outage, then it got stuck syncing but at 100%... tried to repair but finally flashed a new SD card and recover it. After some weeks, the same process (disconnected, stuck and recover with a fresh new SD card)... and now went out and stuck again.

I don't want to recover it and have the same issue in some weeks. And I don't know where is the problem... That's why I need some help.

This is the debug output: https://l.termbin.com/5hqm

I see some errors connecting to bitcoin network:
*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-10-08T18:35:30Z UpdateTip: new best=000000000000000000014573a7ba1754217e75973af9975cb14fd3c98dbae0b1 height=757742 version=0x20000004 log2_work=93.773429 tx=770674183 date='2022-10-08T18:35:20Z' progress=1.000000 cache=68.5MiB(464939txo)
2022-10-08T18:36:38Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-10-08T18:39:59Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-10-08T18:46:19Z UpdateTip: new best=00000000000000000003d3862030faf89d0f2ceeb6f14621f571436a9ad7f727 height=757743 version=0x20a00000 log2_work=93.773440 tx=770677780 date='2022-10-08T18:46:03Z' progress=1.000000 cache=69.6MiB(473867txo)
2022-10-08T18:50:08Z New outbound peer connected: version: 70016, blocks=757743, peer=19992 (block-relay-only)
2022-10-08T18:50:34Z UpdateTip: new best=000000000000000000083f3909ae84d852ca623fae6d1ce2e09ba41c31548923 height=757744 version=0x20c00000 log2_work=93.773452 tx=770678778 date='2022-10-08T18:50:18Z' progress=1.000000 cache=69.7MiB(474823txo)
2022-10-08T18:54:31Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-10-08T18:57:28Z UpdateTip: new best=000000000000000000012666a496608e688f2758385626e6098929fe765c25d0 height=757745 version=0x20004000 log2_work=93.773463 tx=770681194 date='2022-10-08T18:57:05Z' progress=1.000000 cache=70.3MiB(479306txo)
2022-10-08T19:11:08Z UpdateTip: new best=00000000000000000001fb6490327da79c11590f7b31cb8687fa0e3f6b1b49b5 height=757746 version=0x27ffe004 log2_work=93.773475 tx=770683941 date='2022-10-08T19:10:14Z' progress=1.000000 cache=71.4MiB(488337txo)
2022-10-08T19:11:44Z UpdateTip: new best=0000000000000000000354d80a5f629961bfb885fd0d850427b24025509b4ceb height=757747 version=0x20600000 log2_work=93.773486 tx=770684906 date='2022-10-08T19:11:07Z' progress=1.000000 cache=70.8MiB(483583txo)
2022-10-08T19:16:02Z UpdateTip: new best=000000000000000000006d65bb87d08492a6d9e0b83784667353ee2bb81d10ce height=757748 version=0x216c2000 log2_work=93.773497 tx=770686446 date='2022-10-08T19:15:47Z' progress=1.000000 cache=70.9MiB(483871txo)
2022-10-08T19:18:02Z UpdateTip: new best=000000000000000000087cf80c43322b4e11060fc2c1380f33836501a3e7ca74 height=757749 version=0x20800000 log2_work=93.773509 tx=770687038 date='2022-10-08T19:17:50Z' progress=1.000000 cache=71.0MiB(484623txo)
2022-10-08T19:26:08Z UpdateTip: new best=00000000000000000003310c76ce02fd6a082748809edd6959cb0ef868328a41 height=757750 version=0x20800000 log2_work=93.773520 tx=770688674 date='2022-10-08T19:25:32Z' progress=1.000000 cache=71.5MiB(488811txo)
2022-10-08T19:41:12Z UpdateTip: new best=000000000000000000086252fbc68b6f207f3da470f1d9dc9b6aa5b82a708d16 height=757751 version=0x2000c000 log2_work=93.773532 tx=770691595 date='2022-10-08T19:40:43Z' progress=1.000000 cache=72.5MiB(497432txo)
2022-10-08T19:45:37Z UpdateTip: new best=0000000000000000000620ebe7e1f82eeff8ebf085f80ac036e76016d84f567a height=757752 version=0x20eb2000 log2_work=93.773543 tx=770692550 date='2022-10-08T19:45:13Z' progress=1.000000 cache=72.9MiB(500194txo)
2022-10-08T19:46:24Z UpdateTip: new best=000000000000000000032c781dbe11459fba50312acfca3cd96fa2bc4367d5b1 height=757753 version=0x20c00004 log2_work=93.773555 tx=770692631 date='2022-10-08T19:45:59Z' progress=1.000000 cache=72.9MiB(500598txo)
2022-10-08T20:03:24Z UpdateTip: new best=000000000000000000050289600504107f55a540b141a21ef810baa901c63adb height=757754 version=0x20000000 log2_work=93.773566 tx=770695169 date='2022-10-08T20:02:58Z' progress=1.000000 cache=73.9MiB(508301txo)
2022-10-08T20:16:55Z UpdateTip: new best=0000000000000000000226a466efd188c0ad38abe51da4da566899278e9226b9 height=757755 version=0x20000000 log2_work=93.773578 tx=770697243 date='2022-10-08T20:16:45Z' progress=1.000000 cache=74.4MiB(512657txo)
2022-10-08T20:20:56Z New outbound peer connected: version: 70016, blocks=757755, peer=20038 (outbound-full-relay)
2022-10-08T20:22:17Z UpdateTip: new best=00000000000000000008398f910e7323b942e29f8a906b516479d3d3f3074bd9 height=757756 version=0x20002000 log2_work=93.773589 tx=770699288 date='2022-10-08T20:21:38Z' progress=1.000000 cache=74.6MiB(514211txo)

And that LND is inactive since this wednesday...
*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Wed 2022-10-05 14:02:14 BST; 3 days ago
    Process: 2692606 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 2692855 ExecStart=/usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf (code=exited, status=0/SUCCESS)
   Main PID: 2692855 (code=exited, status=0/SUCCESS)
     Status: "Wallet unlocked"
        CPU: 3d 7min 8.691s

Obviously, LND is inactive, but why? how can I avoid this happening again?

[image]

[image]

Thanks in advance!!

ZeroPointEnergy, 2022-10-10:
There is a bug in a library LND uses that fails to parse a taproot transaction:

https://github.com/lightningnetwork/lnd/issues/7002
https://twitter.com/brqgoo/status/1579216353780957185

There is an LND update available with a fix for this issue 0.15.2

I did the following to get LND running again from the console:
```
sudo systemctl stop lnd
./00mainMenu.sh
```
In the menu, I selected Update -> LND, selected the 0.15.2 release and pressed OK. Raspiblitz updated the LND and rebooted automatically. Now it's working again.

bigsome, 2022-10-10:
Thanks @ZeroPointEnergy I updated yesterday and it's working fine... didn't mention here because I wanted to check if it worked fine for a few days. Now, with your comment I think the problem is issued correctly and fixed with the update. Thanks!

bigsome, 2022-11-10:
Yes! Thought it was closed… Thanks!

closed 2022-11-10 by bigsome