raspiblitz/raspiblitz #3351: Is it possible to add a VPN client to the raspiblitz firmware?

tomm2001, 2022-10-09:
Hello, TOR is blocked in Russia, can you create a VPN server outside of Russia, and install a VPN client on raspiblitz to bypass the TOR blocking? Otherwise, even the blockchain has not been synchronized for six months. I asked people in our chat (Raspiblitz_RU) in a telegram, they say that this is happening because of DPI and we need to pack everything into the VPN tunnel. Can you help with solving this problem?

ChuckNorrison, 2022-10-09:
Iam using OpenVPN to a VPS as OpenVPN Server
 
> \# install openvpn and setup client
> sudo apt install openvpn
> sudo cp /mnt/hdd/yourclient.ovpn /etc/openvpn/client.conf
> sudo systemctl start openvpn@client
> sudo systemctl enable openvpn@client
> 
> \# set firewall
> \# need to check own public IP (UPDATE  EXTERNAL IP HERE replace 1.2.3.4)
> sudo ufw allow out on eth0 to 1.2.3.4/32
> 
> \# allow OpenVPN Client connection
> sudo ufw allow out 1194/udp
> 
> \# allow traffic via openvpn tunnel only
> sudo ufw allow out on tun0 from any to any
> sudo ufw allow in on tun0 from any to any

And a script to check connectivity

> \#!/bin/sh
> ping -q -c 4 10.8.0.1 > /dev/null
> if [ $? -eq 0 ]; then
>   echo "`date`: Connectivity check successful"
> else
>   echo "`date`: Connectivity check failed"
>   dhclient -r; dhclient > /dev/null
>   sleep 15
>   systemctl restart openvpn@client
>   sleep 15
>   systemctl restart nginx
>   echo "`date`: Connectivity repaired... hopefully"
>   exit
> fi

ChuckNorrison, 2022-11-22:
i can share my custom-install scripts to setup and connect openvpn and add a network check routine via cron
https://github.com/ChuckNorrison/raspiblitz-scripts

tomm2001, 2023-01-15:
an open VPN will not cope, because VPN traffic is detected by the provider and blocked in Russia. We need a VPN that could mask VPN traffic, accessing the server and establishing a connection should take place using an unencrypted protocol, after that the server should receive a certain command to open a certain port and only after that the VPN should be turned on. There is a description of the problem and ways to solve it, can you do as recommended in the new raspiblitz build? I installed version 1.9, but it also cannot establish a connection with knives due to VPN blocking. https://habr.com/ru/post/415977/