raspiblitz/raspiblitz #3352: rescan failed, node wont boot

bitsam, 2022-10-10:
I noticed that my node hung and rebooted it, It is now showing the following two lines in  /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log ,  and won't progress any futher.

2022-10-10 08:16:51.468 [INF] LNWL: Started rescan from block 00000000000000000007477b90ff0ea53cdd1db88e03799af18ff58df0cebaa7 (height 757921) for 213 addresses

2022-10-10 08:16:51.963 [ERR] LNWL: Unable to complete chain rescan: readScript: script witness item is larger than the max allowed size [count 33970, max 11000]

bitsam, 2022-10-10:
seems related to issue #3350 , followed instructions and node back online.

closed 2022-10-10 by bitsam

rootzoll, 2022-10-10:
This seems reolated to the LND bug: https://twitter.com/roasbeef/status/1579274303006474242?s=20&t=G2Lh9oxf4iO5K-9CGgmAkQ

On the LCD of the Raspiblitz having this buck you can see "Scanning blocks 100.00" with an lnd version lower than 0.15.2

To hotfix this pllease try: CTRL+c to exit to terminal (if needed). Call command `menu` and then select: UPADTE > LND > RECKLESS LND update to v0.15.2-beta & let it reboot