raspiblitz/raspiblitz #3358: Node status stuck at 100% progress. Not moving to menu screen. Info screen loop error

dannuttall, 2022-10-10:
After syncing the blockchain, my node is stuck displaying 100%. It's been in this state for 24 hours now. 
[image]

It looks the same if I SSH in. If I CTRL+C and enter the menu manually, then look at the status screen, I get the following error:
[image]

I installed the RTL web interface and that says the node is fine. 

The above 2 things are slightly annoying but if my node is running fine I'm not too bothered, but if there's something I could try to fix it then I'd love to know!

Here's my log:
https://l.termbin.com/hat5c

NilByte, 2022-10-10:
Check this:

https://github.com/rootzoll/raspiblitz/issues/3352

dannuttall, 2022-10-10:
@NilByte Thanks updating LDN seemed to resolve the stuck on 100% screen. I still get the info screen update loop error. Is that a problem?

NilByte, 2022-10-10:
If you followed the steps:

Steps:
- connect with ssh 
- press CTRL+C
- type: menu
- select UPDATE - LND 
- update to LND v0.15.2-beta

And Raspiblitz rebooted now showing your channels active in status screen, then update should have been successful.

What is the info screen update loop error you notice?

dannuttall, 2022-10-10:
@NilByte yes I followed those steps and the status screen is now showing channels, although 0/0, not sure if that's a problem?

The update loop error is this:

[image]

NilByte, 2022-10-10:
`Screen is updating in a loop .... press 'x' now to get back to menu.`

Is normal behavior when you access the status screen via ssh. 

If you haven't opened any channels yet, then `0/0 Channels 0 sat 3 peers` is normal status.

closed 2022-10-10 by dannuttall

Delivator, 2023-04-12:
I have the same issue but I'm on `RaspiBlitz 1.8.0` with `lnd version 0.16.0-beta commit=v0.16.0-beta`

Anyone know if this is a known issue? Should I open a new issue?

Delivator, 2023-04-12:
My screen looks like this
[image]
[image]
and this is my debug log: https://l.termbin.com/e90e

Delivator, 2023-04-14:
okay looks like it just took a very long time even though it said 100%