raspiblitz/raspiblitz #3363: RTL unreachable after updating to 1.8

SomeBTChomer, 2022-10-12:
RTL isn't reachable on ports 3000/3001 or the provided tor address after flashing to 1.8, opening the onion link shows an nginx error.

Uninstalling and reinstalling via the mainmenu and the webUI also didn't help.

Surenic, 2022-10-13:
I have had a problem few days ago, that RTL (neither LND, nor CLN) wasn't running, because clrest was not able to start. Are you using CLN as lightning implementation, too?

Then try this, worked for me

https://github.com/rootzoll/raspiblitz/issues/3324#issuecomment-1266869029

SomeBTChomer, 2022-10-13:
Thanks, but no CLN here, only lnd is active.

rootzoll, 2022-11-15:
Did it repair with the the new v1.8.0c sd card image update?

cstenglein, 2023-06-07:
@SomeBTChomer does it still appear with 1.9.0?

Closing for now, if it still appears, please re-open

closed 2023-06-07 by cstenglein