raspiblitz/raspiblitz #3366: How do I restore default settings/ app selection?

1thales, 2022-10-13:
I cant reactivate WebUI and generally noticed it is not a good idea to decativate any app since Raspiblitz is not well tested for this. How do I restore default settings/ app selection? would really appreciate any help

cstenglein, 2022-10-16:
Do you have more info on this?
What does "reactivate WebUI" mean? What errors do you get?

1thales, 2022-10-19:
I deactivated WebUI then reactivated it from the menu. I get "unknown error" in the UI

1thales, 2022-10-19:
I would love to be able to go back to "factory" settings

rootzoll, 2022-11-15:
running a recover with a fresh sd card is thge normal way to get back to "factory" settings

1thales, 2022-11-15:
Its not. There is some conf file in the hard drive that is making blitz restore its state

rootzoll, 2022-11-15:
In the process of starting with a fresh sd card image it will ask you if you like to start the recovery - but you can cancel that and then you will get offered the "Fresh Setup" option.