raspiblitz/raspiblitz #3367: Lighting sync stuck at 99.99%

Haightz, 2022-10-14:
Hello
Blockchain is 100% but lighting sync has been stuck at 99.99% for a few days now. I have restarted the node twice and still the same issue. This is not an initial setup. 

/home/admin/XXdebugLogs.sh | nc termbin.com 9999

BCsRock, 2022-10-14:
It's a recent bug in LND implementation which cannot process one particular taproot transaction. Just update the Image to version 1.0.8b and it will work. 
See https://github.com/rootzoll/raspiblitz/commit/ace3e21a9c18bee8a5cce7d43c286036c9a847c7

Haightz, 2022-10-14:
> It's a recent bug in LND implementation which cannot process one particular taproot transaction. Just update the Image to version 1.0.8b and it will work. See [ace3e21](https://github.com/rootzoll/raspiblitz/commit/ace3e21a9c18bee8a5cce7d43c286036c9a847c7)

Thank you back up running!

closed 2022-10-14 by Haightz