raspiblitz/raspiblitz #3368: Lightning wallet recovery

jerbearstare, 2022-10-15:
### *BASIC ISSUE*

I misplaced my password D, but still have my recovery seed. Will I be able to retrieve my BTC if something happens to my node, without password D?

If so how do I create a new password D / reset my lightning node? So I can load a good amount of sats to get my channels going

openoms, 2022-10-28:
you can change the password D in the menu. Do it now so you won't have issues later.

jerbearstare, 2022-10-29:
in  "*set password*" within the menu there is only password **A, B &C**

is the password **D** a figment of my imagination?

openoms, 2022-10-29:
You are right, actually the password D is not used for a good while now. 
It was how we called the optional passphrase with the seed. 
If you generated your wallet from the menu recently there was no passwordD given.

closed 2022-11-15 by rootzoll