raspiblitz/raspiblitz #3369: Give root password for maintenance or press Control-D to continue

mattias232, 2022-10-16:
Hi, i have tried to update my node and get stuck in this above loop. I now cannot get out of this mode.

I have entered root pw via keyboard but no matter what i enter to restart from maintenance mode, it wont come out of it.

Any Advice

Many thanks
PS i am not a developer lol

ChuckNorrison, 2022-10-17:
Signals a corrupted state of your node. Could happen after power outtage or hardware related issue. Best advice is to reflash your sd card

https://github.com/rootzoll/raspiblitz/blob/v1.8/README.md#write-the-sd-card-image-to-your-sd-card

mattias232, 2022-10-20:
You guys rock and are ninja AF - that works and is all back working again.Thanks so much!

closed 2022-10-20 by mattias232