raspiblitz/raspiblitz #3371: CLN Does not have export option in connect menu for macaroons & TLS.cert

benjamin-wilson, 2022-10-17:
Currently LND has an 'Export' option under CONNECT > EXPORT to export macaroons & TLS.cert. When using CLN this option is not available.

openoms, 2022-10-28:
CLN works differently. You might want to refer to the commando plugin documentation.

benjamin-wilson, 2022-10-29:
When connecting Zeus to my node it displays a macaroon. It also requests a certificate for verification.

openoms, 2022-10-29:
Yes, that is for C-LightningREST
get it again from CONNECT - MOBILE - CLREST

benjamin-wilson, 2022-11-15:
Where can I get the TLS cert?

openoms, 2022-11-15:
> Where can I get the TLS cert?

it is shown as a QR code when you go through. CONNECT as well but you don't really need it.

benjamin-wilson, 2022-11-15:
I don't see a QR for it and it isn't mentioned anywhere either. (Going though ZEUS_CLREST option)

benjamin-wilson, 2023-02-04:
Revisiting this issue, there is no option to export the TLS cert to prevent MITM attacks.

openoms, 2023-02-04:
For mainnet the file is under:
```
/home/bitcoin/c-lightning-REST/bitcoin/certs
```
extract it as a string:
```
cert=$(grep -v 'CERTIFICATE' /home/bitcoin/c-lightning-REST/bitcoin/certs/certificate.pem | tr -d '=' | tr '/+' '_-' | tr -d '\n')
echo ${cert}

generate a QRcode with the shortcut in the admin user:
```
qr ${cert}
```

Related docs: https://github.com/Ride-The-Lightning/c-lightning-REST#security

Curious are you planning to use it with Zeus (import to Android?) or please describe the use case.

benjamin-wilson, 2023-02-04:
Yes I am planning on importing it to Android for Zeus. When you scan the QR in Zeus it warns you to do this to prevent MITM. If I am not mistaken there were instructions in raspiblitz how to do this when I was using LND but not CLN. Thank you for the docs link.

openoms, 2023-02-04:
It needs to be in DER format for Android to read the .crt file:
```
# convert to DER format
sudo -u bitcoin openssl x509 -outform der -in  /home/bitcoin/c-lightning-REST/bitcoin/certs/certificate.pem -out  /home/bitcoin/c-lightning-REST/bitcoin/certs/certificate.crt
```

Did not find a good way yet to transfer the file remotely. The QRcode does not work with binary and Tor is too broken for onionshare.