raspiblitz/raspiblitz #3374: LND does not start

AnimaI, 2022-10-17:
Hi folks,

Raspiblitz does not start anymore.
Can someone help me with this?

[](https://l.termbin.com/5o5g)

thank you for any advice :-)

Tetrix42, 2022-10-24:
don't undestand exactly what your problem is, but I suggest you do an upgrade.
You are running lnd version 0.14.2-beta on raspiblitz 1.7.2.
There is raspiblitz 1.8.0  with lnd 0.15.0. But it requires lnd to update to 0.15.2 to fix a bug on processing Taproot transactions. (how to upgrade to 0.15.2 see https://github.com/rootzoll/raspiblitz/issues/3350)

AnimaI, 2022-10-26:
@Tetrix42 Thanks for your information. 

Before I update, I want to disable the automatic recoveryscan.

The problem is that every time I reboot it runs the recoveryscan.

How can I stop the recoveryscan.

have found information here:

https://github.com/rootzoll/raspiblitz/issues/2901

I am still unsure.

AnimaI, 2022-10-26:
lnd.monitor.sh mainnet info

#scanTimestamp=1666780136
ln_lnd_address=''
ln_lnd_tor='0'
ln_lnd_sync_chain='0'
ln_lnd_sync_progress='100.00'
ln_lnd_sync_graph='0'
ln_lnd_channels_pending='0'
ln_lnd_channels_active='0'
ln_lnd_channels_inactive='19'
ln_lnd_channels_total='19'
ln_lnd_peers='0'
ln_lnd_recovery_mode='1'
ln_lnd_recovery_done='0'
`

closed 2022-11-03 by AnimaI