raspiblitz/raspiblitz #3378: Ledger hww login to raspiblitz not working after v1.8.0

chrisBonGit, 2022-10-21:
Hi, before updating to v.1.8.0 I had my raspiblitz set up so that log in was via ssh with my key provided/confirmed through an ssh app on my hww (ledger nano). After the upgrade I repeated the steps I took when setting up this secure access process, earlier this year - but it doesn't seem to work. Specifically, after removing the password option, I'm given the error error ‘admin@192.168.1.2: Permission denied (publickey). Steps taken detailed below.

System: 8GB raspberry pi. SW v1.8.0.
HWW: Ledger Nano X (also same issue with S)
home computer: imac mini, ssh opened through a mac terminal

Steps used to set up ssh key login provided from ledger hww.
- generate public key with ledger-agent 
- add this to ~/.ssh/authorized_keys, making sure permissions for both .ssh and authorized_keys were correct, 
- then remove ability to use password in /etc/ssh/ssh_config and sshd_config 
- restart both services.
- attempt to ssh in to LN from a new terminal -> error   ‘admin@192.168.1.2: Permission denied (publickey)
When connecting with ssh, it prompts the ledger for authorisation (good) but I don’t get any request for authorisation to log into the LN itself (i.e. before update I would need to approve twice on my ledger, now I am only asked for one approval and am then give then error ‘admin@192.168.1.2: Permission denied (publickey).). 
I kept access to the LN thorugh a few ‘back up’ terminals / consoles so now I’ve rolled back to using pwdA log in. So I’m not in trouble but I would like to harden access to the LN. Have I missed a step, or has there has been a change in how I need to configure this (e.g. is pasting the public key into authorized_keys now not the right place to put it?). I'm still in the 'learning by doing' phase, so it is possible I've missed something obvious to someone more familiar with ssh config & management.

Nerfzooka, 2022-11-01:
You got a ledger to work as an ssh key?  That's tremendous! If this issue gets fixed I would also like to also set that up.  I'm "learning by doing" with a trezor to ssh to my linux box.  but so far I'm afraid to turn off the password option.
Slight change of topic: How did you get your ledger to become a security key?  I thought I saw it in the ledger app list a while back, but now I can't find it.

chrisBonGit, 2022-11-04:
Thanks - I leant heavily on a fusion of the following guides (although there was still, as far as I recall, some trial and error - I took unstructured notes as I attempted various modifications, & can attempt to tidy up if that's of interest)
- https://thoughts.t37.net/a-step-by-step-guide-to-securing-your-ssh-keys-with-the-ledger-nano-s-92e58c64a005
- https://www.digitalocean.com/community/tutorials/how-to-configure-ssh-key-based-authentication-on-a-linux-server
- https://blog.ledger.com/ssh/
If you can afford it / have a trusted friend with another hww, I'd definitely recommend setting up an alternative hww with access before you turn off the pwd (and probably rehearse the full lightning node recovery option, if you've not already). Ledger's app is called 'SSH/PGP Agent' (at least, that's the display name on the device).

rootzoll, 2022-11-05:
@chrisBonGit this is a cool use of a hardware wallet. Its a bit special so I will try to take a quick look into it for the v1.8.1 release. Basically not much changed from v1.72 to v1.8.0 on the SSH part.

rootzoll, 2022-11-27:
I ran into a setup problem within my setup and missing a bit the time to go deeper on it. Pushing this issue a bit up the timeline - but happy if someone else can help out (test) in the meantime.

chrisBonGit, 2022-11-29:
I'm not a SW dev, but I'm happy to learn, I work more on algo design and test of returned SW. If it could help, I'm happy to flash a SD card with the test-image & try it out on  my dev/learning-raspiblitz - I don't mind investing a bit of time to see if I can get hww back in operation. I am travelling for work for a few days, but will have a go this weekend - if you confirm that it could be helpful. Also, github noobie question, which branch should I take?

rootzoll, 2022-12-07:
Any practical report on modifying your RaspiBlitz to being able to use the Ledger "SSH/PGP Agent" to SSH login would help.

chrisBonGit, 2023-03-03:
Sorry for taking so long to respond. Below are the notes I made when I was able to successfully get this to work (I think this was sometime in early 2021, say Feb / March). I was not able to get this working after the upgrade to v1.8.0, I will try again if I can make some time.

**Security Steps for RaspiBlitz**
Lock down access to LNDs within Home Network
	• Use switch to isolate LNDs from most of home, [Managed Switch ports] 
Set up SSH key access (ledger agent)
	• Using https://thoughts.t37.net/a-step-by-step-guide-to-securing-your-ssh-keys-with-the-ledger-nano-s-92e58c64a005 (also, but less clear help here https://blog.ledger.com/ssh/)
	• Once ssh app on HWW & ledger agent is on computer, then the enabling tech is to have python running in terminal and use
		○ ledger-agent username@hostname
		○ Create a directory (as admin) on raspiblitz /~/.ssh
		○ Copy public key into ~/.ssh/authorized_keys on remote server (LNDXX). NOTE must include ecdsa-sha2-nistp256 at start.
		○ Use chmod to set permissions of .ssh folder to 700 (drwx) :q& authorised keys to 600
		○ Restart ssh service: 
			§ service sshd restart
		○ NOT REQUIRED. [Add public key locally (ie. On mac / linux machine) in ~/.ssh/ledger.conf] That was my misunderstanding. Doing this forces use of ssh key to start the ssh session, then another confirmation from node to log in…
	• Repeat above for redundant hww
	• Then, with one terminal already open and connected to node…
		○ Remove password as an option
			1. edit /etc/ssh/ssh_config, you need to do this as a super user (e.g. sudo vi …)
				1. uncomment the line #PasswordAuthentication no (line 25)
				2. Change no to yes above
			2. Restart ssh server 
				1. service ssh restart
			3. Test log in from a new terminal. KEEP safety line (i.e. terminal that is already connected to node) attached.

Remove Password access
	1) On raspiblitz, edit /etc/ssh/ssh_config, you need to do this as a super user (e.g. sudo vi …)
		a. uncomment the line 25, #PasswordAuthentication no
		b. Change no to yes above
	2) Restart ssh server 
		a. service ssh restart
	3) Test log in from a new terminal
	4) Note, some refs state that you also need to check sshd_config (I found lines commented out in my raspiblitz install)
		a. PAM = privileged access management.
		b. In sshd_config
			i. Uncomment & Modified line 56; PassWordAuthentication from yes to no
			ii. :Did not need to modify line 61 (ChallengeResponseAuthentication was already set to no)
			iii. Modified line 84; UsePAM yes to UsePAM no
Steps in 2 successfully disables password (after a service ssh restart)

Nerfzooka, 2023-03-03:
Thank you so much @chrisBonGit  I've added your notes to [my repository](https://github.com/Nerfzooka/Using-Hardware-Wallets-as-Security-Keys) of DIY security key stuff.  If anyone has an issue or question on that topic, you can add an issue over there.