raspiblitz/raspiblitz #3390: Joinmarket : wallet recovery from seed

HonmaSokyu, 2022-10-27:
Hello guys need some help. I was running JM and yieldgenerator on my raspberry with raspiblitz like a year ago. Then my memmory card broke and i lost my wallet.jmdat. I still have my 12 word seed and password.

Now trying to recover from mnemonic. All goes well and "recovered wallet ok".
When i go check the balance its zero, nothing is there. I tried increasing the Gap limit to 1000 and 5000 and 10000 and after each i rescanned the blockcahin and still zero balance.

Then i tried the "No-history wallet synchronization" and using large gaps but still zero balance.

What am i doing wrong and how can i get my bitcoins back or are they lost forever? 😢

I am wiling to give some reward if the problem is fixed...

Thanks for helping!

ChuckNorrison, 2022-10-27:
The wallet.dat should be found in `/mnt/hdd/app-data/.joinmarket`. Did you checked this path?

HonmaSokyu, 2022-10-28:
Just checked that and its just this new.jmdat which i recovered with seed.

openoms, 2022-10-28:
you coins are not lost until you have the seed (and optional passphrase). It is a BIP39 compatible seed so can use any wallet supporting it including Sparrow, Specter, Electrum etc only need to be aware of the possible high gap limit and the 5 different account of the JM wallet.

See: 
https://github.com/JoinMarket-Org/joinmarket-clientserver/blob/master/docs/USAGE.md#portability
and
https://github.com/openoms/joininbox/blob/master/FAQ.md#wallet-recovery

raulcano, 2022-11-07:
Hi @HonmaSokyu , the same happened to me, but increasing the gap solved the issue. That and waiting for the synchronization, of course, which took a few hours. 

In any case, as @openoms said, if you have the keys, you can recover the btc, so don't panic.

closed 2022-11-15 by rootzoll