raspiblitz/raspiblitz #3394: How to connect to a TEoS watchtower with CLN

wchancao, 2022-10-29:
Steps to install and connect a CLN node to a The Eye of Satoshi (TEoS) watchtower, using the watchtower-client plugin.

`sudo su bitcoin`

Install rustc and cargo
`curl https://sh.rustup.rs -sSf | sh`
`source "$HOME/.cargo/env"`

Clone the repo from [https://github.com/talaia-labs/rust-teos](url)
`cd /home/bitcoin/cl-plugins-available`
`git clone https://github.com/talaia-labs/rust-teos`

Compile the watchtower-client plugin
`cd rust-teos`
`cargo install --path watchtower-plugin`

Symlink the compiled plugin
`ln -s /home/bitcoin/cl-plugins-available/rust-teos/target/release/watchtower-client /home/bitcoin/cl-plugins-enabled/watchtower-client`
`exit`

Restart lightningd
`sudo systemctl restart lightningd`

To connect to a tower, use
`cl registertower tower_id`

If you want to connect to a tower through Tor, add this line to the cl.conf file
`watchtower-proxy=127.0.0.1:9050`

rootzoll, 2022-12-08:
PR merged - fInal dev continues in `eos-watchtower` branch

rootzoll, 2022-12-08:
OK tested on recover and install keeps consistent. Merged into `dev` branch - to be functional tested as Release Canidate for v1.9.0 release.

closed 2022-12-15 by rootzoll