raspiblitz/raspiblitz #3403: How to update raspiblitz to LND 15.4 to fix bug

skifreak27, 2022-11-01:
Tried to follow the instructions from LND page but my linux skills just arent up to par.  my node is basically nuked from the 15.2 LND bug.  Cant figure out how to update it to 15.4 beta for the hotfix, anyone feel like being a big help?

skifreak27, 2022-11-01:
can anyone explain why the LND interim V0.15.4-beta update is listed as "RECKLESS" and "experimental"?  Isn't there an update option to run a script that checks signatures?

rootzoll, 2022-11-02:
I will release a new sd card image v1.8.0c today that you can use for update. The "RECKLESS" update option is just an optional emergency option until a new sd card image is available - its "experimental" because it takes just the latest lnd version from the lightning labs github and runs with it.

closed 2022-11-02 by rootzoll

skifreak27, 2022-11-02:
Cool, thanks for the update!