raspiblitz/raspiblitz #3406: Node sync stuck at 100%.  Electrs won't start.

Zestforlife808, 2022-11-02:


vv01f, 2022-11-02:
Hi @Zestforlife808, 
as quite somepeople already found out yesterday via Twitter or Telegram …
plz update LND (to 0.15.4-beta) and electrs (to 0.9.9). The issue is similar to the need for updating LND to 0.15.2-beta … a faulty common component of the two packages (btcd) …

1. For LND you find the Update option in the menu: Update / LND / …
2. For electrs you can use the cli or wait for a new image release enabling the [update in the menu](https://github.com/rootzoll/raspiblitz/pull/3402) as well; cli (leave menu with Ctrl+c) commands:

    cd /home/electrs/electrs
    sudo -u electrs git fetch
    sudo -u electrs git reset --hard v0.9.9
    sudo -u electrs /home/electrs/.cargo/bin/cargo build --locked --release

rootzoll, 2022-11-02:
LND had a critical bug and also electrs was affected - see: https://twitter.com/openoms/status/1587415789484785667?s=20&t=mVPCiRGA2K8HFJznD4vFaQ

Will release a new sd card image v1.8.0c today that can be used to update and will come with fixes for lnd & electrs.

Zestforlife808, 2022-11-02:
@vv01f @rootzoll Thanks for the quick replies.  I will wait for the image, update, and report back.

Mexxblitz, 2022-11-03:
I have the same Issue with Electrum, LND is working again but Electrum not. Where can I find the new Image release to update Electrum?

vv01f, 2022-11-03:
The new image 1.8.0**c** at: https://github.com/rootzoll/raspiblitz#downloading-the-software includes electrs 0.9.9 as well.

lupn3, 2022-11-03:
#https://github.com/romanz/electrs/issues/787

Zestforlife808, 2022-11-04:
I updated with the linked image and I'm back in action.  Thanks to all.

closed 2022-11-04 by Zestforlife808