raspiblitz/raspiblitz #3409: YES!

Sep21-2009, 2022-11-02:
YES!

```
# download
wget https://github.com/sparrowwallet/sparrow/releases/download/1.7.0/sparrow-server_1.7.0_arm64.deb
wget https://github.com/sparrowwallet/sparrow/releases/download/1.7.0/sparrow-1.7.0-manifest.txt
wget https://github.com/sparrowwallet/sparrow/releases/download/1.7.0/sparrow-1.7.0-manifest.txt.asc
# checksum check
sha256sum -c sparrow-1.7.0-manifest.txt --ignore-missing
# get PGP pubkey
curl https://keybase.io/craigraw/pgp_keys.asc | gpg --import
# signature check
gpg --verify sparrow-1.7.0-manifest.txt.asc

# install
sudo dpkg -i sparrow-server_1.7.0_arm64.deb

# options
/opt/sparrow/bin/Sparrow --help

# start
/opt/sparrow/bin/Sparrow
```
[image]

The config file is in the home folder of the user started Sparrow (default is `admin`):
```
/home/admin/.sparrow/config
```

Example after the first run with the local Tor proxy set:
```
cat .sparrow/config 
{
  "mode": "ONLINE",
  "loadRecentWallets": true,
  "validateDerivationPaths": true,
  "groupByAddress": true,
  "includeMempoolOutputs": true,
  "notifyNewTransactions": true,
  "checkNewVersions": true,
  "openWalletsInNewWindows": false,
  "hideEmptyUsedAddresses": false,
  "showTransactionHex": true,
  "showLoadingLog": true,
  "showAddressTransactionCount": false,
  "showDeprecatedImportExport": false,
  "preventSleep": true,
  "dustAttackThreshold": 1000,
  "enumerateHwPeriod": 30,
  "serverType": "ELECTRUM_SERVER",
  "electrumServer": "tcp://192.168.3.170:50021",
  "recentElectrumServers": [
    "tcp://192.168.3.170:50021"
  ],
  "useProxy": true,
  "proxyServer": "127.0.0.1:9050",
  "maxServerTimeout": 34,
  "maxPageSize": 100,
  "usePayNym": false,
  "sameAppMixing": false
}
```

_Originally posted by @openoms in https://github.com/rootzoll/raspiblitz/issues/3391#issuecomment-1293613368_

Sep21-2009, 2022-11-02:
"preventSleep": true,

openoms, 2022-11-04:
closing this as a duplicate of https://github.com/rootzoll/raspiblitz/issues/3391

closed 2022-11-04 by openoms