raspiblitz/raspiblitz #3417: IP2TOR Subscription unable to succeed

Mexxblitz, 2022-11-05:
After countless hours trying to establish a subscription for the IP2TOR Bridge Service for LNBits I'm still getting the same Error Message. 

First apiGetOrder loop to around 50 times. 

Then afterwards, one of the following Error messages pops up: 
1. Error: Failed HTTP Request
2. Error: Failed to get Invoice

Any idea how to fix this?

raulcano, 2022-11-07:
I experience a similar error when trying to buy a subscription IP2TOR for BTCPay server.
After the apiGetOrder attemps, I get a screen telling that the invoice could not be retrieved.
[image]

The details screen shows this:
[image]

Mexxblitz, 2022-11-09:
Did you manage to get it working again? I still can't get a subscription!

raulcano, 2022-11-10:
Nope. I tried every single day since I posted my comment, but same error unfortunately.

Do you know if there are other shops we could try for this?
It is quite interesting that the subscription menu allows you to input the onion address of a shop, but it comes prefilled with the fulmo one. If other providers offered that service, we could try ....

Mexxblitz, 2022-11-10:
That is a good question. I havnt heard about other shops but will do my research. 

Maybe a maintainer could asnwer our question. Atleast if this is a known error.

rootzoll, 2022-11-10:
At the moment the IP2Tor service run by FULMO is broken and its unsure if it will be further maintained by FULMO. Sorry. But basically its open source and somebody from the community can run an alternative IP2Tor shop & bridge (users can change the onion shop address in the beginning of the subscription dialog). If someone provides a stable alternative I can set that shop as a default in the next RaspiBlitz release. If there is no running shop within the next weeks I might remove the service from the feature set of RaspiBlitz with next version.

Here is the Ip2Tor Shop+Brigde open source code by @frennkie if anybody want to try it out:
https://github.com/frennkie/django-ip2tor

If you are a big fan/user of the Ip2Tor server and you dont want to see it go away - let @fulmolightning on twitter know and motivate for further FULMO support: https://twitter.com/fulmolightning

raulcano, 2022-11-10:
Thanks for that @rootzoll . 
I wrote Fulmo to see if they are planning resuming the service. 
Also, I'll have a look at the code you linked and see if I could do that myself. If so, it won't be soon, so I'd have to let you know in due time...
Cheers!

Miguel-sdj, 2022-11-17:
Does anyone have any more updates on the case?

raulcano, 2022-11-18:
I asked fulmo and it's likely they won't resume the service. Time to check
for alternatives or to build your own IP2TOR with the instructions in the
linked repo.

raulcano, 2022-11-24:
Hi @rootzoll , I forked the Frennkie's source code and I am slowly building a IP2TOR Shop + Bridge, using Docker to isolate each service (nginx, django, tor, redis, postgres, ...)

Here are the repos:
- https://github.com/raulcano/docker-ip2tor-shop
- https://github.com/raulcano/docker-ip2tor-host

I'm still stuck after running the Daphne server (in localhost [+ Tor]). It doesn't accept HTTP requests and I honestly don't know how to proceed. If/When I get to something more stable I'll report back.

rootzoll, 2022-11-24:
@raulcano great to hear - keep us updated.

@frennkie any help/support you can give @raulcano in getting his IP2Tor forks running would be highly appreciated.

raulcano, 2022-11-24:
Thanks , will do.

@frennkie , in case you read this. I'm stuck in this step of your instructions:

```
Initial setup

-> go to Sites and change the initial domain name (and display name)
-> go to user and create operator (and add to "operators" group)
-> go to Hosts and create your first host
```

Daphne and nginx are running, and receiving connections even (can see in the logs when I try to access via browser).
The point is that I don't know where these "Sites" , "User" or "Hosts" pages are, so I'm not sure I can progress without adding the right entries.

For what is worth:
- redis is running and throws no errors in the logs
- the database is running fine (or so it seems). Migrations are executed and tables created successfully
- the celery-beat and celery-worker are not running (I disabled them to add services one by one)

Also, I'm new to most of those things so I'm pretty much learning as I go ...

raulcano, 2022-11-24:
Once started (see screenshots above), I visit from my browser 
http://localhost:8000/

Results:
[image]

[image]

The key seems to be in that "No application configured for scope type 'http'"
It looks the whole setup is not listening to http, or I am missing the correct paths.

This is the nginx config file I'm running at the moment:
https://github.com/raulcano/docker-ip2tor-shop/blob/master/.docker/shop.localhost.conf

`server {
    # Listen on port for incoming requests.
    listen 88;
    server_name  localhost;

    # Django media
    location /media/  {
      root /usr/share/nginx;

    }
    # Django static
    location /static/ {
      root /usr/share/nginx;
    }

    location / {
        proxy_pass http://django:8001;

        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";

        proxy_redirect off;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Host $server_name;
        # proxy_set_header X-Forwarded-Proto https;

    }
}`

raulcano, 2022-11-28:
Since this thread was about the IP2Tor in Raspiblitz not working, I opened an issue in my docker implementation of the shop to bring the discussion there, with my last 2 posts, which belong to a different scope: https://github.com/raulcano/docker-ip2tor-shop/issues/1

rootzoll, 2022-12-07:
@raulcano I can see you closed the issue. Can you sum up the status of your work?

raulcano, 2022-12-07:
Well, it's still work in progress.
I haven't understood from frennkie's code in GitHub which server he used to get to the admin pages, so I managed to access the admin pages via the integrated django development server, which takes care of http requests. I need to change that to a more production ready server (gunicorn maybe?) and update Nginx properly, also for https.

The docker compose structure itself is pretty advanced, but won't get the whole system working yet.
Things that work:
-django development server for http requests
- Tor hidden service
- connection to redis container
- connection to the database container

Things that don't work
- calls to Daphne sockets throw an error
- also I haven't gone through the celery parts just yet .
-pretty much everything else

rootzoll, 2022-12-19:
@raulcano so I wanna go into release candidate phase for v1.9 within the next days. So I wil start to deactivate IP2Tor feature for now with the possibility to maybe get revived in future versions.

raulcano, 2022-12-19:
Thanks for the heads up. I'm actively working on the revamped IP2TOR shop, but it won't be ready soon enough.
I'll let you know once I'm closer to a stable version.
Cheers!

rootzoll, 2023-05-03:
@raulcano https://ip2tor.com/#tor_bridges seems to time out .. any update on the state of the service. BTW sorry for the delayed v1.9.0 release - but hopefully we now getting closer to a final release.

rootzoll, 2023-05-08:
@raulcano any idea yet whats wrong?

raulcano, 2023-05-08:
I think it's related to the certificates. Will try again tomorrow...

raulcano, 2023-05-11:
Hi @rootzoll, I'm having trouble finding the time to fix this, sorry. If you can intend to launch a new release please deactivate my service. As soon as I fix it and leave it on a more stable way I'll let you know. Cheers!

raulcano, 2023-05-27:
> @raulcano <https://github.com/raulcano> OK looks like we finally will
> release v1.9.0 this week and so also the ip2tor.com as a subscribe option
> ... but it still has the problem when I test it: ('failed HTTP code',
> {'status_code': 502, 'timestamp': '2023-05-23T13:09:13Z'})

Hi @Rootzoll, it was impossible for me to work on that during this month,
but I'm actively trying to solve this issue this weekend. Hopefully I'll
manage...
Also, congrats on the new release!

raulcano, 2023-05-27:
Ok, I think the problem is solved. There was a huge log file that broke
Redis and everything else with it.
I will take care of that in the server and document it in the github page

raulcano, 2023-05-30:
> @raulcano <https://github.com/raulcano> I tested the 24h bridge and it
> worked fine - If you like you can make a tweet about the new service now
> available thru the RaspiBlitz SSH subscription menus and I am happy to
> retweet the annoucement.

Ok, will do in a second. Thanks!

closed 2023-05-30 by rootzoll