raspiblitz/raspiblitz #3418: Electrum Service Failed

mattias232, 2022-11-05:
Hi there, im having an issue to get the electrum service to run properly. LN and BTC are both running fine.

I get the following code:

> *** LAST 20 ElectRS LOGS ***
> sudo journalctl -u electrs -b --no-pager -n20
> -- Journal begins at Sat 2022-11-05 06:47:31 GMT, ends at Sat 2022-11-05 20:17:40 GMT. --
> Nov 05 20:16:42 raspberrypi systemd[1]: electrs.service: Consumed 8.208s CPU time.
> Nov 05 20:16:42 raspberrypi systemd[1]: Started Electrs.
> Nov 05 20:16:42 raspberrypi electrs[973427]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.9.7 - the Electrum Rust Server on your RaspiBlitz", args: [] }
> Nov 05 20:16:42 raspberrypi electrs[973427]: [2022-11-05T20:16:42.479Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
> Nov 05 20:16:42 raspberrypi electrs[973427]: [2022-11-05T20:16:42.479Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
> Nov 05 20:16:42 raspberrypi electrs[973427]: [2022-11-05T20:16:42.636Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 1953 SST files, 56.666234871 GB, 4.696112756 Grows
> Nov 05 20:16:46 raspberrypi electrs[973427]: [2022-11-05T20:16:46.604Z INFO  electrs::chain] loading 761240 headers, tip=000000000000000000022fa13b5abb217c073ed005a73834cb886470af426678
> Nov 05 20:16:50 raspberrypi electrs[973427]: [2022-11-05T20:16:50.412Z INFO  electrs::chain] chain updated: tip=000000000000000000022fa13b5abb217c073ed005a73834cb886470af426678, height=761240
> Nov 05 20:16:50 raspberrypi electrs[973427]: [2022-11-05T20:16:50.455Z INFO  electrs::index] indexing 635 blocks: [761241..761875]
> Nov 05 20:16:50 raspberrypi electrs[973427]: thread 'p2p_loop' panicked at 'invalid message: OversizedVectorAllocation { requested: 12000072, max: 4000000 }', src/p2p.rs:250:70
> Nov 05 20:16:50 raspberrypi electrs[973427]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
> Nov 05 20:16:50 raspberrypi electrs[973427]: [2022-11-05T20:16:50.647Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
> Nov 05 20:16:50 raspberrypi electrs[973427]: Error: electrs failed
> Nov 05 20:16:50 raspberrypi electrs[973427]: Caused by:
> Nov 05 20:16:50 raspberrypi electrs[973427]:     0: sync failed
> Nov 05 20:16:50 raspberrypi electrs[973427]:     1: failed to get block 000000000000000000070742427fa10ec3c66d006160155a704d9f56d090a3ea
> Nov 05 20:16:50 raspberrypi electrs[973427]:     2: receiving on an empty and disconnected channel
> Nov 05 20:16:50 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
> Nov 05 20:16:50 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
> Nov 05 20:16:50 raspberrypi systemd[1]: electrs.service: Consumed 8.190s CPU time.
> 
> *** ElectRS Status ***
> ##### STATUS ELECTRS SERVICE
> configured=1
> serviceInstalled=1
> serviceRunning=0
> serviceRunning=0
> tipSynced=0
> initialSynced=0
> electrumResponding=0
> infoSync='Not running - check: sudo journalctl -u electrs'

mattias232, 2022-11-05:
Any help would be greatly appreciated. thanks

istie, 2022-11-06:
I have the same problem running raspiblitz version 1.8.0b.

*** LAST 20 ElectRS LOGS ***
sudo journalctl -u electrs -b --no-pager -n20
-- Journal begins at Sun 2022-11-06 09:19:03 GMT, ends at Sun 2022-11-06 19:49:33 GMT. --
Nov 06 19:48:56 raspberrypi systemd[1]: electrs.service: Consumed 7.935s CPU time.
Nov 06 19:48:57 raspberrypi systemd[1]: Started Electrs.
Nov 06 19:48:57 raspberrypi electrs[235745]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.9.7 - the Electrum Rust Server on your RaspiBlitz", args: [] }
Nov 06 19:48:57 raspberrypi electrs[235745]: [2022-11-06T19:48:57.149Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 06 19:48:57 raspberrypi electrs[235745]: [2022-11-06T19:48:57.149Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 06 19:48:57 raspberrypi electrs[235745]: [2022-11-06T19:48:57.204Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 150 SST files, 37.594497274 GB, 4.696187477 Grows
Nov 06 19:49:01 raspberrypi electrs[235745]: [2022-11-06T19:49:01.261Z INFO  electrs::chain] loading 761248 headers, tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f
Nov 06 19:49:05 raspberrypi electrs[235745]: [2022-11-06T19:49:05.143Z INFO  electrs::chain] chain updated: tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f, height=761248
Nov 06 19:49:05 raspberrypi electrs[235745]: [2022-11-06T19:49:05.195Z INFO  electrs::index] indexing 772 blocks: [761249..762020]
Nov 06 19:49:05 raspberrypi electrs[235745]: thread 'p2p_loop' panicked at 'invalid message: OversizedVectorAllocation { requested: 12000072, max: 4000000 }', src/p2p.rs:250:70
Nov 06 19:49:05 raspberrypi electrs[235745]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Nov 06 19:49:05 raspberrypi electrs[235745]: [2022-11-06T19:49:05.225Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 06 19:49:05 raspberrypi electrs[235745]: Error: electrs failed
Nov 06 19:49:05 raspberrypi electrs[235745]: Caused by:
Nov 06 19:49:05 raspberrypi electrs[235745]:     0: sync failed
Nov 06 19:49:05 raspberrypi electrs[235745]:     1: failed to get block 000000000000000000070742427fa10ec3c66d006160155a704d9f56d090a3ea
Nov 06 19:49:05 raspberrypi electrs[235745]:     2: receiving on an empty and disconnected channel
Nov 06 19:49:05 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 06 19:49:05 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 06 19:49:05 raspberrypi systemd[1]: electrs.service: Consumed 7.895s CPU time.

*** ElectRS Status ***
##### STATUS ELECTRS SERVICE
configured=1
serviceInstalled=1
serviceRunning=0
serviceRunning=0
tipSynced=0
initialSynced=0
electrumResponding=0
infoSync='Not running - check: sudo journalctl -u electrs'

Do not know if it's the same problem as mattias232 but also any help with this would be greatly appreciated as well...

IanMichaelHarper, 2022-11-07:
> Electrs crashed out due to the same block LND bugged out on. If you are on 1.8, I think you can just run `patch` to update Electrs to v.0.9.9.
> 
> Otherwise you can update Electrs manually to 0.9.9 as follows:
> 
> ```
> cd /home/electrs/electrs
> sudo -u electrs git fetch
> sudo -u electrs git reset --hard v0.9.9
> sudo -u electrs /home/electrs/.cargo/bin/cargo build --locked --release
> sudo systemctl restart electrs
> ```
> 
> With this latter option, electrs will have to re-index which may take some time.
> 
> I believe that should fix your issues.

I tried this, but it didn't fix the issue.

I've flashed the 1.8.0c image three times and it still doesn't work out of the box.

I get the same error as @istie - `receiving on an empty and disconnected channel`

more logs of the error:
```Nov 07 13:56:48 raspberrypi electrs[527579]: [2022-11-07T13:56:48.444Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 07 13:56:48 raspberrypi electrs[527579]: [2022-11-07T13:56:48.446Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 07 13:56:48 raspberrypi electrs[527579]: [2022-11-07T13:56:48.565Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 234 SST files, 0.00018837 GB, 0.000000021 Grows
Nov 07 13:56:48 raspberrypi electrs[527579]: [2022-11-07T13:56:48.635Z WARN  electrs::p2p] failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)
Nov 07 13:56:48 raspberrypi electrs[527579]: [2022-11-07T13:56:48.637Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 07 13:56:48 raspberrypi electrs[527579]: Error: electrs failed
Nov 07 13:56:48 raspberrypi electrs[527579]: Caused by:
Nov 07 13:56:48 raspberrypi electrs[527579]:     receiving on an empty and disconnected channel
Nov 07 13:56:48 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 07 13:56:48 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
```

At this point I'm thinking of using an older version of the image just so I can use my node again.

istie, 2022-11-07:
Updated electrum manually as @Lobbelt suggested, but to v0.9.10
It started without errors, 
I deleted the index and now it's rebuilding it, so it will take a while but it looks fine to me.
Thank you for your help!

IanMichaelHarper, 2022-11-07:
deleting the index seems to have fixed it for me too. Let's see when it finally rebuilds...

istie, 2022-11-08:
Small update
Needed to update LND to 0.15.4-beta as well to get it working again.
This was also suggested by @vv01f in #3406 
Now it's all is working fine again.

I suppose you get the same result with the new image (1.8.0c) from @rootzoll which he prepared very quickly.
Thanks again for all the hard work and support, I really appreciate it.

mattias232, 2022-11-10:
thanks for your hlp guys - i've followed advise here and all is now working

closed 2022-11-10 by mattias232