raspiblitz/raspiblitz #3420: Electrum: Missing second parameter

nbommersbach, 2022-11-06:
Hello,

after the recent issue with LND and the update of my node using the new image v1.8.0 provided here, Electrum does not work anymore. I wasn't able to get more information than this:

`sudo journalctl -u electrs -b --no-pager -n20
-- Journal begins at Sun 2022-11-06 06:46:28 GMT, ends at Sun 2022-11-06 21:17:27 GMT. --
Nov 06 19:04:21 raspberrypi electrs[482414]: [2022-11-06T19:04:21.392Z INFO  electrs::index] indexing 1 blocks: [762015..762015]
Nov 06 19:04:21 raspberrypi electrs[482414]: [2022-11-06T19:04:21.479Z INFO  electrs::chain] chain updated: tip=00000000000000000003b2010cce94f3012fa34ef4464fd4daa21ef4e521719a, height=762015
Nov 06 19:11:28 raspberrypi electrs[482414]: [2022-11-06T19:11:28.312Z INFO  electrs::index] indexing 1 blocks: [762016..762016]
Nov 06 19:11:28 raspberrypi electrs[482414]: [2022-11-06T19:11:28.405Z INFO  electrs::chain] chain updated: tip=000000000000000000043c29e9f5dfdaf201d90dd6d601821b1863076a523776, height=762016
Nov 06 19:34:05 raspberrypi electrs[482414]: [2022-11-06T19:34:05.431Z INFO  electrs::index] indexing 1 blocks: [762017..762017]
Nov 06 19:34:05 raspberrypi electrs[482414]: [2022-11-06T19:34:05.646Z INFO  electrs::chain] chain updated: tip=00000000000000000006f75ceb9689c6ef7852dd2d31623faaf724923ed2311e, height=762017
Nov 06 19:38:40 raspberrypi electrs[482414]: [2022-11-06T19:38:40.324Z INFO  electrs::index] indexing 1 blocks: [762018..762018]
Nov 06 19:38:40 raspberrypi electrs[482414]: [2022-11-06T19:38:40.459Z INFO  electrs::chain] chain updated: tip=000000000000000000023ac2baaaeb00e332d37ccabce3b515e16c5ecc0a0141, height=762018
Nov 06 19:38:49 raspberrypi electrs[482414]: [2022-11-06T19:38:49.797Z INFO  electrs::index] indexing 1 blocks: [762019..762019]
Nov 06 19:38:49 raspberrypi electrs[482414]: [2022-11-06T19:38:49.807Z INFO  electrs::chain] chain updated: tip=000000000000000000052cab167e6041ba9c75546d579380dbf43e417cc30c82, height=762019
Nov 06 19:46:12 raspberrypi electrs[482414]: [2022-11-06T19:46:12.197Z INFO  electrs::index] indexing 1 blocks: [762020..762020]
Nov 06 19:46:12 raspberrypi electrs[482414]: [2022-11-06T19:46:12.255Z INFO  electrs::chain] chain updated: tip=00000000000000000004355faef991a06265a42132b88a40f29ace65f053513a, height=762020
Nov 06 20:00:48 raspberrypi electrs[482414]: [2022-11-06T20:00:48.439Z INFO  electrs::index] indexing 1 blocks: [762021..762021]
Nov 06 20:00:48 raspberrypi electrs[482414]: [2022-11-06T20:00:48.654Z INFO  electrs::chain] chain updated: tip=0000000000000000000477d68302d2487f065b3ced2e9bb3419d33c8f1fc904d, height=762021
Nov 06 20:04:16 raspberrypi electrs[482414]: [2022-11-06T20:04:16.603Z INFO  electrs::index] indexing 1 blocks: [762022..762022]
Nov 06 20:04:16 raspberrypi electrs[482414]: [2022-11-06T20:04:16.678Z INFO  electrs::chain] chain updated: tip=000000000000000000027f24bdf229543d5a948f451879a65eaa01b1fa5cf679, height=762022
Nov 06 20:17:26 raspberrypi electrs[482414]: [2022-11-06T20:17:26.986Z INFO  electrs::index] indexing 1 blocks: [762023..762023]
Nov 06 20:17:27 raspberrypi electrs[482414]: [2022-11-06T20:17:27.115Z INFO  electrs::chain] chain updated: tip=0000000000000000000279613284524e5a428e5e5b97158c2c45f2e6b4cf4973, height=762023
Nov 06 20:37:18 raspberrypi electrs[482414]: [2022-11-06T20:37:18.508Z INFO  electrs::index] indexing 1 blocks: [762024..762024]
Nov 06 20:37:18 raspberrypi electrs[482414]: [2022-11-06T20:37:18.735Z INFO  electrs::chain] chain updated: tip=00000000000000000003d63b56f830f3d62c3f1df257ad1a2996438d1d23069b, height=762024`

...which looks good to me. I then tried to connect to Electrum using the Connect-Option in Electrum Menu. But Electrum doesn't show addresses for both local and Tor Connection. 

```
######## How to Connect to Electrum Rust Server #######

Install the Electrum Wallet App on your laptop from:
https://electrum.org

On Network Settings > Server menu:
- deactivate automatic server selection
- as manual server set '' & '50002'
- laptop and RaspiBlitz need to be within same local network

To start directly from laptop terminal use:
electrum --oneserver --server :50002:s

The Tor Hidden Service address for electrs is (see LCD for QR code):

To connect through TOR open the Tor Browser and start with the options:
electrum --oneserver --server :50002:s --proxy socks5:127.0.0.1:9150
error='missing second parameter - see help'

For more details check the RaspiBlitz README on ElectRS:
https://github.com/rootzoll/raspiblitz
```

...which shows the error "missing second parameter - see help". I couldn't find any information about this issue and I also don't know which help I should look at.

Can someone support please?
Thanks in advance!

ChuckNorrison, 2022-11-09:
in the manual to start electrum is the IP missing, only Port 50002 is given. So this is meant with second parameter. If it does not work, you could try to simply start the gui app and go into tools -> network and connect to your electrum

nbommersbach, 2022-11-09:
> Does the electrum wallet show a connection to the local raspiblitz? If you start electrum with the command: electrum --oneserver... it will only connect to one server.

It doesn't even know the command "electrum"... I meanwhile tried to renew the index from Electrs-MENU, but that also did not work.

`admin@IP:~ ₿ electrum --oneserver
-bash: electrum: command not found`

nbommersbach, 2022-11-09:
> in the manual to start electrum is the IP missing, only Port 50002 is given. So this is meant with second parameter. If it does not work, you could try to simply start the gui app and go into tools -> network and connect to your electrum

Ok so the second parameter is the IP... But how do I start the GUI app and what GUI app do you mean? 
I do still think that this has to do with Raspiblitz as my LCD turns black for 5 seconds about every 1 minute since I have the problem with Electrum.
Also, this was immediately after the update to v.1.8.0

ChuckNorrison, 2022-11-09:
we have two parts, the service on raspiblitz called electrs which can be observed with the journal

`sudo journalctl -fu electrs -n 200`

Than you have a client application, here named electrum which wants to connect to the service. The client application is not running on the raspiblitz. So if your PC results with electrum command not found, you need to properly install the client application electrum.

i would suggest to Download the appimage from [here](https://electrum.org/#download), make a `sudo chmod +x electrum-4.3.2-x86_64.AppImage` on the file and call it with `./electrum-4.3.2-x86_64.AppImage`

nbommersbach, 2022-11-09:
but this is all console output while being logged in as admin on my raspiblitz.
Then in the menu, I go to "ELECTRS" -> "Connect" which then leaves me with this output.

I have connected to Electrum Rust Server multiple times before, but since the update I can't and ELECTRS isn't showing me addresses to connect to either.

```
######## How to Connect to Electrum Rust Server #######

Install the Electrum Wallet App on your laptop from:
https://electrum.org

On Network Settings > Server menu:
- deactivate automatic server selection
- as manual server set '' & '50002'
- laptop and RaspiBlitz need to be within same local network

To start directly from laptop terminal use:
electrum --oneserver --server :50002:s

The Tor Hidden Service address for electrs is (see LCD for QR code):

To connect through TOR open the Tor Browser and start with the options:
electrum --oneserver --server :50002:s --proxy socks5:127.0.0.1:9150
error='missing second parameter - see help'

For more details check the RaspiBlitz README on ElectRS:
https://github.com/rootzoll/raspiblitz
```

Before the update, ELECTRS showed me both TOR and local addresses.

I guess I don't need specific software on my PC in order for ELECTRS to show me an address to connect to, do I?

nbommersbach, 2022-11-12:
both commands work and give back the addresses. The tor address is the one I used for my raspiblitz over the last year...

via Sparrow wallet, I can now connect again to my blitz. But still there is no address shown in the menu...

rootzoll, 2022-11-14:
FYI the local IP not showing was fixed in v1.8 branch - on terminal just run `patch code`

closed 2022-12-15 by rootzoll