raspiblitz/raspiblitz #3425: Can not install any app that was installed before

KainerW, 2022-11-08:
I have the problem that I can not install mempool.space, rtl or thunderhub. I have had them all installed at one point in the past.
Now I want to install e.g. mempool.space again but the system wont let me. I get the error message:

*Fail on mempool explorer install*
Try manual install on terminal after reboot with /home/admin/config.scripts/bonus.mempool.sh on

After reboot, trying to install in terminal does not work neither.

Tried to install in web-UI with click on "+ Install" button. Does not work. Error-msg: "install script did not run trough".

ChuckNorrison, 2022-11-09:
You should toggle off and on. Better to run the script manual to get the output shown.

`/home/admin/config.scripts/bonus.mempool.sh off`

`/home/admin/config.scripts/bonus.mempool.sh on`

Check output for errors

KainerW, 2022-11-12:
That sadly did work only for mempool app, not for rtl or thunderhub. Any hints on what I can try? 

E.g. when trying to install thunderhub I get this:

npm ERR! While resolving: react-qr-reader@2.2.1
npm ERR! Found: react@17.0.2
npm ERR! node_modules/react
npm ERR!   react@"^17.0.2" from the root project
npm ERR!   peerOptional react@"^16.8.0 || ^17.0.0" from @apollo/client@3.5.8
npm ERR!   node_modules/@apollo/client
npm ERR!     @apollo/client@"^3.5.8" from the root project
npm ERR!   30 more (@emotion/react, @visx/axis, @visx/bounds, @visx/chord, ...)
npm ERR! 
npm ERR! Could not resolve dependency:
npm ERR! peer react@"~16" from react-qr-reader@2.2.1
npm ERR! node_modules/react-qr-reader
npm ERR!   react-qr-reader@"^2.2.1" from the root project
npm ERR! 
npm ERR! Conflicting peer dependency: react@16.14.0
npm ERR! node_modules/react
npm ERR!   peer react@"~16" from react-qr-reader@2.2.1
npm ERR!   node_modules/react-qr-reader
npm ERR!     react-qr-reader@"^2.2.1" from the root project
...

ChuckNorrison, 2022-11-12:
I would try to uninstall nodejs and try again... Something have produced  conflicts of installation.

For this action:
Uninstall RTL and Thunderhub and try uninstall nodejs with 

`/home/admin/config.scripts/bonus.nodejs.sh off`

Then try install again. 

Or just reflash SD card with current release image.

KainerW, 2022-11-13:
Thank you very much, reinstalling nodejs did work.

cstenglein, 2023-06-07:
Looks resolved => closing

closed 2023-06-07 by cstenglein