raspiblitz/raspiblitz #3427: Reduce bloat of the SDcard image

openoms, 2022-11-09:
The bitcoin user directory has a lot of data left over from compiling the CLN GRPC plugin:
```
₿ sudo du -hd1 /home/bitcoin
261M	/home/bitcoin/bitcoin
4.0K	/home/bitcoin/cl-plugins-enabled
1.7G	/home/bitcoin/cl-plugins-available
4.0K	/home/bitcoin/scl-plugins-enabled
40K	/home/bitcoin/.config
562M	/home/bitcoin/lightning
248M	/home/bitcoin/.cargo
106M	/home/bitcoin/.local
1.4G	/home/bitcoin/.rustup
4.0K	/home/bitcoin/tcl-plugins-enabled
36K	/home/bitcoin/.gnupg
482M	/home/bitcoin/.cache
4.9G	/home/bitcoin
```

Can be deleted after (in) the SDcard build:
```
/home/bitcoin/.rustup
/home/bitcoin/.cargo/
/home/bitcoin/.cache
```

Then need to test if there is anything else needed from here:
```
sudo du -ahd1 /home/bitcoin/cl-plugins-available/cln-grpc/debug/
4.0K	/home/bitcoin/cl-plugins-available/cln-grpc/debug/cln-grpc.d
0	/home/bitcoin/cl-plugins-available/cln-grpc/debug/.cargo-lock
336M	/home/bitcoin/cl-plugins-available/cln-grpc/debug/incremental
4.0K	/home/bitcoin/cl-plugins-available/cln-grpc/debug/examples
1.1G	/home/bitcoin/cl-plugins-available/cln-grpc/debug/deps
3.4M	/home/bitcoin/cl-plugins-available/cln-grpc/debug/.fingerprint
199M	/home/bitcoin/cl-plugins-available/cln-grpc/debug/build
```
apart from the binary:
```
/home/bitcoin/cl-plugins-available/cln-grpc/debug/cln-grpc
```

rootzoll, 2022-11-27:
Added to a lot of `git clones` the parameter `--single-branch` to reduce the amount of data a git repo with a lot of branches might take

rootzoll, 2022-11-28:
OK I tested with the cleanup cl install scripts ... Spark & Sparko work, but RTL has some socket issues. Can this be caused by the clean ups? Or just a cRTL thing to investigate seperatly?

rootzoll, 2022-12-06:
OK merged. Closing issue for now.

closed 2022-12-06 by rootzoll