raspiblitz/raspiblitz #3428: migration from virtual umbrel to virtual raspiblitz not working

cwgriswold, 2022-11-09:
tried to migrate from virtualized umbrel 0.5.3 to virtualized raspiblitz 1.8.0c on debian 11.5 x64. the mounted 1tb umbrel data hdd is not recognized by the setup routine. setup screen only showing freshsetup, frombackup, shutdown.

any ideas?

cwgriswold, 2022-11-09:
mounting manually after reboot doesn't work. umbrel hdd is xfs. do i need to convert to ext4 before hand?

openoms, 2022-11-09:
right unfortunately the raspiblitz scripts don't support XFS. You would be best to set up new and copy the chain and LND data manually.

cwgriswold, 2022-11-11:
gave up. i manually moved lnd folder from umbrel to raspi but lnd service doesn't start. can't even unlock wallet, neither with umbrel web password nor with the default moneyprintergobrrr. since the umbrel code massively changed in version >0.5.0 the migration fails.

cwgriswold, 2022-11-11:
what files from the lnd directory need to be copied to rapsiblitz?

cwgriswold, 2022-11-12:
Finally managed to manually migrate from virtual umbrel 0.5.3 on ubuntu 22.04 LTS x64 to virtual raspiblitz 0.8.0c on debian 11.5 x64

here is how to.
1. backup umbrel as usual
2. change umbrel password to something raspiblitz understands (no special characters!!!!)
3. stop lnd on umbrel (preserve lnd channel db status to prevent fund loss)
 `sudo /umbrel01/scripts/app compose lightning stop lnd`
4. prepare lndrescue file for raspiblitz on umbrel
5. sudo mkdir /mnt/hdd and mkdir /mnt/hdd/lnd
6. sudo cp -r ./app-data/lightning/data/lnd/data /mnt/hdd/lnd
7. sudo tar -czvf lnd-rescue-.tar.gz /mnt/hdd/lnd
8. md5sum lnd-rescue-.tar.gz (will show the md5 hash)
9. copy the hash sum to lnd-rescue-.tar.gz so it lokks like lnd-rescue-2asdasd234kk234.tar.gz
10. copy the file to the new raspiblitz /mnt/hdd/upload
11. shutdown umbrel just in case
12. on raspiblitz menu/repair/repair-lnd/lndrescue follow the instructions
13. if the rescuefile is beeing accepted you should be able to see your open channels and onchain funds.
14. if lnd is up and running burn umbrel.

closed 2022-11-13 by cwgriswold

rootzoll, 2022-11-14:
@cwgriswold thanks for sharing your experience. just one question: was this XFS your choice of your virtual setup or is XFS a new option/feature in umbrel I missed so far?

cwgriswold, 2022-11-15:
it was my choice. have to check whats the default filesystem on ubuntu 22.04 but i think it was zfs. i changed it to xfs for having good experience on large drives with databases on.

umbrel doesn't bother with the filesystem as long as the 1tb hard drive can be mounted, all should be fine.

used fstransform to convert xfs to ext4. took about 90 minutes.