raspiblitz/raspiblitz #3429: WebUi Installscripts delivering version -> API -> WebUI

rootzoll, 2022-11-09:
https://github.com/cstenglein/raspiblitz-web/blob/master/src/utils/availableApps.ts

rootzoll, 2022-12-11:
@cstenglein whats the status here - still targeting for 1.9. or should I push up to 1.10?

cstenglein, 2022-12-11:
You & @fusion44 need to decide that since most of the work is on your part:

- RaspiBlitz: providing the version info somewhere so that blitzAPI can get it
- BlitzAPI: providing the info somewhere
- webUI: using the info, which is the least work IMO

fusion44, 2022-12-11:
Sure, `raspiBlitzVersion` is already being [delivered](https://github.com/fusion44/blitz_api/blob/36be181e77291da242de9b430285b9fd05011d1c/app/system/impl/raspiblitz.py#L65) correctly. Adding `codeVersion` won't be a problem.

fusion44, 2022-12-11:
@rootzoll What is the difference between `raspiBlitzVersion` and `codeVersion`?

rootzoll, 2022-12-11:
`raspiBlitzVersion` comes from the config file - says which version the config is from. `codeVersion` is directoly from the code files. Normally both are the same ... just on starting a recovery `raspiBlitzVersion` could differ from `codeVersion`. `codeVersion` is the the better source to use in this case.

cstenglein, 2022-12-12:
So `codeVersion` is specific to the app? Or is that the version of the installscripts?

If it's the second case, then this issue isn't what I thought it would be. I thought the app versions are provided by raspiblitz to the UI. Can we open another issue in that case?

rootzoll, 2022-12-14:
Ah OK sorry .. I now added version information to the status of the install scripts for the apps (already merged into dev). Next step is to add this to the API ..@fusion44 I will provide an PR for that.

rootzoll, 2022-12-14:
OK PR is open to be merged for Blitz API: https://github.com/fusion44/blitz_api/pull/174

@cstenglein basically you can already implement this on webUI side now .. just expect that a request to `http://192.168.178.147/api/latest/apps/status` will now also have in here:
```
  {
    "id": "thunderhub",
    "installed": true,
    "status": "online",
    "address": "http://192.168.178.147:3010",
    "httpsForced": "0",
    "httpsSelfsigned": "1",
    "hiddenService": "653obthlroknoz6lofrnq57molekgqk72hcawchxuynqzps5sael2nqd.onion",
    "authMethod": "password_b",
    "details": {},
    "error": ""
  },
  {
```
also a `version` field that will have a version string to be displayed.

DocBrown101, 2022-12-29:
I'm not sure if this fits here, but with the version raspiblitz-min-v1.9.0rc1 the API and the webUI can not be activated.

[image]
A success message is given, but it is false and visible far too quickly.

rootzoll, 2023-01-13:
- [ ] retest install API+UI on min image

rootzoll, 2023-05-08:
Tested with v1.9.0rc4 minimal image. API & WebUI installs and works after reboot. Also Verisons of API & WebUI get shown under settings and are part of the debug report. Closing issue for v1.9.0 release.

closed 2023-05-08 by rootzoll