raspiblitz/raspiblitz #3431: Once again, couldn't connect to LND

bigsome, 2022-11-10:
I think there are problems again with LND... I have the last version (0.15.4) but yesterday it disconnected and it hasn't recovered yet...

debug: 
https://l.termbin.com/vuhju

I'm considering to install and run cln instead of LND... is possible to do that with raspiblitz?

Thanks!

bigsome, 2022-11-10:
Well, I see there is an option to activate cln... if I install cln, it will keep all the current channels I have? or I'll have to do some kind of migration? close and re-open channels?

Thanks!

ChuckNorrison, 2022-11-10:
whats with this issue? Why create a new one? 
https://github.com/rootzoll/raspiblitz/issues/3386

LND works great in version 0.15.4. if it does not work for your, reflash your sd card and wait for the syncing. You can install C-Lightning, but with new channels. LND and CLN using their own channels and backups

bigsome, 2022-11-10:
Yes, sorry but didn't update the other issue... I updated and it managed to connect and work for a week.

I'll try to reflash my SD. Thanks!

cstenglein, 2022-11-10:
@bigsome there have been two LND bugs the last two weeks, that's why it worked for some time after you updated :)

The easiest way to update would be to flash the 1.8.0c image version found here: https://github.com/rootzoll/raspiblitz#downloading-the-software

Closing the issue for now, if you still got issues, just re-open this one :)

closed 2022-11-10 by cstenglein

bigsome, 2022-11-20:
Hello,
I am reopening this problem because I am encountering the same issue over and over again.

After upgrading to version 1.8.0c and flashing two cards just in case. My node disconnected from LND, I reinstalled with the second card, it was working for about 1 week and again, disconnection from LND.  This time I did a reboot and it reconnected to last about 3 days and then went down again.

I don't know what else I can do, I have other noderunners quite annoyed with this downtime. I've had the node up to date for more than a year with no problems, and for the last few months it's been impossible with LND to keep it stable. And I think it's not my fault... what makes this situation a bit unfair.

What do you suggest?

Thanks for your help...

My debug log: https://l.termbin.com/qdt1

[image]

** I am restarting the node to keep it connected, however long it lasts.

ChuckNorrison, 2022-11-20:
For me this looks like a TOR related connectivity issue... could be interesting what lnd log can show us.

`sudo journalctl -fu lnd -n 400`

The second reason could be an SSD failure. This can be tested with smartmontools, further infos here
https://github.com/rootzoll/raspiblitz/issues/2599

bigsome, 2022-11-22:
Hi there @ChuckNorrison thanks for helping me...

A peer recommended me to compact the channel.db so I've changed db.bolt.auto-compact-min-age to 0 on lnd.conf... restarted lnd service (I had a small issue editing lnd.conf but solved and restarted like a charm).

Following your suggestion, this is the lnd log:

```
-- Journal begins at Wed 2022-11-16 17:31:40 GMT. --
Nov 18 12:18:14 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 18 12:20:41 raspberrypi systemd[1]: Started Lightning Network Daemon.
Nov 19 17:49:15 raspberrypi systemd[1]: lnd.service: Succeeded.
Nov 19 17:49:15 raspberrypi systemd[1]: lnd.service: Consumed 1h 8min 25.210s CPU time.
-- Boot 6c23fce284f0420299f498eeb18cccbb --
Nov 20 21:49:03 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 20 21:54:42 raspberrypi systemd[1]: Started Lightning Network Daemon.
Nov 22 09:01:13 raspberrypi systemd[1]: Stopping Lightning Network Daemon...
Nov 22 09:01:16 raspberrypi systemd[1]: lnd.service: Succeeded.
Nov 22 09:01:16 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
Nov 22 09:01:16 raspberrypi systemd[1]: lnd.service: Consumed 1h 35min 3.826s CPU time.
Nov 22 09:01:16 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 22 09:01:54 raspberrypi systemd[1]: Started Lightning Network Daemon.
Nov 22 09:01:56 raspberrypi systemd[1]: Stopping Lightning Network Daemon...
Nov 22 09:01:56 raspberrypi lnd[3386763]: error creating wallet config: unable to set up wallet password listeners: shutting down
Nov 22 09:01:56 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 09:01:56 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 22 09:01:56 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
Nov 22 09:01:56 raspberrypi systemd[1]: lnd.service: Consumed 6.869s CPU time.
-- Boot cb6d22f16c674bb1a9d104c4d2dc59fa --
Nov 22 09:16:05 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 22 09:16:05 raspberrypi lnd.check.sh[38231]: expr: syntax error: unexpected argument ‘62’
Nov 22 09:16:05 raspberrypi lnd.check.sh[38235]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:16:05 raspberrypi lnd.check.sh[38240]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:16:07 raspberrypi lnd[38297]: failed to load config: /home/bitcoin/.lnd/lnd.conf:17: unknown option: ignore-historical-gossip-filters
Nov 22 09:16:07 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 09:16:07 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 22 09:16:07 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.
Nov 22 09:17:07 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 1.
Nov 22 09:17:07 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
Nov 22 09:17:07 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 22 09:17:08 raspberrypi lnd.check.sh[45621]: expr: syntax error: unexpected argument ‘62’
Nov 22 09:17:08 raspberrypi lnd.check.sh[45625]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:17:08 raspberrypi lnd.check.sh[45630]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:17:09 raspberrypi lnd[45687]: failed to load config: /home/bitcoin/.lnd/lnd.conf:17: unknown option: ignore-historical-gossip-filters
Nov 22 09:17:09 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 09:17:09 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 22 09:17:09 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.
Nov 22 09:18:09 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 2.
Nov 22 09:18:09 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
Nov 22 09:18:09 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 22 09:18:10 raspberrypi lnd.check.sh[47290]: expr: syntax error: unexpected argument ‘62’
Nov 22 09:18:10 raspberrypi lnd.check.sh[47294]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:18:10 raspberrypi lnd.check.sh[47299]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:18:11 raspberrypi lnd[47379]: failed to load config: /home/bitcoin/.lnd/lnd.conf:17: unknown option: ignore-historical-gossip-filters
Nov 22 09:18:11 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 09:18:11 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 22 09:18:11 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.
Nov 22 09:19:11 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 3.
Nov 22 09:19:11 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
Nov 22 09:19:11 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 22 09:19:12 raspberrypi lnd.check.sh[49527]: expr: syntax error: unexpected argument ‘62’
Nov 22 09:19:12 raspberrypi lnd.check.sh[49531]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:19:12 raspberrypi lnd.check.sh[49536]: sed: -e expression #1, char 2: extra characters after command
Nov 22 09:19:13 raspberrypi lnd[49622]: failed to load config: /home/bitcoin/.lnd/lnd.conf:17: unknown option: ignore-historical-gossip-filters
[... cut ...]
Nov 22 11:05:12 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Nov 22 11:15:29 raspberrypi systemd[1]: Started Lightning Network Daemon.
```

Here you can see the error with the lnd.conf I mentioned above (I edited the log to remove the iterative fail...). I don't see any other errors, so my guess is that the problem is/was about channel.db size (somehow SSD related) so I'm giving a grace time to check if it goes down again or not... although I'll test the SDD health with smartmontools to see if everything's ok.

What do you think? do you agree?

Thanks!

ChuckNorrison, 2022-11-22:
This looks very bad... could be a faulty ssd or power outtage which corrupted files on your ssd. I would check the lnd.conf for obvious failures.

> Nov 22 10:57:03 raspberrypi systemd[1]: Starting Lightning Network Daemon...
> Nov 22 10:57:03 raspberrypi lnd.check.sh[49554]: expr: syntax error: unexpected argument ‘62’
> Nov 22 10:57:03 raspberrypi [...]

A good lnd.conf till line 17 looks like this:

```
# lnd configuration

[Application Options]
accept-keysend=true
externalip=1.2.3.4:9735
listen=0.0.0.0:9735
tlsdisableautofill=1
tlsautorefresh=1
stagger-initial-reconnect=1
sync-freelist=1
ignore-historical-gossip-filters=1

```

bigsome, 2022-11-22:
Yes @ChuckNorrison that was because of some text I added accidentally to my lnd.conf this morning, when modifying db.bolt.auto-compact-min-age... completely my fault =) but as I explained, as soon as I realized (in one hour more or less) I fixed it...

If you see everything else ok, that's great...

bigsome, 2022-11-23:
Today down again... adding the last journal lines about lnd:

```
Nov 22 11:15:29 raspberrypi systemd[1]: Started Lightning Network Daemon.
Nov 23 12:06:48 raspberrypi systemd[1]: lnd.service: Succeeded.
Nov 23 12:06:48 raspberrypi systemd[1]: lnd.service: Consumed 1h 1min 28.526s CPU time.
```

Nothing to see here so I looked everything for the exact time when it went down:

```
Nov 23 12:06:24 raspberrypi _background.scan.sh[696]: SystemScan Loop done in 5 seconds
Nov 23 12:06:28 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/blitz>
Nov 23 12:06:29 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitco>
Nov 23 12:06:33 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitco>
Nov 23 12:06:35 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/lnd.m>
Nov 23 12:06:37 raspberrypi _background.scan.sh[696]: skipping because cl=off
Nov 23 12:06:37 raspberrypi _background.scan.sh[696]: skipping because testnet=off
Nov 23 12:06:37 raspberrypi _background.scan.sh[696]: skipping because signet=off
Nov 23 12:06:38 raspberrypi _background.scan.sh[696]: SystemScan Loop done in 11 seconds
Nov 23 12:06:42 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/blitz>
Nov 23 12:06:44 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitco>
Nov 23 12:06:47 raspberrypi sudo[2941689]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/du -s>
Nov 23 12:06:47 raspberrypi sudo[2941689]: pam_unix(sudo:session): session opened for user root(>
Nov 23 12:06:47 raspberrypi sudo[2941689]: pam_unix(sudo:session): session closed for user root
Nov 23 12:06:48 raspberrypi systemd[1]: lnd.service: Succeeded.
Nov 23 12:06:48 raspberrypi systemd[1]: lnd.service: Consumed 1h 1min 28.526s CPU time.
Nov 23 12:06:49 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitco>
Nov 23 12:06:49 raspberrypi python[740]: ERROR:asyncio:Task exception was never retrieved
Nov 23 12:06:49 raspberrypi python[740]: future: <Task finished name='Task-1959' coro=<_handle_f>
Nov 23 12:06:49 raspberrypi python[740]: Traceback (most recent call last):
Nov 23 12:06:49 raspberrypi python[740]:   File "./app/repositories/ln_impl/lnd_grpc.py", line 6>
```

Here I see that ERROR:asyncio:Task exception was never retrieved but is **after** the service goes down... Not before, so I don't think it's the cause but a consequence... And the other question is that it doesn't recover automatically... only rebooting.

Any idea?

ChuckNorrison, 2022-11-23:
Could search for some system trouble like under voltage events or something... i dont know why your lnd cant reach the bitcoind RPC. Take the timestamp from the event and check syslog and journal of bitcoind. 

`cat /var/log/syslog | grep -c "Under-voltage detected!"`

bigsome, 2022-11-24:
Thanks for your help... This is what I see:

There's no entries with "Under-voltage detected!" I think that's a dead end

And this is the syslog from just before and 1 minute after the event:

```
Nov 23 12:06:42 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/blitz.datadrive.sh status
Nov 23 12:06:44 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitcoin.monitor.sh mainnet info
Nov 23 12:06:48 raspberrypi systemd[1]: lnd.service: Succeeded.
Nov 23 12:06:48 raspberrypi systemd[1]: lnd.service: Consumed 1h 1min 28.526s CPU time.
Nov 23 12:06:49 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitcoin.monitor.sh mainnet mempool
Nov 23 12:06:49 raspberrypi python[740]: ERROR:asyncio:Task exception was never retrieved
Nov 23 12:06:49 raspberrypi python[740]: future: <Task finished name='Task-1959' coro=<_handle_forward_event_listener() done, defined at ./app/repositories/lightning.py:236> exception=HTTPException(status_code=500, detail='htlc event subscription terminated')>
Nov 23 12:06:49 raspberrypi python[740]: Traceback (most recent call last):
Nov 23 12:06:49 raspberrypi python[740]:   File "./app/repositories/ln_impl/lnd_grpc.py", line 687, in listen_forward_events
Nov 23 12:06:49 raspberrypi python[740]:     async for e in _router_stub.SubscribeHtlcEvents(request):
Nov 23 12:06:49 raspberrypi python[740]:   File "/usr/local/lib/python3.9/dist-packages/grpc/aio/_call.py", line 326, in _fetch_stream_responses
Nov 23 12:06:49 raspberrypi python[740]:     await self._raise_for_status()
Nov 23 12:06:49 raspberrypi python[740]:   File "/usr/local/lib/python3.9/dist-packages/grpc/aio/_call.py", line 236, in _raise_for_status
Nov 23 12:06:49 raspberrypi python[740]:     raise _create_rpc_error(await self.initial_metadata(), await
Nov 23 12:06:49 raspberrypi python[740]: grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
Nov 23 12:06:49 raspberrypi python[740]: #011status = StatusCode.UNKNOWN
Nov 23 12:06:49 raspberrypi python[740]: #011details = "htlc event subscription terminated"
Nov 23 12:06:49 raspberrypi python[740]: #011debug_error_string = "{"created":"@1669205207.094664567","description":"Error received from peer ipv4:127.0.0.1:10009","file":"src/core/lib/surface/call.cc","file_line":952,"grpc_message":"htlc event subscription terminated","grpc_status":2}"
Nov 23 12:06:49 raspberrypi python[740]: >
Nov 23 12:06:49 raspberrypi python[740]: During handling of the above exception, another exception occurred:
Nov 23 12:06:49 raspberrypi python[740]: Traceback (most recent call last):
Nov 23 12:06:49 raspberrypi python[740]:   File "./app/repositories/lightning.py", line 256, in _handle_forward_event_listener
Nov 23 12:06:49 raspberrypi python[740]:     async for i in ln.listen_forward_events():
Nov 23 12:06:49 raspberrypi python[740]:   File "./app/repositories/ln_impl/lnd_grpc.py", line 715, in listen_forward_events
Nov 23 12:06:49 raspberrypi python[740]:     raise HTTPException(
Nov 23 12:06:49 raspberrypi python[740]: fastapi.exceptions.HTTPException: 500
Nov 23 12:06:56 raspberrypi python[740]: ERROR:asyncio:Task exception was never retrieved
Nov 23 12:06:56 raspberrypi python[740]: future: <Task finished name='Task-1957' coro=<_handle_info_listener() done, defined at ./app/repositories/lightning.py:207> exception=HTTPException(status_code=500, detail='failed to connect to all addresses')>
Nov 23 12:06:56 raspberrypi python[740]: Traceback (most recent call last):
Nov 23 12:06:56 raspberrypi python[740]:   File "./app/repositories/ln_impl/lnd_grpc.py", line 600, in get_ln_info_impl
Nov 23 12:06:56 raspberrypi python[740]:     response = await _lnd_stub.GetInfo(req)
Nov 23 12:06:56 raspberrypi python[740]:   File "/usr/local/lib/python3.9/dist-packages/grpc/aio/_call.py", line 290, in __await__
Nov 23 12:06:56 raspberrypi python[740]:     raise _create_rpc_error(self._cython_call._initial_metadata,
Nov 23 12:06:56 raspberrypi python[740]: grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
Nov 23 12:06:56 raspberrypi python[740]: #011status = StatusCode.UNAVAILABLE
Nov 23 12:06:56 raspberrypi python[740]: #011details = "failed to connect to all addresses"
Nov 23 12:06:56 raspberrypi python[740]: #011debug_error_string = "{"created":"@1669205216.388474510","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3217,"referenced_errors":[{"created":"@1669205216.388470509","description":"failed to connect to all addresses","file":"src/core/lib/transport/error_utils.cc","file_line":165,"grpc_status":14}]}"
Nov 23 12:06:56 raspberrypi python[740]: >
Nov 23 12:06:56 raspberrypi python[740]: During handling of the above exception, another exception occurred:
Nov 23 12:06:56 raspberrypi python[740]: Traceback (most recent call last):
Nov 23 12:06:56 raspberrypi python[740]:   File "./app/repositories/lightning.py", line 211, in _handle_info_listener
Nov 23 12:06:56 raspberrypi python[740]:     info = await ln.get_ln_info_impl()
Nov 23 12:06:56 raspberrypi python[740]:   File "./app/repositories/ln_impl/lnd_grpc.py", line 604, in get_ln_info_impl
Nov 23 12:06:56 raspberrypi python[740]:     raise HTTPException(
Nov 23 12:06:56 raspberrypi python[740]: fastapi.exceptions.HTTPException: 500
Nov 23 12:06:58 raspberrypi _background.scan.sh[696]: skipping because cl=off
[... cut ...]
Nov 23 12:07:02 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/blitz.datadrive.sh status
Nov 23 12:07:03 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/bitcoin.monitor.sh mainnet status
Nov 23 12:07:05 raspberrypi kernel: [91196.591897] [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00:fb:5e:30:ab:bf:9b:07:08:00 SRC=192.168.0.46 DST=224.0.0.251 LEN=32 TOS=0x00 PREC=0x00 TTL=1 ID=19008 PROTO=2 
Nov 23 12:07:09 raspberrypi _background.scan.sh[696]: updating: /home/admin/config.scripts/lnd.monitor.sh mainnet status
```

ChuckNorrison, 2022-11-24:
Cant help here, i dont know whats wrong on your node. You can join the Raspiblitz Telegram and try to get better help there.

bigsome, 2022-11-24:
Thanks @ChuckNorrison  you helped a lot!

bigsome, 2022-11-30:
No help from the Telegram group... The only one who replied (openoms) suggested to flash a new endurance card and then format the SSD. Flashing is ok, but to format a new SSD and don't lose any funds and get it synced, up and running as fast as possible... What is the best way? If I backup the SSD and restore it, maybe I will be getting the same error, right?

Although... I'm a newbie so if this is a nonsense just don't laugh at me =)
I wanted to ask your opinion about an idea I got today:

Since rebooting manages to recover lnd, I thought that maybe there's something that could launch lnd without rebooting (because reboot asks for the password and would get stuck anyway). So, this is what I was thinking:

**First:** See if I can (and how to) recover lnd with the command line 

**Second:** I know there's a background scan that checks if everything is running and ok but I'm wondering to create another "scan":
Creating a CRON job that runs a script checking for the exact conditions/state of the lnd service once it's down to run the command/s I found that recover lnd service.

Right now I'm waiting for lnd to go down... let's see

Thanks!