raspiblitz/raspiblitz #3433: Error: electrs failed

NokSuKao-O, 2022-11-11:
Hi, 

Im neither a programmer nor an computer expert so forgive me for all what is coming ;-)

Here is "my" problem:

I try to start Electrum via the Raspiblitz Menu (worked fine for a all the time)

It says that it failed to get block 761249 (see Hash of the block in following debug info):

Nov 11 07:41:33 raspberrypi electrs[3228861]:     1: failed to get block 000000000000000000070742427fa10ec3c66d006160155a704d9f56d090a3ea
Nov 11 07:41:33 raspberrypi electrs[3228861]:     2: receiving on an empty and disconnected channel
Nov 11 07:41:33 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE

Best regards
Nok

cstenglein, 2022-11-11:
Hi @NokSuKao-O 

Which version of electrs are you running? Did you update to the latest version? There has been a bug recently which caused electrum & LND to stop processing blocks.

You can re-flash the 1.8.0c image with the latest versions of LND & electrs, which fixes that: https://github.com/rootzoll/raspiblitz#downloading-the-software

NokSuKao-O, 2022-11-11:
Hi @cstenglein,

as I said, forgive me XD ....

I thought I had done the necessary updates after the bug... but I did not, obviously.

The update is running now. Im assuming that this will fix it.

Thank you so much for taking your time.

Send a lightning address and I'll be happy to "tip" you ;-)

cstenglein, 2022-11-11:
Hi

If you like you can tip the RaspiBlitz development :)

https://tallycoin.app/@rootzoll/raspiblitz-development-r5lx23/

I'll close the issue for now, if you still got the problem, don't hesitate to re-open it.

closed 2022-11-11 by cstenglein