raspiblitz/raspiblitz #3435: Google API - Incompatible python packages

rek79, 2022-11-12:
https://github.com/rootzoll/raspiblitz/blob/1a04af5c99c2b74a89a0f9a7f0398dc37e602f3a/build_sdcard.sh#L318

Building under Armbian Buster Odroid HC2, I get the following error:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
google-api-core 2.10.2 requires googleapis-common-protos<2.0dev,>=1.56.2, but you have googleapis-common-protos 1.53.0 which is incompatible.
google-api-core 2.10.2 requires protobuf!=3.20.0,!=3.20.1,!=4.21.0,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0dev,>=3.19.5, but you have protobuf 3.20.1 which is incompatible.
```
I changed following two dependencies which seemed to work:
```
googleapis-common-protos==1.56.4
protobuf==4.21.9
```

`python_libs="grpcio==1.42.0 googleapis-common-protos==1.56.4 toml==0.10.2 j2cli==0.3.10 requests[socks]==2.21.0 protobuf==4.21.9"`

rek79, 2022-11-12:
I'm not sure if this was due to installing googleapis manually after patching from 1.7.1 to 1.8 did no succeed.  I could have installed a later version that was incompatible.

rootzoll, 2022-11-14:
@openoms is there anyting special building these under "Armbian Buster Odroid HC2"?

rek79, 2022-11-14:
@rootzoll I've been checking build_sdcard.sh line by line.  Besides the issues above I've found that the standard repo for apt-get is limited to python 3.7.  I was able to build 3.10 from source though.

rek79, 2022-11-15:
Looks like `sysbench` is also not available for armhf buster.  I'll keep compiling a list and start to suggest fallback code in a new PR to install from source or alt PPA.