raspiblitz/raspiblitz #3436: After 1.8.0 Update USB Backup is not possible

lazyrudi, 2022-11-12:
I tried to create a USB Backup and ran in the problem that more than one backup targets are connected. Than I tried a backup without the USB and got the information the Zram was detected as Backup USB device... What do I have to do to backup my Raspiblitz to my external USB drive?
[image]
[image]

fjone, 2022-11-12:
Hi there!
I'm having the same issues with version 1.8.0c running CLN.

There is no option to setup a automatic backup of the sqlite3 file on a external thumbdrive.

Also tried to format a thumbdrive with ext4 but the system don't discover the new thumbdrive in the /mnt directory

cwgriswold, 2022-11-13:
Delete the tram package.

 sudo /home/admin/download/zram-swap/zram-swap.sh stop
 sudo /home/admin/download/zram-swap/install.sh —uninstall

openoms, 2022-11-13:
The fix is part of this PR for the next release: https://github.com/rootzoll/raspiblitz/pull/3393

lazyrudi, 2022-11-14:
Perfect now it works. Will this be fixed in 1.8.1, or do I have to uninstall zram again?

rootzoll, 2022-11-14:
@lazyrudi see changed file https://github.com/rootzoll/raspiblitz/pull/3393/files#diff-6323998d129bbc54941c000be2316e847564ddf42bcfb454559f5e8b78831d9b

its on final testing but it should fix with release of next version (which was renamed from 1.8.1 to 1.9.0) with zram running

lazyrudi, 2022-11-14:
Thanks for the Info !!! 👍🏻Mit freundlichen GrüßenRüdiger PassehlJohann-Schütte-Str. 1271034 BöblingenTel. +49 160 5538767Mail.

closed 2022-12-15 by rootzoll