raspiblitz/raspiblitz #3440: Unintenional rebooting due to typo which leads to script execution

curemici, 2022-11-14:
I made a small typo on CLI.

`lcd..` (instead of cd..) 

and raspiblitz was starting the script `blitz.display.sh `with parameter `on`.

After script execution it was rebooting.

I have not found an alias "lcd" so it is a little strange. Small typo, big reboot.

rootzoll, 2022-11-15:
This is where the `lcd` command comes from:

https://github.com/rootzoll/raspiblitz/blob/6e426871a9c24854e67be0ddf8ccd816296319a0/home.admin/_commands.sh#L185

I can see that its a bit of a tragic typo resulting in bad UX experience .. but the `l` is not in the fat finger area of the `c` when typing `cd` command for example - its quite an edge case ... so I dont think there is a need for a change from my side - but open for ideas if this happens more often to multiple people.

rootzoll, 2022-11-15:
its an idea to better improve for the future against command typos - but I think with that change more people will have a bad UX because suddenly the commands they are already used to (and are namend in doc and old support issues) are not working anymore.

I dont have a high personal priority on this. But I will merge a PR that changes all that commands in code but also in documentation.

rootzoll, 2022-11-15:
> what if a yes no question gets added if such a command is used?

I would not do a Yes/No question to all commands - but for some like the `lcd` that are not often used and change configuration .. it can make sense to add a Yes/No to prevent execution on typo and also make sure the user will get informed what the command will do.

ChuckNorrison, 2022-11-15:
yes, only for commands which will change the system or cause interruption like reboot or power off. I can make a PR with this confirmation message.

rootzoll, 2022-11-26:
PR merged. Looking good - keep open for final testing.

rootzoll, 2022-11-27:
Tests looked good - closing.

closed 2022-11-27 by rootzoll