raspiblitz/raspiblitz #3447: Electrs indexing taking forever

IanMichaelHarper, 2022-11-16:
I had to delete my old electrs index and re-index. I started this last week and its been indexing ever since, but it's taking forever. The index size is only at 52MB now. If it's meant to be 10-20% of the blockchain size, then it seems it will take years to complete.

Is there an issue with my node? Is there a way to speed it up? Has anyone encountered this issue before?

rootzoll, 2022-11-16:
Can you post the result of the terminal command: `/home/admin/config.scripts/bonus.electrs.sh status`

IanMichaelHarper, 2022-11-16:
```
##### STATUS ELECTRS SERVICE
configured=1
serviceInstalled=1
serviceRunning=0
serviceRunning=0
tipSynced=0
initialSynced=0
electrumResponding=0
infoSync='Not running - check: sudo journalctl -u electrs'
```

rootzoll, 2022-11-16:
And please just `/home/admin/config.scripts/bonus.electrs.sh` (to see the version) 

and also `sudo journalctl -u electrs -b --no-pager -n100`

IanMichaelHarper, 2022-11-16:
```
admin@192.168.0.203:~ ₿ /home/admin/config.scripts/bonus.electrs.sh
config script to switch the Electrum Rust Server on or off
bonus.electrs.sh status -> dont call in loops
bonus.electrs.sh status-sync
bonus.electrs.sh [on|off|menu]
installs the version v0.9.9
admin@192.168.0.203:~ ₿ sudo journalctl -u electrs -b --no-pager -n100
-- Journal begins at Wed 2022-11-16 07:07:39 GMT, ends at Wed 2022-11-16 16:56:33 GMT. --
Nov 16 16:48:30 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 16:48:30 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 16 16:49:30 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 11016.
Nov 16 16:49:30 raspberrypi systemd[1]: Stopped Electrs.
Nov 16 16:49:30 raspberrypi systemd[1]: Started Electrs.
Nov 16 16:49:30 raspberrypi electrs[2173559]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.9.9 - the Electrum Rust Server on your RaspiBlitz", args: [] }
Nov 16 16:49:30 raspberrypi electrs[2173559]: [2022-11-16T16:49:30.768Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 16 16:49:30 raspberrypi electrs[2173559]: [2022-11-16T16:49:30.768Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 16 16:49:31 raspberrypi electrs[2173559]: [2022-11-16T16:49:31.563Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 12758 SST files, 0.01027019 GB, 0.000000021 Grows
Nov 16 16:49:31 raspberrypi electrs[2173559]: [2022-11-16T16:49:31.615Z WARN  electrs::p2p] failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)
Nov 16 16:49:31 raspberrypi electrs[2173559]: [2022-11-16T16:49:31.615Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 16 16:49:31 raspberrypi electrs[2173559]: Error: electrs failed
Nov 16 16:49:31 raspberrypi electrs[2173559]: Caused by:
Nov 16 16:49:31 raspberrypi electrs[2173559]:     receiving on an empty and disconnected channel
Nov 16 16:49:31 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 16:49:31 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 16 16:50:31 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 11017.
Nov 16 16:50:31 raspberrypi systemd[1]: Stopped Electrs.
Nov 16 16:50:31 raspberrypi systemd[1]: Started Electrs.
Nov 16 16:50:31 raspberrypi electrs[2181948]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.9.9 - the Electrum Rust Server on your RaspiBlitz", args: [] }
Nov 16 16:50:31 raspberrypi electrs[2181948]: [2022-11-16T16:50:31.777Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 16 16:50:31 raspberrypi electrs[2181948]: [2022-11-16T16:50:31.777Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 16 16:50:32 raspberrypi electrs[2181948]: [2022-11-16T16:50:32.636Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 12759 SST files, 0.010270995 GB, 0.000000021 Grows
Nov 16 16:50:32 raspberrypi electrs[2181948]: [2022-11-16T16:50:32.696Z WARN  electrs::p2p] failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)
Nov 16 16:50:32 raspberrypi electrs[2181948]: [2022-11-16T16:50:32.696Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 16 16:50:32 raspberrypi electrs[2181948]: Error: electrs failed
Nov 16 16:50:32 raspberrypi electrs[2181948]: Caused by:
Nov 16 16:50:32 raspberrypi electrs[2181948]:     receiving on an empty and disconnected channel
Nov 16 16:50:32 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 16:50:32 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 16 16:51:32 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 11018.
Nov 16 16:51:32 raspberrypi systemd[1]: Stopped Electrs.
Nov 16 16:51:32 raspberrypi systemd[1]: Started Electrs.
Nov 16 16:51:33 raspberrypi electrs[2190282]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1 [... cut ...]
Nov 16 16:55:37 raspberrypi electrs[2224118]: [2022-11-16T16:55:37.820Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 16 16:55:37 raspberrypi electrs[2224118]: Error: electrs failed
Nov 16 16:55:37 raspberrypi electrs[2224118]: Caused by:
Nov 16 16:55:37 raspberrypi electrs[2224118]:     receiving on an empty and disconnected channel
Nov 16 16:55:37 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 16:55:37 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
admin@192.168.0.203:~ ₿
```

rootzoll, 2022-11-16:
Constant restart with error `failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)`.

This looks like an eletrs bug - please report in their repo here: https://github.com/romanz/electrs/issues/769

rootzoll, 2022-11-16:
TODO: check for electrs release update before next reaspiblitz RC

Kixunil, 2022-11-17:
I don't think this is electrs bug. I suspect you don't allow electrs connecting to bitcoind which would be a configuration bug on your side.

IanMichaelHarper, 2022-11-17:
posting my `bitcoin.conf` here
```
admin@192.168.0.203:~ ₿ cat assets/bitcoin.conf
# bitcoind configuration
# some values might be overruled directly systemd-service exec call parameters

# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=0
disablewallet=1
peerbloomfilters=1
datadir=/mnt/hdd/bitcoin

# Connection settings
rpcuser=raspibolt
rpcpassword=passwordB
main.rpcport=8332
test.rpcport=18332
rpcallowip=127.0.0.1
main.rpcbind=127.0.0.1:8332
test.rpcbind=127.0.0.1:18332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
bind=0.0.0.0:50001

# Raspberry Pi optimizations
dbcache=128
maxorphantx=10
maxmempool=300
maxconnections=40
maxuploadtarget=5000

# tor by default
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
```

romanz, 2022-11-17:
Please try to set `whitelist=download@127.0.0.1`:
https://github.com/romanz/electrs/blob/master/doc/config.md#bitcoind-configuration

If the issue persists, could you please increase log verbosity (by setting `--log-filters TRACE`) and attach the logs?

IanMichaelHarper, 2022-11-18:
I added `whitelist=download@127.0.0.1` to my `bitcoin.conf` but that doesn't seem to have worked.

Which command do I need to execute with `--log-filters TRACE`?

romanz, 2022-11-19:
> Which command do I need to execute with `--log-filters TRACE`?

`electrs`, which should be configured in `/home/electrs/.electrs/config.toml`:

https://github.com/rootzoll/raspiblitz/blob/15b460098c4efc841a9ee86fdb5bc91c12361e34/home.admin/config.scripts/bonus.electrs.sh#L326

IanMichaelHarper, 2022-11-20:
```
admin@192.168.0.203:~ ₿  /home/electrs/electrs/target/release/electrs --log-filters TRACE
Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "./db/bitcoin", daemon_dir: "/home/admin/.bitcoin", daemon_auth: CookieFile("/home/admin/.bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.9 (Electrum Rust Server)!", args: [] }
[2022-11-20T16:02:01.482Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
[2022-11-20T16:02:01.483Z INFO  electrs::server] serving Electrum RPC on 127.0.0.1:50001
[2022-11-20T16:02:01.483Z DEBUG tiny_http] Server listening on 127.0.0.1:4224
[2022-11-20T16:02:01.487Z DEBUG tiny_http] Running accept thread
[2022-11-20T16:02:01.799Z INFO  electrs::db] "./db/bitcoin": 0 SST files, 0 GB, 0 Grows
[2022-11-20T16:02:01.800Z DEBUG electrs::db] DB None
[2022-11-20T16:02:01.818Z INFO  electrs::db] closing DB at ./db/bitcoin
Error: electrs failed

Caused by:
    0: failed to open bitcoind cookie file: /home/admin/.bitcoin/.cookie
    1: No such file or directory (os error 2)
```

romanz, 2022-11-20:
I think it's better to modify `/home/electrs/.electrs/config.toml` to contain:
```
log_filters="TRACE"
```
instead of:
https://github.com/rootzoll/raspiblitz/blob/15b460098c4efc841a9ee86fdb5bc91c12361e34/home.admin/config.scripts/bonus.electrs.sh#L315

You can the restart `electrs` via:
```
sudo systemctl restart electrs
```

rootzoll, 2022-11-21:
@romanz is that a general recommendation (loglevel TRACE) or just for this one debug case?

rootzoll, 2022-11-21:
@romanz `whitelist=download@127.0.0.1`should already be part of the bitcoind config, as its part of the electrs install script - see: https://github.com/rootzoll/raspiblitz/blob/15b460098c4efc841a9ee86fdb5bc91c12361e34/home.admin/config.scripts/bonus.electrs.sh#L446

IanMichaelHarper, 2022-11-22:
Adding that to the config doesn't change the output of the `electrs` command

```
admin@192.168.0.203:~ ₿ /home/electrs/electrs/target/release/electrs
Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "./db/bitcoin", daemon_dir: "/home/admin/.bitcoin", daemon_auth: CookieFile("/home/admin/.bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.9 (Electrum Rust Server)!", args: [] }
Error: electrs failed

Caused by:
    0: failed to open bitcoind cookie file: /home/admin/.bitcoin/.cookie
    1: No such file or directory (os error 2)
```

Kixunil, 2022-11-24:
@IanMichaelHarper I suspect your `bitcoind` died. Unless you changed your configuration to have cookie file somewhere else.

IanMichaelHarper, 2022-11-25:
My `bitcoind` is still running and I never altered the config file to change the cookie location.

romanz, 2022-11-25:
Please note that [running electrs from command-line](https://github.com/rootzoll/raspiblitz/issues/3447#issuecomment-1323525920) seems to use a different config:
```
Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "./db/bitcoin", daemon_dir: "/home/admin/.bitcoin", daemon_auth: CookieFile("/home/admin/.bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.9 (Electrum Rust Server)!", args: [] }
```
compared to [running it using systemd](https://github.com/rootzoll/raspiblitz/issues/3447#issuecomment-1317348715):
```
Nov 16 16:55:37 raspberrypi electrs[2224118]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.9.9 - the Electrum Rust Server on your RaspiBlitz", args: [] }

```

IanMichaelHarper, 2022-11-27:
```
admin@192.168.0.203:/home/electrs/.electrs ₿ sudo journalctl -u electrs -b --no-pager -n100
-- Journal begins at Sun 2022-11-27 13:52:11 GMT, ends at Sun 2022-11-27 22:20:02 GMT. --
Nov 27 22:16:56 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 27 22:16:56 raspberrypi systemd[1]: electrs.service: Consumed 1.003s CPU time.
Nov 27 22:17:56 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 38.
Nov 27 22:17:56 raspberrypi systemd[1]: Stopped Electrs.
Nov 27 22:17:56 raspberrypi systemd[1]: electrs.service: Consumed 1.003s CPU time.
Nov 27 22:17:56 raspberrypi systemd[1]: Started Electrs.
Nov 27 22:17:56 raspberrypi electrs[2363564]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.9.9 - the Electrum Rust Server on your RaspiBlitz", args: [] }
Nov 27 22:17:56 raspberrypi electrs[2363564]: [2022-11-27T22:17:56.925Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 27 22:17:56 raspberrypi electrs[2363564]: [2022-11-27T22:17:56.925Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 27 22:17:56 raspberrypi electrs[2363564]: [2022-11-27T22:17:56.925Z DEBUG tiny_http] Server listening on 127.0.0.1:4224
Nov 27 22:17:56 raspberrypi electrs[2363564]: [2022-11-27T22:17:56.926Z DEBUG tiny_http] Running accept thread
Nov 27 22:17:57 raspberrypi electrs[2363564]: [2022-11-27T22:17:57.405Z DEBUG electrs::server] 0: recv {"jsonrpc":"2.0","method":"server.version","params":["btc-rpc-explorer-v2","1.4"],"id":5}
Nov 27 22:17:57 raspberrypi electrs[2363564]: [2022-11-27T22:17:57.859Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 18382 SST files, 0.01479751 GB, 0.000000021 Grows
Nov 27 22:17:57 raspberrypi electrs[2363564]: [2022-11-27T22:17:57.869Z DEBUG electrs::db] DB Some(Config { compacted: false, format: 0 })
Nov 27 22:17:57 raspberrypi electrs[2363564]: [2022-11-27T22:17:57.913Z DEBUG bitcoincore_rpc] JSON-RPC request: getblockchaininfo []
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.037Z TRACE bitcoincore_rpc] JSON-RPC response for getblockchaininfo: {"chain":"main","blocks":764955,"headers":764955,"bestblockhash":"00000000000000000004a8bb673a583bfeccb92d93dc785818877ed04b7502ed","difficulty":36950494067222.41,"time":1669587252,"mediantime":1669581155,"verificationprogress":0.9999991662898858,"initialblockdownload":false,"chainwork":"00000000000000000000000000000000000000003a5484a4e76e55e1b162ba5c","size_on_disk":500365234120,"pruned":false,"warnings":""}
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.037Z DEBUG bitcoincore_rpc] JSON-RPC request: getnetworkinfo []
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.039Z TRACE bitcoincore_rpc] JSON-RPC response for getnetworkinfo: {"version":230000,"subversion":"/Satoshi:23.0.0/","protocolversion":70016,"localservices":"000000000000040d","localservicesnames":["NETWORK","BLOOM","WITNESS","NETWORK_LIMITED"],"localrelay":true,"timeoffset":-1,"networkactive":true,"connections":10,"connections_in":0,"connections_out":10,"networks":[{"name":"ipv4","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"ipv6","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"onion","limited":false,"reachable":true,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"i2p","limited":true,"reachable":false,"proxy":"","proxy_randomize_credentials":false},{"name":"cjdns","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true}],"relayfee":0.00001000,"incrementalfee":0.00001000,"localaddresses":[{"address":"4b5ssswuoaomkds7ux5j7tjj55mcdr7lyjwgax7fcoepzw5y3fn34vad.onion","port":8333,"score":4}],"warnings":""}
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.039Z DEBUG bitcoincore_rpc] JSON-RPC request: getnetworkinfo []
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.041Z TRACE bitcoincore_rpc] JSON-RPC response for getnetworkinfo: {"version":230000,"subversion":"/Satoshi:23.0.0/","protocolversion":70016,"localservices":"000000000000040d","localservicesnames":["NETWORK","BLOOM","WITNESS","NETWORK_LIMITED"],"localrelay":true,"timeoffset":-1,"networkactive":true,"connections":10,"connections_in":0,"connections_out":10,"networks":[{"name":"ipv4","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"ipv6","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"onion","limited":false,"reachable":true,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"i2p","limited":true,"reachable":false,"proxy":"","proxy_randomize_credentials":false},{"name":"cjdns","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true}],"relayfee":0.00001000,"incrementalfee [... cut ...]
Nov 27 22:20:00 raspberrypi electrs[2380942]: Error: electrs failed
Nov 27 22:20:00 raspberrypi electrs[2380942]: Caused by:
Nov 27 22:20:00 raspberrypi electrs[2380942]:     receiving on an empty and disconnected channel
Nov 27 22:20:00 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 27 22:20:00 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 27 22:20:00 raspberrypi systemd[1]: electrs.service: Consumed 1.047s CPU time.
```

romanz, 2022-11-28:
Seems like bitcoind disconnects from electrs:
```
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.047Z DEBUG electrs::p2p] closing p2p_loop thread: peer has disconnected
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.047Z DEBUG electrs::p2p] closing p2p_recv thread: connection closed
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.047Z TRACE electrs::p2p] send: Version(VersionMessage { version: 70001, services: ServiceFlags(0), timestamp: 1669587478, receiver: Address {services: ServiceFlags(NONE), address: 0.0.0.0, port: 0}, sender: Address {services: ServiceFlags(NONE), address: 0.0.0.0, port: 0}, nonce: 560359173648027389, user_agent: "/electrs:0.9.9/", start_height: 0, relay: false })
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.058Z DEBUG electrs::p2p] closing p2p_send thread: no more messages to send
Nov 27 22:17:58 raspberrypi electrs[2363564]: [2022-11-27T22:17:58.058Z WARN  electrs::p2p] failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)
Nov 27 22:18:59 raspberrypi electrs[2372287]: [2022-11-27T22:18:59.251Z DEBUG electrs::p2p] closing p2p_loop thread: peer has disconnected
Nov 27 22:18:59 raspberrypi electrs[2372287]: [2022-11-27T22:18:59.251Z DEBUG electrs::p2p] closing p2p_recv thread: connection closed
Nov 27 22:18:59 raspberrypi electrs[2372287]: [2022-11-27T22:18:59.251Z DEBUG electrs::p2p] closing p2p_send thread: no more messages to send
Nov 27 22:18:59 raspberrypi electrs[2372287]: [2022-11-27T22:18:59.251Z TRACE electrs::p2p] send: Version(VersionMessage { version: 70001, services: ServiceFlags(0), timestamp: 1669587539, receiver: Address {services: ServiceFlags(NONE), address: 0.0.0.0, port: 0}, sender: Address {services: ServiceFlags(NONE), address: 0.0.0.0, port: 0}, nonce: 14330661110205509539, user_agent: "/electrs:0.9.9/", start_height: 0, relay: false })
Nov 27 22:18:59 raspberrypi electrs[2372287]: [2022-11-27T22:18:59.251Z WARN  electrs::p2p] failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)
Nov 27 22:20:00 raspberrypi electrs[2380942]: [2022-11-27T22:20:00.602Z DEBUG electrs::p2p] closing p2p_recv thread: connection closed
Nov 27 22:20:00 raspberrypi electrs[2380942]: [2022-11-27T22:20:00.602Z TRACE electrs::p2p] send: Version(VersionMessage { version: 70001, services: ServiceFlags(0), timestamp: 1669587600, receiver: Address {services: ServiceFlags(NONE), address: 0.0.0.0, port: 0}, sender: Address {services: ServiceFlags(NONE), address: 0.0.0.0, port: 0}, nonce: 11125353345037788140, user_agent: "/electrs:0.9.9/", start_height: 0, relay: false })
Nov 27 22:20:00 raspberrypi electrs[2380942]: [2022-11-27T22:20:00.608Z DEBUG electrs::p2p] closing p2p_loop thread: peer has disconnected
Nov 27 22:20:00 raspberrypi electrs[2380942]: [2022-11-27T22:20:00.609Z DEBUG electrs::p2p] closing p2p_send thread: no more messages to send
Nov 27 22:20:00 raspberrypi electrs[2380942]: [2022-11-27T22:20:00.609Z WARN  electrs::p2p] failed to shutdown p2p connection: Transport endpoint is not connected (os error 107)
```

romanz, 2022-11-28:
Could you please add a `debug=net` line to your `bitcoin.conf`, and search for `electrs` in bitcoind logs?
It should look like:
```
[net] receive version message: /electrs:0.9.10/: version 70001, blocks=0, us=0.0.0.0:0, txrelay=0, peer=2
```
Then, please filter the relevant network debug messages from electrs by filtering over `peer=2` - and attach the result here :)

romanz, 2022-11-28:
> I have `maxuploadlimit=5` and this worked last year with electrs. Could something has changed in the last year so that now it needs to be higher?

https://github.com/romanz/electrs/issues/399#issuecomment-843640380 may be relevant.

IanMichaelHarper, 2022-11-28:
> > I have `maxuploadlimit=5` and this worked last year with electrs. Could something has changed in the last year so that now it needs to be higher?
> 
> [romanz/electrs#399 (comment)](https://github.com/romanz/electrs/issues/399#issuecomment-843640380) & [romanz/electrs#429 (comment)](https://github.com/romanz/electrs/issues/429#issuecomment-882090579) may be relevant.

Okay, looks like that fixed it. I commented out `maxuploadlimit` and it worked. Its building the index now. Thanks for the help

IanMichaelHarper, 2022-12-03:
The index finished building and I can now lookup address balances on my local mempool.space.

Closing this issue now as it's resolved.

closed 2022-12-03 by IanMichaelHarper