raspiblitz/raspiblitz #3449: lightning wallet locked

ShogunWarlord, 2022-11-16:
[image]

my node is failing to unlock the lightning wallet, any idea how to fix this?

thanks in advance

rootzoll, 2022-11-16:
On the terminal - can you simply try: `lncli unlock`?

ShogunWarlord, 2022-11-16:
output:
[image]

rootzoll, 2022-11-16:
Please restart your raspiblitz one and then run from command line `menu` and go `CONNECT` > `RESET` (recreate LND Macaroons & TLS certs)

ShogunWarlord, 2022-11-16:
[image]
there is no reset option :(

rootzoll, 2022-11-16:
Ah OK - your using core lightning (not LND) ..  @openoms can you take a look why CLN wallet does not unlock?

openoms, 2022-11-16:
The problem is as in the in the OP,  lightningd fails with a Tor error.
@ShogunWarlord can you enter the CLN menu again and post if the error is the same after restart?

ShogunWarlord, 2022-11-17:
[image]
wallet unlocked, thank you guys!

closed 2022-11-17 by rootzoll