raspiblitz/raspiblitz #3454: Setup Raspiblitz: state(error) message(tbc-testrun-fail)

TonyGranato, 2022-11-18:
Dear Raspiblitz-Team,

I ordered a pre-synced RaspiBlitz 1.8.0 (8GB) and tried to run the setup. Unfortunately I run into problems. There is always the same error: 

***********************
ERROR -Please report to development team
***********************
state(error) message(tbc-testrun-fail)

First I did restart the raspiblitz and got the information to refresh the SD-Card. I did this with the  balenaEtcher-Tool and was able to start the setup again. I was asked whether I want to use the pre-synced blockchain (I did yes) and run into the same error. Then I read that someone was able to run the setup again using a different Imager (raspberry Pi), but I run into the same error. So I decided to sync the blockchain from scratch, but I run into the same error either. 

Reading the debug-file I can see, that I run into a Fatal LevelDB error. This was the decision to run the sync from scratch in the first place. 

Can someone give me further assistance to solve the problem or at least give me a hint? 

Thanks for your support

Debug-file attached:
https://l.termbin.com/uv69n

Best
T.

TonyGranato, 2022-11-20:
I could solve the problem, but it  is probably a systematic failure at the assembled and pre-synced Raspiblitz. I was looking for further assistance at the english RaspiBlitz-Telegram Group where someone recently had the same problem. 

His solution was formatting the installed SSD and run the setup from scratch, i.e. building the bitcoin blockchain from scratch. His solution worked for me either. 

Best
T.

rootzoll, 2022-11-22:
@TonyGranato so if the pre-synced blockchain from FULMO didnt worked for you - please get in contact with the sales support (link this issue) and request a refund for the pre-sync.

closed 2022-11-22 by rootzoll