raspiblitz/raspiblitz #3460: Problem with mempool space, addresses do not load

FlorianHubl, 2022-11-20:
Im not a techy so I need a good explanation. When I go to mempool space on my raspiblitz and click one a adress the adress do not load. After some time I get a timeout. I have electrs 0.9.9. Any ideas how to fix this?
[image]

rootzoll, 2022-11-21:
TODO: retest with v1.8.0c image

FlorianHubl, 2022-11-22:
Thank you. I had to do it with v1.8.0c, then install electrs 0.9.10 and then restart electrs and mempool.
cd /home/electrs/electrs
sudo -u electrs git fetch
sudo -u electrs git reset --hard v0.9.10
sudo -u electrs /home/electrs/.cargo/bin/cargo build --locked --release
(*Wait till this completes before executing the system restart*)
sudo systemctl restart electrs
sudo systemctl restart mempool

mitohund, 2022-11-25:
I know this is closed, but since it also says "bug - unconfirmed" I thought I'd just chime in to say that I also have this issue.

rootzoll, 2022-11-26:
I see if I can give this another check during release candidate phase

mitohund, 2022-11-27:
Don't know if it helps, but here's what happened for me since my last message (I'm on 1.8.0c): I reinstalled electrs (keeping the index), but mempool still would not load addresses (everything else though). An hour or so later, it started loading addresses with a 1-2 minute loading time. Another day later, they seem to be loading fine. I had checked htop load averages at the time and they were fine.

The thing is, I had only updated to 1.8.0 maybe two weeks ago, and at the time I had reinstalled both mempool and electrs.

rootzoll, 2022-12-15:
I was able to reproduce it shortly - with diffrent error messages (did also not seem cossitent) but now it works again on my test machines. Maybe this is electrs related - tested with eletcrs 0.9.10 which will be part of RaspiBlitz v1.9.0 release. So I will push this up - hope this gets more stable with future electrs versions. To be further monitored.

mitohund, 2022-12-19:
I believe I read somewhere in the mempool.space docs that electrs might be a bottleneck and that fulcrum is recommended? I also read that address loading behaviour depends on the number of transactions at that address.

I think I can reproduce that. Addresses with one transaction load pretty instantly. Addresses with a few thousand transactions give an error. In between, high loading times. I'm on electrs 0.9.10 now (I think...).

malt33k, 2022-12-26:
Running raspiblitz 1.8.0 and having the same issue. Do not have electrum rust server enabled, does this have to be a requirement? Thought setting txindex=1 would include the address information.
[image]

mitohund, 2023-01-09:
Since I just stumbled upon this again, here's an exemplary (totally random) address that always times out on my Raspiblitz: https://mempool.space/de/address/bc1q9s2kmdhn8xt0qjuks0faetq76ycq7zvq4jdfwf

myxmaster, 2023-01-22:
Like @malt33k I get the same error message immediately for any address I try to look up. I installed mempool a few days ago, so index building is done for sure.

So I also want to know: Is the electrum rust server a requirement? If yes, users should get notified in case mempool is installed without electrum rust server. If no, what can I do to fix it?

mitohund, 2023-01-23:
I believe it's required. Why don't you just try and install it?

It's true though that if it is required, there should be some kind of notification of dependencies.

myxmaster, 2023-01-23:
Because of the weak hardware I don't want to install more than needed.

Anyway, I read more about it and it seems that, to look up addresses, electrum rust server as backend is needed.
-> When a user triggers mempool installation, it should be checked if electrum rust server is installed already, if not it should inform about the limited functionality and offer to install electrum rust server first.

If you install electrum rust server later, like I did, you not only need to wait another multiple hours for indexing, but also restart the pi, otherwise mempool won't talk to electrum rust server. (Maybe only mempool service restart would have been sufficient, but I couldn't figure out easily how to do that.)

mitohund, 2023-01-24:
fwiw I run a RPi4 and have not run into any performance issues at all. And at times I had most services running concurrently (apart from the additional ones for Lightning, since I don't use Lightning).

myxmaster, 2023-01-24:
Ok good to know, thank you!

ExchangeREC, 2023-01-26:
> Since I just stumbled upon this again, here's an exemplary (totally random) address that always times out on my Raspiblitz: https://mempool.space/de/address/bc1q9s2kmdhn8xt0qjuks0faetq76ycq7zvq4jdfwf

I can confirm this for the mentioned example address as well! Others with not that much transaction work. Using 1.8.0.c image with Electrs and Mempool.space

myxmaster, 2023-01-30:
@mitohund 
I guess it's totally normal to run into problems if you lookup addresses with lots of transactions on weak hardware, even if you use lightweight electrum server (electrs). If it is important for you, you might try fulcrum as backend, see this comment: https://github.com/mempool/mempool/issues/1612#issuecomment-1120346897

rootzoll, 2024-04-23:
@openoms recheck if still relevant

closed 2024-06-10 by rootzoll

chuegel, 2025-02-01:
I'm also facing this issue with:
Raspiblitz  1.11.1
Fulcrum 1.10.0

installed on a Pi4 with 8GB RAM

rootzoll, 2025-02-03:
@openoms was mempool+fulcrum ever tested?