raspiblitz/raspiblitz #3463: Electrs won't start after udpate

datvebi, 2022-11-21:
I updated to v1.8.0c but electrs isn't running. I've restarted the node and I've run sudo systemctl restart electrs to no effect. Can't exactly figure out what I should be doing from the logs

Nov 21 01:05:45 raspberrypi systemd[1]: Started Electrs.
Nov 21 01:05:45 raspberrypi electrs[50449]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_ad>
Nov 21 01:05:45 raspberrypi electrs[50449]: [2022-11-21T01:05:45.286Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 21 01:05:45 raspberrypi electrs[50449]: [2022-11-21T01:05:45.286Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 21 01:05:45 raspberrypi electrs[50449]: [2022-11-21T01:05:45.373Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 157 SST files, 37.605140975 GB, 4.69618748 Grows
Nov 21 01:05:51 raspberrypi electrs[50449]: [2022-11-21T01:05:51.693Z INFO  electrs::chain] loading 761248 headers, tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f
Nov 21 01:05:57 raspberrypi electrs[50449]: [2022-11-21T01:05:57.645Z INFO  electrs::chain] chain updated: tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f, height=761248
Nov 21 01:05:57 raspberrypi electrs[50449]: [2022-11-21T01:05:57.777Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 21 01:05:57 raspberrypi electrs[50449]: Error: electrs failed
Nov 21 01:05:57 raspberrypi electrs[50449]: Caused by:
Nov 21 01:05:57 raspberrypi electrs[50449]:     0: bitcoin p2p failed to connect: 127.0.0.1:8333
Nov 21 01:05:57 raspberrypi electrs[50449]:     1: Connection refused (os error 111)
Nov 21 01:05:57 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 01:05:57 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 21 01:05:57 raspberrypi systemd[1]: electrs.service: Consumed 19.409s CPU time.
Nov 21 01:06:57 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 3.
Nov 21 01:06:57 raspberrypi systemd[1]: Stopped Electrs.
Nov 21 01:06:57 raspberrypi systemd[1]: electrs.service: Consumed 19.409s CPU time.
Nov 21 01:06:57 raspberrypi systemd[1]: Started Electrs.
Nov 21 01:06:58 raspberrypi electrs[57473]: Starting electrs 0.9.9 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_ad>
Nov 21 01:06:58 raspberrypi electrs[57473]: [2022-11-21T01:06:58.022Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Nov 21 01:06:58 raspberrypi electrs[57473]: [2022-11-21T01:06:58.022Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Nov 21 01:06:59 raspberrypi electrs[57473]: [2022-11-21T01:06:59.601Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 158 SST files, 37.605141779 GB, 4.696187481 Grows
Nov 21 01:07:05 raspberrypi electrs[57473]: [2022-11-21T01:07:05.701Z INFO  electrs::chain] loading 761248 headers, tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f
Nov 21 01:07:10 raspberrypi electrs[57473]: [2022-11-21T01:07:10.260Z INFO  electrs::chain] chain updated: tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f, height=761248
Nov 21 01:07:47 raspberrypi electrs[57473]: [2022-11-21T01:07:47.743Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Nov 21 01:07:47 raspberrypi electrs[57473]: Error: electrs failed
Nov 21 01:07:47 raspberrypi electrs[57473]: Caused by:
Nov 21 01:07:47 raspberrypi electrs[57473]:     0: bitcoind RPC polling failed
Nov 21 01:07:47 raspberrypi electrs[57473]:     1: daemon not available
Nov 21 01:07:47 raspberrypi electrs[57473]:     2: JSON-RPC error: transport error: Didn't receive response data in time, timed out.
Nov 21 01:07:47 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 01:07:47 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.

ChuckNorrison, 2022-11-21:
Your electrs can not access RPC of your bitcoind. Maybe wrong Password B in the electrs config?

you can check your electrs config file here

`sudo cat /home/electrs/.electrs/config.toml`

Maybe check your bitcoind logs

`sudo journalctl -fu bitcoind -n 200`

or check your open ports for 8333

`netstat -tupl | grep 8333`

ChuckNorrison, 2022-11-21:
i think your bitcoind is not running. Check again with

`sudo systemctl status bitcoind`

If your bitcoind is not running, you should reflash your sd card

datvebi, 2022-11-21:
Looks like it's running:

bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-11-21 05:37:00 GMT; 12h ago
    Process: 5924 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 5941 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 5945 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6017 (bitcoind)
      Tasks: 17 (limit: 4531)
        CPU: 55min 18.630s
     CGroup: /system.slice/bitcoind.service
             └─6017 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Nov 21 05:33:48 raspberrypi systemd[1]: Starting Bitcoin daemon...
Nov 21 05:33:48 raspberrypi bitcoind[5945]: Warning: /mnt/hdd/bitcoin/bitcoin.conf:31 Section [main.main] is not recognized.
Nov 21 05:33:48 raspberrypi bitcoind[5945]: /mnt/hdd/bitcoin/bitcoin.conf:32 Section [main.main] is not recognized.
Nov 21 05:33:48 raspberrypi bitcoind[5945]: /mnt/hdd/bitcoin/bitcoin.conf:38 Section [main.main] is not recognized.
Nov 21 05:33:48 raspberrypi bitcoind[5945]: /mnt/hdd/bitcoin/bitcoin.conf:39 Section [main.test] is not recognized.
Nov 21 05:37:00 raspberrypi systemd[1]: Started Bitcoin daemon.

ChuckNorrison, 2022-11-21:
> Looks like it's running:
>
> bitcoind.service - Bitcoin daemon Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled) Active: active (running) since Mon 2022-11-21 [...]

think something is wrong with the config file... you should reflash your sd card or take a look into the bitcoind config file

datvebi, 2022-11-22:
Alright. This will be the third reflash but I'll do it and update here. First I'll check the config file though.

datvebi, 2022-11-22:
Re-flash didn't work. Electrs is still not running. Any other ideas? Where is the config file and what should I be looking for?

ChuckNorrison, 2022-11-23:
Ok, so its definitely an error on your existent ssd installation of electrs. Remove and install again should fix it. Maybe switch to the raspiblitz dev branch and update electrs to v0.9.10.

datvebi, 2022-11-26:
So I flashed the dev branch and ran commands from other tickets to update lnd and electrs. No change. I then deleted electrs (but kept the index) and re-installed. Still not running. I don't know what to do from here other than abandon my node and install from scratch, no recovery. Any other ideas? 

For what it's worth, here's the output from sudo journalctl -u electrs

-- Journal begins at Wed 2022-07-20 12:10:51 BST, ends at Sat 2022-11-26 18:18:10 GMT. --
Nov 26 04:58:41 raspberrypi systemd[1]: Started Electrs.
Nov 26 04:58:41 raspberrypi electrs[28236]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_>Nov 26 04:58:41 raspberrypi electrs[28236]: [2022-11-26T04:58:41.845Z INFO  electrs::metrics::metrics_impl] serving Pro>Nov 26 04:58:41 raspberrypi electrs[28236]: [2022-11-26T04:58:41.845Z INFO  electrs::server] serving Electrum RPC on 0.>Nov 26 04:58:46 raspberrypi electrs[28236]: [2022-11-26T04:58:46.865Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/>Nov 26 04:58:55 raspberrypi electrs[28236]: [2022-11-26T04:58:55.205Z INFO  electrs::chain] loading 761248 headers, tip>Nov 26 04:59:00 raspberrypi electrs[28236]: [2022-11-26T04:59:00.064Z INFO  electrs::chain] chain updated: tip=00000000>Nov 26 04:59:04 raspberrypi electrs[28236]: [2022-11-26T04:59:04.620Z INFO  electrs::db] closing DB at /mnt/hdd/app-sto>Nov 26 04:59:04 raspberrypi electrs[28236]: Error: electrs failed
Nov 26 04:59:04 raspberrypi electrs[28236]: Caused by:
Nov 26 04:59:04 raspberrypi electrs[28236]:     0: bitcoin p2p failed to connect: 127.0.0.1:8333
Nov 26 04:59:04 raspberrypi electrs[28236]:     1: Connection refused (os error 111)
Nov 26 04:59:04 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 26 04:59:04 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 26 04:59:04 raspberrypi systemd[1]: electrs.service: Consumed 23.242s CPU time.
Nov 26 05:00:04 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 1.
Nov 26 05:00:04 raspberrypi systemd[1]: Stopped Electrs.
Nov 26 05:00:04 raspberrypi systemd[1]: electrs.service: Consumed 23.242s CPU time.
Nov 26 05:00:04 raspberrypi systemd[1]: Started Electrs.
Nov 26 05:00:04 raspberrypi electrs[36810]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_>Nov 26 05:00:04 raspberrypi electrs[36810]: [2022-11-26T05:00:04.949Z INFO  electrs::metrics::metrics_impl] serving Pro>Nov 26 05:00:04 raspberrypi electrs[36810]: [2022-11-26T05:00:04.950Z INFO  electrs::server] serving Electrum RPC on 0.>Nov 26 05:00:06 raspberrypi electrs[36810]: [2022-11-26T05:00:06.804Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/>Nov 26 05:00:14 raspberrypi electrs[36810]: [2022-11-26T05:00:14.837Z INFO  electrs::chain] loading 761248 headers, tip>Nov 26 05:00:20 raspberrypi electrs[36810]: [2022-11-26T05:00:20.237Z INFO  electrs::chain] chain updated: tip=00000000>Nov 26 05:00:21 raspberrypi electrs[36810]: [2022-11-26T05:00:21.021Z INFO  electrs::db] closing DB at /mnt/hdd/app-sto>Nov 26 05:00:21 raspberrypi electrs[36810]: Error: electrs failed
Nov 26 05:00:21 raspberrypi electrs[36810]: Caused by:
Nov 26 05:00:21 raspberrypi electrs[36810]:     0: bitcoin p2p failed to connect: 127.0.0.1:8333
Nov 26 05:00:21 raspberrypi electrs[36810]:     1: Connection refused (os error 111)
Nov 26 05:00:21 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 26 05:00:21 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 26 05:00:21 raspberrypi systemd[1]: electrs.service: Consumed 20.721s CPU time.
Nov 26 05:01:21 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 2.
Nov 26 05:01:21 raspberrypi systemd[1]: Stopped Electrs.
Nov 26 05:01:21 raspberrypi systemd[1]: electrs.service: Consumed 20.721s CPU time.
Nov 26 05:01:21 raspberrypi systemd[1]: Started Electrs.
Nov 26 05:01:21 raspberrypi electrs[44576]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_>Nov 26 05:01:21 raspberrypi electrs[44576]: [2022-11-26T05:01:21.273Z INFO  electrs::metrics::metrics_impl] serving Pro>Nov 26 05:01:21 raspberrypi electrs[44576]: [2022-11-26T05:01:21.273Z INFO  electrs::server] serving Electrum RPC on 0.>Nov 26 05:01:21 raspberrypi electrs[44576]: [2022-11-26T05:01:21.351Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/>Nov 26 05:01:26 raspberrypi electrs[44576]: [2022-11-26T05:01:26.425Z INFO  electrs::chain] loading 761248 headers, tip>Nov 26 05:01:31 raspberrypi electrs[44576]: [2022-11-26T05:01:31.052Z INFO  electrs::chain] chain updated: tip=00000000>Nov 26 05:01:31 raspberrypi electrs[44576]: [2022-11-26T05:01:31.099Z INFO  electrs::db] closing DB at /mnt/hdd/app-sto>Nov 26 05:01:31 raspberrypi electrs[44576]: Error: electrs failed
Nov 26 05:01:31 raspberrypi electrs[44576]: Caused by:
Nov 26 05:01:31 raspberrypi electrs[44576]:     0: bitcoin p2p failed to connect: 127.0.0.1:8333
Nov 26 05:01:31 raspberrypi electrs[44576]:     1: Connection refused (os error 111)
lines 1-49...skipping...
-- Journal begins at Wed 2022-07-20 12:10:51 BST, ends at Sat 2022-11-26 18:18:10 GMT. --
Nov 26 04:58:41 raspberrypi systemd[1]: Started Electrs.
Nov 26 04:58:41 raspberrypi electrs[28236]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daem [... cut ...]
Nov 26 05:02:41 raspberrypi electrs[53005]:     0: bitcoin p2p failed to connect: 127.0.0.1:8333
Nov 26 05:02:41 raspberrypi electrs[53005]:     1: Connection refused (os error 111)
Nov 26 05:02:41 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Nov 26 05:02:41 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Nov 26 05:02:41 raspberrypi systemd[1]: electrs.service: Consumed 18.523s CPU time.
Nov 26 05:03:41 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 4.

datvebi, 2022-11-28:
Update: I just deleted the config files from my ssd and flashed anew and built from scratch. No recovery, new everything. Installed electrs and indexed, it's now running and can be connected to. Not sure what went wrong with my node above, but if all else fails, starting from the beginning (but keeping the blockchain on ssd) is a not too long way of getting past it.

closed 2022-11-28 by datvebi