raspiblitz/raspiblitz #3465: Raspiblitz LND Error - elpasobitcoin

357384N, 2022-11-21:
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 16:40:35 up 1 day, 10:24,  3 users,  load average: 1.49, 2.05, 2.82

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-11-20 06:21:45 GMT; 1 day 10h ago
    Process: 6560 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6577 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6581 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6736 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 4h 7min 36.921s
     CGroup: /system.slice/bitcoind.service
             └─6736 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Nov 20 06:18:12 raspberrypi systemd[1]: Starting Bitcoin daemon...
Nov 20 06:21:45 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Sat 2022-11-19 20:02:28 GMT, ends at Mon 2022-11-21 16:40:35 GMT. --
Nov 20 06:18:12 raspberrypi systemd[1]: Starting Bitcoin daemon...
Nov 20 06:21:45 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-11-21T14:38:48Z New outbound peer connected: version: 70016, blocks=764149, peer=1333 (outbound-full-relay)
2022-11-21T14:41:23Z UpdateTip: new best=000000000000000000076127fd05551a62da91ea6854b08a0c5dae1cb79459b2 height=764150 version=0x3fff0000 log2_work=93.855924 tx=782095897 date='2022-11-21T14:38:39Z' progress=0.999999 cache=58.2MiB(382452txo)
2022-11-21T14:41:48Z New outbound peer connected: version: 70016, blocks=764150, peer=1337 (block-relay-only)
2022-11-21T14:58:37Z UpdateTip: new best=00000000000000000000f52e421f97d21e0a58b507d977cd64dc344b5f94ec8b height=764151 version=0x2e39c000 log2_work=93.855936 tx=782099198 date='2022-11-21T14:53:36Z' progress=0.999999 cache=60.3MiB(399819txo)
2022-11-21T15:00:57Z New outbound peer connected: version: 70016, blocks=764151, peer=1355 (block-relay-only)
2022-11-21T15:08:43Z UpdateTip: new best=00000000000000000004e962c1a0fc6a201d937bf08ffe4b1221e956615c7cd9 height=764152 version=0x2c81c000 log2_work=93.855949 tx=782102271 date='2022-11-21T15:07:24Z' progress=1.000000 cache=61.7MiB(411197txo)
2022-11-21T15:27:15Z New outbound peer connected: version: 70016, blocks=764152, peer=1374 (block-relay-only)
2022-11-21T15:31:35Z New outbound peer connected: version: 70016, blocks=764152, peer=1381 (block-relay-only)
2022-11-21T15:34:29Z UpdateTip: new best=00000000000000000000199349a95526c4f83959f0ef06697048a297f25e7fac height=764153 version=0x32950000 log2_work=93.855962 tx=782105142 date='2022-11-21T15:33:32Z' progress=1.000000 cache=64.2MiB(430838txo)
2022-11-21T15:38:19Z New outbound peer connected: version: 70016, blocks=764153, peer=1385 (outbound-full-relay)
2022-11-21T15:47:52Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-11-21T15:48:20Z New outbound peer connected: version: 70016, blocks=764153, peer=1392 (outbound-full-relay)
2022-11-21T16:14:34Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2405 seconds ago)
2022-11-21T16:14:57Z New outbound peer connected: version: 70016, blocks=764153, peer=1405 (outbound-full-relay)
2022-11-21T16:17:09Z New outbound peer connected: version: 70016, blocks=764153, peer=1408 (block-relay-only)
2022-11-21T16:23:38Z New outbound peer connected: version: 70016, blocks=764153, peer=1415 (outbound-full-relay)
2022-11-21T16:24:45Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-11-21T16:24:48Z Potential stale tip detected, will try using extra outbound peer (last tip update: 3019 seconds ago)
2022-11-21T16:27:01Z New outbound peer connected: version: 70016, blocks=764153, peer=1419 (outbound-full-relay)
2022-11-21T16:33:02Z UpdateTip: new best=0000000000000000000790afcaeb4c1aa542b36ae14227298732cce26d0a0bbc height=764154 version=0x2bf1c000 log2_work=93.855975 tx=782107965 date='2022-11-21T16:32:46Z' progress=1.000000 cache=68.7MiB(467602txo)

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Mon 2022-11-21 07:06:54 GMT; 9h ago
    Process: 3350735 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 3351164 ExecStart=/usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf (code=exited, status=0/SUCCESS)
   Main PID: 3351164 (code=exited, status=0/SUCCESS)
     Status: "Wallet unlocked"
        CPU: 3h 6min 45.254s

[... 345 lines cut ...]
ln_default_wallet_channels_pending="0"
ln_default_online="0"
ln_default_channels_total="3"
ln_default_version="0.15.4-beta"
ln_default_error_full=""
ln_default_fees_total="0"
btc_default_peers="14"
btc_default_online="1"
btc_default_mempool_transactions="17451"
btc_default_version="v23.0.0"
btc_default_sync_progress="0.9999983439854043"
btc_default_ready="1"
btc_default_sync_initial_done="1"
btc_default_error_short=""
btc_default_sync_percentage="100.00"
btc_default_blocks_headers="764154"
btc_default_running="1"
btc_default_activated="1"
btc_default_blocks_verified="764154"
btc_default_error_full=""
btc_default_blocks_data_kb="488680880"
btc_default_synced="1"
btc_default_blocks_behind="0"
btc_default_sync_initialblockdownload="0"
btc_default_port="8333"

elpasobitcoin, 2022-11-21:
2022-11-21 17:32:29.434 [INF] NTFN: Dispatching historical spend rescan for outpoint=844853510330eb4e0aff5bddd2f9b6b9b989d9f141046940ef2999867071c020:1, script=0 593f54fee8751d033cb40eabe1b881b77f352459ca161e07e8cb72a0d8fd3d56, start=764107, end=764158
2022-11-21 17:32:29.749 [DBG] NTFN: Updated spend hint to height=764158 for unconfirmed spend request outpoint=c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0, script=0 127bcca2e7b7d24ffe289c896d9c61ffe75ec3caf74feb835777360f7ce7f0f0
2022-11-21 17:32:29.749 [INF] CNCT: Close observer for ChannelPoint(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0) active
2022-11-21 17:32:29.882 [DBG] NTFN: Updated spend hint to height=764158 for unconfirmed spend request outpoint=8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1, script=0 3ff1d5cd1735e08a82b027ba5d52f3c606fca9d853e259f28b688db1f48b83c5
2022-11-21 17:32:29.883 [INF] CNCT: Close observer for ChannelPoint(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1) active
2022-11-21 17:32:30.021 [DBG] NTFN: Updated spend hint to height=764158 for unconfirmed spend request outpoint=844853510330eb4e0aff5bddd2f9b6b9b989d9f141046940ef2999867071c020:1, script=0 593f54fee8751d033cb40eabe1b881b77f352459ca161e07e8cb72a0d8fd3d56
2022-11-21 17:32:30.021 [INF] CNCT: Close observer for ChannelPoint(844853510330eb4e0aff5bddd2f9b6b9b989d9f141046940ef2999867071c020:1) active
2022-11-21 17:32:30.021 [DBG] CNCT: Starting ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1), htlc_set=(map[contractcourt.HtlcSetKey]contractcourt.htlcSet) (len=2) {
 (contractcourt.HtlcSetKey) LocalHtlcSet: (contractcourt.htlcSet) {
  incomingHTLCs: (map[uint64]channeldb.HTLC) {
  },
  outgoingHTLCs: (map[uint64]channeldb.HTLC) {
  }
 },
 (contractcourt.HtlcSetKey) RemoteHtlcSet: (contractcourt.htlcSet) {
  incomingHTLCs: (map[uint64]channeldb.HTLC) {
  },
  outgoingHTLCs: (map[uint64]channeldb.HTLC) {
  }
 }
}
, state=StateDefault
2022-11-21 17:32:31.582 [INF] CNCT: ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1): starting state=StateDefault, trigger=chainTrigger, triggerHeight=764158
2022-11-21 17:32:31.582 [DBG] CNCT: ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1): attempting state step with trigger=chainTrigger from state=StateDefault
2022-11-21 17:32:31.582 [DBG] CNCT: ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1): new block (height=764158) examining active HTLC's
2022-11-21 17:32:31.584 [DBG] CNCT: ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1): checking commit chain actions at height=764158, in_htlc_count=0, out_htlc_count=0
2022-11-21 17:32:31.584 [DBG] CNCT: ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1): no actions for chain trigger, terminating
2022-11-21 17:32:31.584 [DBG] CNCT: ChannelArbitrator(8dfab6f6674b126e496d1bdf9a2d20002540054fb7e6606f5f6c27250afbbc01:1): terminating at state=StateDefault
2022-11-21 17:32:31.584 [DBG] CNCT: Starting ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0), htlc_set=(map[contractcourt.HtlcSetKey]contractcourt.htlcSet) (len=2) {
 (contractcourt.HtlcSetKey) LocalHtlcSet: (contractcourt.htlcSet) {
  incomingHTLCs: (map[uint64]channeldb.HTLC) {
  },
  outgoingHTLCs: (map[uint64]channeldb.HTLC) {
  }
 },
 (contractcourt.HtlcSetKey) RemoteHtlcSet: (contractcourt.htlcSet) {
  incomingHTLCs: (map[uint64]channeldb.HTLC) {
  },
  outgoingHTLCs: (map[uint64]channeldb.HTLC) {
  }
 }
}
, state=StateDefault
2022-11-21 17:32:32.207 [INF] CNCT: ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0): starting state=StateDefault, trigger=chainTrigger, triggerHeight=764158
2022-11-21 17:32:32.209 [DBG] CNCT: ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0): attempting state step with trigger=chainTrigger from state=StateDefault
2022-11-21 17:32:32.209 [DBG] CNCT: ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0): new block (height=764158) examining active HTLC's
2022-11-21 17:32:32.209 [DBG] CNCT: ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0): checking commit chain actions at height=764158, in_htlc_count=0, out_htlc_count=0
2022-11-21 17:32:32.209 [DBG] CNCT: ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0): no actions for chain trigger, terminating
2022-11-21 17:32:32.209 [DBG] CNCT: ChannelArbitrator(c9efddcad4527435b45eaffc53797d5b6a5b32330f688fee0b9d45e2d12d580e:0): terminating at state=StateDefault
2022-11-21 17:32:32.209 [DBG] CNCT: Starting ChannelArbitrator(844853510330eb4e0aff5bddd2f9b6b9b989d9f141046940ef2999867071c020:1), htlc_set=(map[contractcourt.HtlcSetKey]contractcourt.htlcSet) (len=2) {
 (contractcourt.HtlcSetKey) LocalHtlcSet: (contractcourt.htlcSet) {
  incomingHTLCs: (map[uint64]channeldb.HTLC) {
  },
  outgoingHTLCs: (map[uint64]channeldb.HTLC) {
  }
 },
 (contractcourt.HtlcSetKey) RemoteHtlcSet: (contractcourt.htlcSet) {
  incomingHTLCs: (map[uint64]channeldb.HTLC) {
  },
  outgoingHTLCs: (map[uint64]channeldb.HTLC) {
  }
 }
}
, state=StateDefault
[... cut ...]
2022-11-21 17:34:41.226 [DBG] RPCS: [channelbalance] local_balance=1386671007 mSAT remote_balance=49273993 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-11-21 17:34:42.332 [DBG] RPCS: [feereport]
2022-11-21 17:35:58.470 [INF] HLCK: Health check: chain backend, call: 1 failed with: health check: chain backend timed out after: 30s, backing off for: 2m0s
2022-11-21 17:38:28.471 [INF] HLCK: Health check: chain backend, call: 2 failed with: health check: chain backend timed out after: 30s, backing off for: 2m0s

rootzoll, 2022-11-21:
hmm - this ow looks like its running with some strange chain backend timeouts at the end. 

@guggero can you see anything it the two log posts above that points to the reason LND going inactive as a service?

rootzoll, 2022-11-21:
OK got report in that its running agian after Auto-Pilot was turned on (not sure why): https://twitter.com/elpasobitcoin/status/1594749432968675329?s=20&t=1IG1Md5sks1MX--0anTgbA

Mark for close if no further error report comes in within the next days.

bitsam, 2022-11-22:
I have also been having the LND service stop on my node every 10 minutes to an hour at best for the past week or more. here is the link to my debug log. [https://l.termbin.com/918j](url)

rootzoll, 2022-11-22:
OK I can see from LND the following error: 

```
SIGILL: illegal instruction PC=0x41c830 m=0 sigcode=1 instruction bytes: 0x0 0x0 0xa3 0x99 0xe1 0x7 0x40 0xb2 0xe2 0x3 0x1 0xaa 0xed 0x7 0xf1 0x97 goroutine 1 [running, locked to thread]: google.golang.org/grpc/binarylog/grpc_binarylog_v1.init() google.golang.org/grpc@v1.38.0/binarylog/grpc_binarylog_v1/binarylog.pb.go:1020 +0x7d0 fp=0x40001bf930 sp=0x40001bf8d0 pc=0x41c830 runtime.doInit(0x1d21ce0) runtime/proc.go:6222 +0x134 fp=0x40001bfa70 sp=0x40001bf930 pc=0x563c4 runtime.doInit(0x1d2ace0) runtime/proc.go:6199 +0x70 fp=0x40001bfbb0 sp=0x40001bfa70 pc=0x56300 runtime.doInit(0x1d35640) runtime/proc.go:6199 +0x70 fp=0x40001bfcf0 sp=0x40001bfbb0 pc=0x56300 runtime.doInit(0x1d2d2e0) runtime/proc.go:6199 +0x70 fp=0x40001bfe30 sp=0x40001bfcf0 pc=0x56300 runtime.doInit(0x1d375e0) runtime/proc.go:6199 +0x70 fp=0x40001bff70 sp=0x40001bfe30 pc=0x56300 runtime.main() runtime/proc.go:233 +0x1fc fp=0x40001bffd0 sp=0x40001bff70 pc=0x483ec runtime.goexit() runtime/asm_arm64.s:1263 +0x4 fp=0x40001bffd0 sp=0x40001bffd0 pc=0x79264 r0 0xf33000 r1 0x3 r2 0x0 r3 0x40001de270 r4 0xe83420 r5 0x400011b1e0 r6 0x0 r7 0x10c589a r8 0x3 r9 0x400011b1e0 r10 0x0 r11 0xd r12 0x1 r13 0x13ac6e2 r14 0x400011b28f r15 0x1000 r16 0x40001b03a0 r17 0x40001bf860 r18 0x0 r19 0x8 r20 0x40001bfa50 r21 0x40001076b0 r22 0x4000004000 r23 0x12209c8 r24 0xffffffffffffffa2 r25 0x36 r26 0x1226200 r27 0x1ddd370 r28 0x40000021a0 r29 0x40001bf8c8 lr 0x41c7f0 sp 0x40001bf8d0 pc 0x41c830 fault 0x0
```

That looks like a very special edge case error. Best to report it over to the LND developer Slack or search/open issue on their repo: https://github.com/lightningnetwork/lnd

guggero, 2022-11-23:
@bitsam can you please post more of the lnd log? Looks like not everything relevant made it into the RaspiBlitz log. And creating an issue as @rootzoll mentioned would be helpful.

Whabster, 2022-11-23:
I have a similar problem. I work was away from my node for a couple of months. When I got back it was not showing its normal display. It was asking for a password. After entering the password it has been synching for days. I tried resetting a couple of times with no luck. I managed to get the debug log. Please see below:

* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.1
chainnetwork: bitcoin / main
 12:12:12 up 17 min,  3 users,  load average: 0.68, 0.95, 0.97

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin-Daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-11-23 11:55:48 GMT; 16min ago
  Process: 3252 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3257 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
 Main PID: 3271 (bitcoind)
    Tasks: 18 (limit: 4532)
   CGroup: /system.slice/bitcoind.service
           └─3271 /usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log

Nov 23 11:55:48 raspberrypi systemd[1]: Starting Bitcoin-Daemon...
Nov 23 11:55:48 raspberrypi systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Wed 2022-11-23 11:17:01 GMT, end at Wed 2022-11-23 12:12:12 GMT. --
Nov 23 11:55:48 raspberrypi systemd[1]: Starting Bitcoin-Daemon...
Nov 23 11:55:48 raspberrypi systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-11-23T12:01:30Z UpdateTip: new best=0000000000000000000408c4b842af89cf159d4ae0b9886fabe17835329d04fc height=764393 version=0x2008a000 log2_work=93.859024 tx=782577111 date='2022-11-23T09:46:38Z' progress=0.999975 cache=113.1MiB(860039txo)
2022-11-23T12:01:32Z UpdateTip: new best=00000000000000000001447483e0a6677df82708809466dcde7a77c636e7441a height=764394 version=0x20800000 log2_work=93.859037 tx=782578042 date='2022-11-23T09:49:53Z' progress=0.999976 cache=113.5MiB(863397txo)
2022-11-23T12:01:32Z Socks5() connect to y7uhpgrniupalqt76pahxrbbvxw7evyawpgveuklkjjrpesued2a7oid.onion:8333 failed: host unreachable
2022-11-23T12:01:34Z UpdateTip: new best=00000000000000000005b0961857ee4fdc637bee13a2672a1dacb8b67fb9c9f0 height=764395 version=0x3fffe000 log2_work=93.859050 tx=782580262 date='2022-11-23T10:01:33Z' progress=0.999978 cache=114.1MiB(867942txo)
2022-11-23T12:01:35Z UpdateTip: new best=00000000000000000000f6665d6cad54bd9f80e4ef36a126fcf95f6b73b738d6 height=764396 version=0x2713c000 log2_work=93.859062 tx=782581367 date='2022-11-23T10:06:30Z' progress=0.999979 cache=113.7MiB(865324txo)
2022-11-23T12:01:36Z UpdateTip: new best=00000000000000000002208b81880bd13c7237125d893bb1cd9285a98064957c height=764397 version=0x2000e000 log2_work=93.859075 tx=782584463 date='2022-11-23T10:34:01Z' progress=0.999984 cache=114.6MiB(871875txo)
2022-11-23T12:01:38Z UpdateTip: new best=00000000000000000000990b393472ab4951a02c8a1589acd77e01f30a3f0a8a height=764398 version=0x20008004 log2_work=93.859088 tx=782586383 date='2022-11-23T10:34:01Z' progress=0.999984 cache=114.9MiB(874816txo)
2022-11-23T12:01:41Z UpdateTip: new best=00000000000000000005e273b1374fb7d3fd1fedfada9e26998d208fc3200d69 height=764399 version=0x20600000 log2_work=93.859101 tx=782588481 date='2022-11-23T10:44:33Z' progress=0.999986 cache=115.2MiB(877219txo)
2022-11-23T12:01:41Z New outbound peer connected: version: 70016, blocks=764405, peer=13 (block-relay-only)
2022-11-23T12:01:43Z UpdateTip: new best=00000000000000000000e3aa18526442e7bf5027b519f284d626f63a346b537f height=764400 version=0x20000000 log2_work=93.859113 tx=782591647 date='2022-11-23T11:04:18Z' progress=0.999989 cache=116.0MiB(884014txo)
2022-11-23T12:01:43Z Leaving InitialBlockDownload (latching to false)
2022-11-23T12:01:43Z Imported mempool transactions from disk: 0 succeeded, 0 failed, 4137 expired, 0 already there, 0 waiting for initial broadcast
2022-11-23T12:01:43Z loadblk thread exit
2022-11-23T12:02:15Z Socks5() connect to 2pk7ehyikkkffngfgcxjur6dpalt426uwhdzgoh2bahlyysdogunhiyd.onion:8333 failed: host unreachable
2022-11-23T12:02:19Z UpdateTip: new best=0000000000000000000624f432b69da73136e19739004e1ec00201c825b40364 height=764401 version=0x30eaa004 log2_work=93.859126 tx=782593974 date='2022-11-23T11:14:09Z' progress=0.999991 cache=116.7MiB(889124txo)
2022-11-23T12:02:21Z UpdateTip: new best=0000000000000000000583d0865a00c01cfc8c1185ad93264cf9c79965a56eb8 height=764402 version=0x287a8000 log2_work=93.859139 tx=782597399 date='2022-11-23T11:43:04Z' progress=0.999996 cache=117.7MiB(897517txo)
2022-11-23T12:02:34Z UpdateTip: new best=000000000000000000002aef90fa4bf2eaf0683d72686b3bac4122ff7e40546e height=764403 version=0x20000000 log2_work=93.859152 tx=782599222 date='2022-11-23T11:48:19Z' progress=0.999997 cache=126.0MiB(907405txo)
[... cut ...]
initialSync=0
syncProgress=100.00 
startcountLightning=1
lndActive=1
walletLocked=0
lndPeers=0
syncedToChain=0
scanTimestamp=1669204204
scanDate='Wed 23 Nov 11:50:04 GMT 2022'
scanProgress=99.99
lndRPCReady=1
clActive=0
CLwalletLocked=0
blitzTUIActive=0
blitzTUIRestarts=0
vagrant=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
debug | torsocks nc termbin.com 9999

guggero, 2022-11-23:
Looks like your node is still starting up. Please run `sudo tail -n 500 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log` and share that here if this doesn't resolve itself over night.

Whabster, 2022-11-23:
Thanks Guggero. Will do.

Whabster, 2022-11-24:
Morning Guggero. No luck. My node is still stuck. Here is the result of the log you asked for.

2022-11-23 11:55:51.059 [INF] LTND: Version: 0.13.3-beta commit=v0.13.3-beta, build=production, logging=default, debuglevel=debug
2022-11-23 11:55:51.059 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2022-11-23 11:55:51.119 [DBG] WLKT: WalletKit RPC server successfully registered with root gRPC server
2022-11-23 11:55:51.119 [DBG] RRPC: Router RPC server successfully register with root gRPC server
2022-11-23 11:55:51.119 [DBG] NTFR: ChainNotifier RPC server successfully register with root gRPC server
2022-11-23 11:55:51.119 [DBG] IRPC: Invoices RPC server successfully registered with root gRPC server
2022-11-23 11:55:51.119 [DBG] SGNR: Signer RPC server successfully register with root gRPC server
2022-11-23 11:55:51.119 [DBG] ARPC: Autopilot RPC server successfully register with root gRPC server
2022-11-23 11:55:51.119 [DBG] VRPC: Versioner RPC server successfully registered with root gRPC server
2022-11-23 11:55:51.119 [INF] RPCS: RPC server listening on 0.0.0.0:10009
2022-11-23 11:55:51.121 [DBG] WLKT: WalletKit REST server successfully registered with root REST server
2022-11-23 11:55:51.127 [DBG] RRPC: Router REST server successfully registered with root REST server
2022-11-23 11:55:51.137 [DBG] NTFR: ChainNotifier REST server successfully registered with root REST server
2022-11-23 11:55:51.137 [DBG] IRPC: Invoices REST server successfully registered with root REST server
2022-11-23 11:55:51.138 [DBG] SGNR: Signer REST server successfully registered with root REST server
2022-11-23 11:55:51.138 [DBG] ARPC: Autopilot REST server successfully registered with root REST server
2022-11-23 11:55:51.138 [DBG] VRPC: Versioner REST server successfully registered with root REST server
2022-11-23 11:55:51.138 [INF] RPCS: gRPC proxy started at 0.0.0.0:8080
2022-11-23 11:55:51.138 [INF] LTND: Opening the main database, this might take a few minutes...
2022-11-23 11:55:51.138 [INF] LTND: Opening bbolt database, sync_freelist=true, auto_compact=false
2022-11-23 11:55:51.713 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2022-11-23 11:55:52.618 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2022-11-23 11:55:52.891 [INF] CHDB: Checking for schema update: latest_version=22, db_version=22
2022-11-23 11:55:52.892 [INF] LTND: Database now open (time_to_open=1.753183776s)!
2022-11-23 11:55:52.892 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2022-11-23 11:56:03.325 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:56:06.990 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:56:10.600 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:56:14.301 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:56:17.899 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:56:21.508 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:56:25.188 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:57:00.001 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:57:07.983 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:57:25.773 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:57:44.380 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:57:55.318 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:58:01.366 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:58:12.086 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:58:29.146 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:58:39.976 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:58:44.919 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:58:53.625 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:59:02.534 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:59:13.354 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:59:24.932 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:59:38.271 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 11:59:49.539 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-11-23 12:00:02.849 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
[... cut ...]
2022-11-24 08:01:14.662 [DBG] LNWL: Filtering block 764511 (00000000000000000006700f3bb707c03d78ca4b6371b4f104a936e456555240) with 1379 transactions
2022-11-24 08:01:15.029 [DBG] LNWL: Received block connected notification for height 764511 while rescanning
2022-11-24 08:03:09.678 [DBG] LNWL: Filtering block 764512 (00000000000000000002fd09008ae723bbc93ade31417a5b4152ba892e9992f9) with 464 transactions
2022-11-24 08:03:10.026 [DBG] LNWL: Received block connected notification for height 764512 while rescanning

guggero, 2022-11-24:
`Version: 0.13.3-beta commit=v0.13.3-beta` you need to update to the latest version, `v0.15.4-beta`, otherwise the node will never get unstuck.

Whabster, 2022-11-24:
Ok, so I've updated the lightning node.
It has been running again for about 4 hours and still hasn't got to 100% on LND. I ran the "sudo tail" command again and it doesn't show the version number. ?!? 

sudo tail -n 500 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.g
2022-11-24 12:01:08.956 [DBG] LNWL: Filtering block 764117 (000000000000000000029315960b6be63fb7be7e361e1c2e9dd8f1e189758b8b) with 56 transactions
2022-11-24 12:01:09.316 [DBG] LNWL: Filtering block 764118 (0000000000000000000550112452ed49d3717a01c8ae8d044577cca711061ec8) with 2619 transactions
2022-11-24 12:01:09.972 [DBG] LNWL: Filtering block 764119 (000000000000000000009e12596f9fb3dca945868aeafcca1c3a89963b8ac690) with 2663 transactions
2022-11-24 12:01:10.629 [DBG] LNWL: Filtering block 764120 (00000000000000000006e172893c0784cbc3cf9696d6f06c71751b4080065eed) with 3230 transactions
2022-11-24 12:01:11.305 [DBG] LNWL: Filtering block 764121 (000000000000000000058f48850c4828c96a1f77f52128c57f03064b87a63187) with 3235 transactions
2022-11-24 12:01:11.931 [DBG] LNWL: Filtering block 764122 (00000000000000000004c41efd103ab861678235889fe8c444a479e721f832d1) with 2398 transactions
2022-11-24 12:01:12.502 [DBG] LNWL: Filtering block 764123 (000000000000000000005e7163bca44fef8ca151e1282ef4fbf6b115c9e87a89) with 1792 transactions
2022-11-24 12:01:13.059 [DBG] LNWL: Filtering block 764124 (00000000000000000000a8103708a70855c96311bdb7d4a2432504def2d968a4) with 2925 transactions
2022-11-24 12:01:13.679 [DBG] LNWL: Filtering block 764125 (00000000000000000005fc705a9568bd282691e64ec048db99e3a2b61787df94) with 1961 transactions
2022-11-24 12:01:14.293 [DBG] LNWL: Filtering block 764126 (00000000000000000006468783cd81c459203ea83feb476148e623b07759925c) with 2861 transactions
2022-11-24 12:01:14.946 [DBG] LNWL: Filtering block 764127 (00000000000000000005800a68941af35a27384ef27f7b452fa8fb11e3516dfd) with 3547 transactions
2022-11-24 12:01:15.637 [DBG] LNWL: Filtering block 764128 (000000000000000000003094bc83d3ce080501c91f7c8475823fd7b596f47ec3) with 2421 transactions
2022-11-24 12:01:16.194 [DBG] LNWL: Filtering block 764129 (00000000000000000005c41e8957c91f196c47dedf33e5da31a0bc54f8da4a8a) with 886 transactions
2022-11-24 12:01:16.704 [DBG] LNWL: Filtering block 764130 (00000000000000000005b0019b4a0cf56774ab593f7342f2989109d5ff2e23a5) with 2916 transactions
2022-11-24 12:01:17.271 [DBG] LNWL: Filtering block 764131 (00000000000000000006b073f8bcfcb405e18511bde4c9a3413f4ac9910dc86c) with 2170 transactions
2022-11-24 12:01:17.528 [DBG] LNWL: Filtering block 764132 (0000000000000000000239622cc32ec18d6b027e290e6089f0f5e4936e73bf3e) with 243 transactions
2022-11-24 12:01:17.858 [DBG] LNWL: Filtering block 764133 (000000000000000000031bcbf6ab0aa65653c45a756730ee9319ce00de0c51b9) with 2699 transactions
2022-11-24 12:01:18.445 [DBG] LNWL: Filtering block 764134 (00000000000000000005d2d4b366a1b6d79435287b493c0edb48c7733519eb38) with 2777 transactions
2022-11-24 12:01:18.999 [DBG] LNWL: Filtering block 764135 (00000000000000000004fa9ddb6182d70cf624d990abab18f4bc5e256ce51d87) with 1194 transactions
2022-11-24 12:01:19.608 [DBG] LNWL: Filtering block 764136 (000000000000000000013ecaaf4c18a53ad06455a2207beddc09455bcb1964a6) with 3099 transactions
2022-11-24 12:01:20.432 [DBG] LNWL: Filtering block 764137 (00000000000000000001d2feb26eedb2235f46e91aebcb213f25020ee3c72a3e) with 2226 transactions
2022-11-24 12:01:21.092 [DBG] LNWL: Filtering block 764138 (0000000000000000000787ac6882fb5f6b187ca5b71e72ecbb6bf28ead9ce56b) with 3359 transactions
2022-11-24 12:01:21.725 [DBG] LNWL: Filtering block 764139 (0000000000000000000133aebe2c2e709eee6031098b17717c4b30e1ece7622d) with 2280 transactions
2022-11-24 12:01:22.330 [DBG] LNWL: Filtering block 764140 (00000000000000000002dc736bcfe40712f0752d5f2dfc0983733a9d502354de) with 3109 transactions
2022-11-24 12:01:23.011 [DBG] LNWL: Filtering block 764141 (000000000000000000049ea605ab7edd85f68880b2e46845d41547ae5a8350c2) with 2017 transactions
2022-11-24 12:01:23.563 [DBG] LNWL: Filtering block 764142 (0000000000000000000333bb1f5bde7d90f70b2bbc9af9d33dad5b43b49417b6) with 1742 transactions
2022-11-24 12:01:24.102 [DBG] LNWL: Filtering block 764143 (000000000000000000013c8ad2ada0426e0755d32e33bf279d664addcd1e7f3a) with 1810 transactions
2022-11-24 12:01:24.719 [DBG] LNWL: Filtering block 764144 (00000000000000000006d2a2b9d6360bf34dc889b5e9223956cb3362ccbf994b) with 2034 transactions
2022-11-24 12:01:25.451 [DBG] LNWL: Filtering block 764145 (00000000000000000000b096de2c749c9af31898ad6bf64145ee7e6008686fd5) with 894 transactions
2022-11-24 12:01:25.911 [DBG] LNWL: Filtering block 764146 (00000000000000000001ae10fe2f01cbbd5dc990f9dda20b1c85ad809c4e8b15) with 433 transactions
2022-11-24 12:01:26.389 [DBG] LNWL: Filtering block 764147 (000000000000000000058c6d53fe6852787bf72fbbf1e5c5cd1739cfea733b4c) with 1683 transactions
2022-11-24 12:01:26.757 [DBG] LNWL: Filtering block 764148 (0000000000000000000633560de1e37e6cb48e44d63748346423ef2b2f834a4e) with 711 transactions
2022-11-24 12:01:27.185 [DBG] LNWL: Filtering block 764149 (00000000000000000000349ddddf7c942b250002263f77e6a8cc45d15ba29e19) with 1919 transactions
2022-11-24 12:01:27.865 [DBG] LNWL: Filtering block 764150 (000000000000000000076127fd05551a62da91ea6854b08a0c5dae1cb79459b2) with 2447 transactions
2022-11-24 12:01:28.478 [DBG] LNWL: Filtering block 764151 (00000000000000000000f52e421f97d21e0a58b507d977cd64dc344b5f94ec8b) with 3301 transactions
[... cut ...]
2022-11-24 12:39:51.386 [INF] NTFN: New block: height=764535, sha=00000000000000000006764fe50e747c85281e4ece36495839b88676772723de
2022-11-24 12:39:51.386 [DBG] SWPR: New block: height=764535, sha=00000000000000000006764fe50e747c85281e4ece36495839b88676772723de
2022-11-24 12:39:51.387 [INF] UTXN: Attempting to graduate height=764535: num_kids=0, num_babies=0
2022-11-24 12:39:51.387 [DBG] DISC: New block: height=764535, hash=00000000000000000006764fe50e747c85281e4ece36495839b88676772723de
2022-11-24 12:45:17.474 [DBG] DISC: No eligible active syncer to rotate

Whabster, 2022-11-24:
Thanks Guggero. It's back up and running. I am a novice at these sorts of things. Couldn't have done it without your help.