raspiblitz/raspiblitz #3468: Electrs isnt working

ShogunWarlord, 2022-11-22:
[image]
[image]

cwgriswold, 2022-11-22:
Update to 0.9.9 or 0.9.10

rootzoll, 2022-11-22:
Electrs 0.9.10 will be part of next RaspiBlitz 1.9.0 release. Merged. Final testing.

ShogunWarlord, 2022-11-23:
> Update to 0.9.9 or 0.9.10

Can you show me how to update it?

ShogunWarlord, 2022-11-23:
> Update to 0.9.9 or 0.9.10

Any documentation on how to do it?

willheisenberg, 2022-11-23:
But still if you do that and I try to connect electrs, i got this error:

######## How to Connect to Electrum Rust Server #######

error='missing second parameter - see help'

so i dont get the onion address

rootzoll, 2022-11-24:
TODO: Test connection

jukeboxnode, 2022-11-27:
having same issues and wondering is there a upgrade script we could try out?

rootzoll, 2022-11-27:
So if anyone cant wait for electrs v0.9.10 coming with the next release - you can try the following manual command (not tested):
```
cd /home/electrs/electrs
sudo -u electrs git fetch
sudo -u electrs git pull -p
sudo -u electrs git reset --hard v0.9.10
sudo -u electrs /home/electrs/.cargo/bin/cargo build --locked --release
sudo systemctl start electrs
```

cwgriswold, 2022-11-28:
Update to 0.9.10 worked

rootzoll, 2022-12-24:
TODO: Had on quicktesting 1.9.0RC1 a problem to connect to ElectrumApp (did not had time to check deeper) - so up for retest for RC2.

Tracachang, 2023-01-14:
Updated manually to Electrs v0.9.11, tested with Electrum and Sparrow without issues. We may set 0.9.11 in 1.90 release?

rootzoll, 2023-01-15:
testing v0.9.11 install right now

rootzoll, 2023-01-15:
PR now open for v0.9.11 update: #3613

rootzoll, 2023-01-16:
On testmachine I get status after Index is build:
```
##### STATUS ELECTRS SERVICE
version='v0.9.10'
configured=1
serviceInstalled=1
serviceRunning=1
localIP='192.168.178.60'
publicIP='91.65.33.149'
portTCP='50001'
localTCPPortActive=1
publicTCPPortAnswering=0
portSSL='50002'
localHTTPPortActive=1
publicHTTPPortAnswering=0
TorRunning=1
nginxTest=1
serviceRunning=1
syncProgress=0%
tipSynced=0
electrumResponding=1
initialSynced=1
```
`tipSynced=0` is looking strange on  my testmachine. Logs looking empty when calling `sudo journalctl -u electrs --no-pager -n2000` -->
```
Jan 16 14:13:03 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Jan 16 14:13:03 raspberrypi systemd[1]: electrs.service: Consumed 14h 45min 16.929s CPU time.
Jan 16 14:14:03 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 1.
Jan 16 14:14:03 raspberrypi systemd[1]: Stopped Electrs.
Jan 16 14:14:03 raspberrypi systemd[1]: electrs.service: Consumed 14h 45min 16.929s CPU time.
Jan 16 14:14:03 raspberrypi systemd[1]: Started Electrs.
```

Check systemd loglevels and general functions.

rootzoll, 2023-02-03:
merged for v1.9.0RC2 release

closed 2023-02-03 by rootzoll