raspiblitz/raspiblitz #3492: An error occured: server is still in the process of starting

Undergotten, 2022-12-04:
Hi all! My RaspiBlitz node was offline for a few months because of personal problems. Got it back up and running, and I would like to close a channel I left unattended and get my funds back so I can start again, but I keep getting this error. I was running 1.7 and upgraded to 1.8 because I thought that might be the problem, but I'm still getting this message whenever I try to close the channel. Any ideas what might be causing this?

I'm not very technical, but I can give logs to help if someone can tell me what directory on the raspiblitz they are in. Thanks in advance

rootzoll, 2022-12-07:
Does this problem still persist? Which version of LND are you running?

Undergotten, 2022-12-07:
> 

Yes, it still persists. I'm not sure how to tell what version I'm running. I just flashed the image to the pi4. I got the error message trying to close the channel in both Thunderhub and in the RaspiBlitz WEBui main menu. 

I just tried to force close it in RideTheLightning and did not recieve an error. I will update you on if that changes things. 

How can I check what version of LND I'm running? I can SSH to the pi4 RaspiBlitz, but the menu doesn't load. It just stays at 100% syncing. If I go to CLI I can tell you, if you let me know what directory to look in for logs/data. 

Thank you for your help. You do amazing work.

rootzoll, 2022-12-08:
> How can I check what version of LND I'm running? I can SSH to the pi4 RaspiBlitz, but the menu doesn't load. It just stays at 100% syncing. If I go to CLI I can tell you, if you let me know what directory to look in for logs/data.

OK - it its syning its not ready to work with thunderhub or RTL. In SSH you can CTRL+c to exit to the terminal and then type `menu`. Under INFO you should see LND Version. Please report back here.

Undergotten, 2022-12-09:
> 

Been using this thing for a year and had no idea about the "menu" command. That's very handy. Thank you! 

I am running LND 0.15.0-beta. Is that the latest?

rootzoll, 2022-12-09:
If you run into further problems (updating the running system) just do the normal system update way by flashing a new sd card with the 1.8.0c image you can download here: https://github.com/rootzoll/raspiblitz#downloading-the-software .. it contains all the LND bugfixes.

closed 2022-12-09 by rootzoll