raspiblitz/raspiblitz #3499: waiting for HDD/SSD

mattias232, 2022-12-07:
Had to reflash the SD card as issue with LN and then when it has loaded i now get this issue? 

Previously i have switched the usb cable to a different port and its worked but not this time. I find it too coincedental to assume the SSD has died at the same time? Any advice please?

[image]

ChuckNorrison, 2022-12-07:
Power off the blitz and connect your ssd to another PC and check the partitions if it is valid and if data can be found. Could be a result of data corruption or hardware failure.

mattias232, 2022-12-07:
i've tried it on my mac and pc and it doesnt recognise the SSD.

i've ran the command and get the following output

```
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 29.81 GiB, 32010928128 bytes, 62521344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: [*******]

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 62521343 61988864 29.6G 83 Linux
```

ChuckNorrison, 2022-12-07:
Your SSD is not available :( Do you have a powered USB Hub to test with?

mattias232, 2022-12-09:
i've tried that but still doesnt seem to see anything

I'm using a ugreen ssd enclosure - should i replace this?  other than this its got to be a full re-install, right?

ChuckNorrison, 2022-12-09:
The urgreen is great. Looks like SSD failure.

mattias232, 2022-12-09:
ok, i'll have to geta new one. is the process fairly straight forward? presume it will take time to sync blockchains?

ChuckNorrison, 2022-12-09:
After fresh install you can restore your lightning with the channel backup file and the seed phrase. Open channels will get force closed. If you are using c-lightning, a current channel backup should be found on your SD card if not  created.

You can find a good description in the README on how to restore.

The Blockchain sync can take some time (1-2 weeks), especially if Tor is used.

Good luck and maybe check your old SSD again with some tools. Maybe it's just partition table defect.

mattias232, 2022-12-10:
ok, ive got a new SSD and it was recognised and i am restoring. 

question i have is i had some funds on LN (no open channels though), i have the backup seed for LN is this enough to recover the funds when the blockchains all sync?

ChuckNorrison, 2022-12-10:
Yes, you can recover without a channel backup the on chain funds with just the seed word phrase.

rootzoll, 2023-03-13:
There was a similar report so dig down where this is coming from:

For this error to be shown on LCD it has to the "noHDD" state and that is only set when the follwing is true:
```
[ "${isMounted}" == "0" ] && [ "${hddCandidate}" == "" ]
```

So if you SSH in and try on terminal the command `sudo /home/admin/config.scripts/blitz.datadrive.sh status` you should see: `isMounted=0` and and empty string for `hddCandidate`

OK so lets take a look at the script how it checks for hddCandidates: 

It does bascially an `lsblk` and just filters for certain names:
https://github.com/rootzoll/raspiblitz/blob/1ae636b265a21edf2102787e5bbff0984727fc1c/home.admin/config.scripts/blitz.datadrive.sh#L106

If you have problems with this script, please post the output of the following command for further debugging:
`sudo lsblk -o NAME,SIZE -b`

moguntiacum, 2023-03-20:
If I ssh in, I only get the above-mentioned message "Waiting for HDD/SSD...".  I cannot type any command. I am on a Windows machine, so sudo also does not work.

rootzoll, 2023-03-20:
Are you using the web browser or a ssh terminal to interact with the raspiblitz?

moguntiacum, 2023-03-20:
I used cmd, Windows PowerShell, and a web browser - all three lead to the same result. The commands that you mention above do not work there.

moguntiacum, 2023-03-20:
Thanks! That worked. The output is:

NAME               SIZE
mmcblk0     32010928128
├─mmcblk0p1   268435456
└─mmcblk0p2 31738298368

moguntiacum, 2023-03-23:
Thanks! This is what I suspected. Yes, it is a setup by FULMO. I already wrote to the shop team but received no answer. I will write again and include a link to this thread.

moguntiacum, 2023-04-11:
Can you advise us on how to return it as of guarantee? Unfortunately, I still got no reply from the shop team on where to find the return label and the shop account. When I bought the setup, there was no shop account to register at, so I do not have a shop account. Thanks!